probe-rs/probe-rs #922: Out of bounds access in probe-rs

korken89, 2021-12-12:
**Describe the bug**

I randomly get crashes like this when using `probe-run` and `cargo embed`:

```
error: Error communicating with probe: A core architecture specific error occured
thread 'main' panicked at 'index out of bounds: the len is 3 but the index is 3', /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.12.0/src/probe/cmsisdap/commands/transfer/mod.rs:186:24
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**To Reproduce**
Steps to reproduce the behavior:
1. No idea, seems to happen randomly

**Expected behavior**
No crash.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - Arch Linux 5.15.7

adamgreig, 2021-12-12:
Huh, weird. It looks like this would happen if the probe returns _more_ transfers than we requested. It's allowed to return fewer when one transfer fails, and ideally returns the same number if they were all OK, but it seems like it should not return more unless something's gone wrong. What probe are you using?

In any case it shouldn't panic with an invalid slice access so I've opened #923 to convert it to a log and an error, but I'm at a loss as to what could cause it to start with. Are you able to capture it with RUST_LOG=trace set?

korken89, 2021-12-12:
Hi, thanks for giving it a look! It's with a dap42 probe, it does however never happen with my HS probe.
I think the fix in #923 looks good.

I'll try to catch it, but it happens so randomly that I can't guarantee it.

commit 8485e490 (2021-12-13): Merge #923

923: cmsisdap: Check for DAP_Transfer response with more transfers than requested r=Yatekii a=adamgreig

In #922 it looks like `transfer_count` which is read from the response data could be larger than `self.transfers.len()`, so we try and access `self.transfers` out of bounds. This isn't expected from a probe (it shouldn't be able to execute _more_ transfers than we asked...) but at least makes sense to emit a useful error log and return an Error instead of panicking.

Still a draft because there might be more we can do to fix #922.
PR #923 (2021-12-13): cmsisdap: Check for DAP_Transfer response with more transfers than requested
In #922 it looks like `transfer_count` which is read from the response data could be larger than `self.transfers.len()`, so we try and access `self.transfers` out of bounds. This isn't expected from a probe (it shouldn't be able to execute _more_ transfers than we asked...) but at least makes sense to emit a useful error log and return an Error instead of panicking.

Still a draft because there might be more we can do to fix #922.
closed 2021-12-13 by bors[bot]