probe-rs/probe-rs #926: ATSAME51N20A: CMSIS-DAP error on `cargo flash`

tmplt, 2021-12-15:
```shell
$ cargo flash --version
cargo-flash 0.12.1
git commit: crates.io

$ cargo flash --list-probes
The following debug probes were found:
[0]: HS-Probe with CMSIS-DAP Support (VID: 1209, PID: 4853, Serial: [REDACTED], CmsisDap)

$ cargo flash --chip atsame51n20a --connect-under-reset
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
    Flashing [REDACTED]
       Error 0: An error with the usage of the probe occured
             1: An error specific to a probe type occured
             2: CMSIS-DAP responded with an error
```

Probably some missing unlock sequence. To be investigated.

tmplt, 2021-12-15:
CC @perlindgren @vccggorski, to keep you in the loop

Yatekii, 2021-12-15:
Hmm I feel like there is the most specific part of the error missing?

Also, this is expected for atmsamd. They need a special sequence to boot, which has not yet been implemented. So them working is kinda hit or miss.

bugadani, 2024-05-17:
I've been using an ATSAMD51 for a while now without many issues, so I'll close this issue for now.

closed 2024-05-17 by bugadani