probe-rs/probe-rs #935: Invalid USB interface descriptors can silently crash the libusb stack and probe-rs

Eekle, 2022-01-02:
The libusb bug is described in https://github.com/libusb/libusb/issues/1039.

Invalid USB interface descriptors cause a read access violation, which immediately terminates execution. Probe-rs silently fails to enumerate probes with no explanation and insufficient debug log.

[image]

---

Steps to reproduce the behavior:
1. Obtain a USB device that has an interface descriptor with a `bInterfaceNumber` that is out of bounds
2. Execute any probe-rs function that calls `list_cmsisdap_devices`
3. The program will crash with a read access violation

---

Ideally, probe-rs should avoid this potential error where possible, and be able to recover from it if it does occur (though this might require a subprocess since it gets immediately terminated).

At least, the debug trace should be more elucidating.

---

**Stacktrace**
``` probe-rs-test.exe!windows_assign_endpoints(libusb_device_handle * dev_handle, unsigned char iface, unsigned char altsetting) Line 506 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:506)
probe-rs-test.exe!winusb_claim_interface(libusb_device_handle * dev_handle, unsigned char iface) Line 1912 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:1912)
probe-rs-test.exe!windows_claim_interface(libusb_device_handle * dev_handle, unsigned char interface_number) Line 665 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_common.c:665)
probe-rs-test.exe!libusb_claim_interface(libusb_device_handle * dev_handle, int interface_number) Line 1683 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\core.c:1683)
probe-rs-test.exe!auto_claim(libusb_transfer * transfer, int * interface_number, int api_type) Line 591 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:591)
probe-rs-test.exe!hid_submit_control_transfer(int sub_api, usbi_transfer * itransfer) Line 3899 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:3899)
probe-rs-test.exe!composite_submit_control_transfer(int sub_api, usbi_transfer * itransfer) Line 4303 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:4303)
probe-rs-test.exe!winusb_submit_transfer(usbi_transfer * itransfer) Line 2026 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_winusb.c:2026)
probe-rs-test.exe!windows_submit_transfer(usbi_transfer * itransfer) Line 725 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\os\windows_common.c:725)
probe-rs-test.exe!libusb_submit_transfer(libusb_transfer * transfer) Line 1553 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\io.c:1553)
probe-rs-test.exe!libusb_control_transfer(libusb_device_handle * dev_handle, unsigned char bmRequestType, unsigned char bRequest, unsigned short wValue, unsigned short wIndex, unsigned char * data, unsigned short wLength, unsigned int timeout) Line 127 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\sync.c:127)
probe-rs-test.exe!libusb_get_string_descriptor(libusb_device_handle * dev_handle, unsigned char desc_index, unsigned short langid, unsigned char * data, int length) Line 1847 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\libusb.h:1847)
probe-rs-test.exe!libusb_get_string_descriptor_ascii(libusb_device_handle * dev_handle, unsigned char desc_index, unsigned char * data, int length) Line 1104 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\libusb1-sys-0.5.0\libusb\libusb\descriptor.c:1104)
probe-rs-test.exe!rusb::device_handle::DeviceHandle<rusb::context::Context>::read_string_descriptor_ascii<rusb::context::Context>(unsigned char self) Line 661 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\rusb-0.8.1\src\device_handle.rs:661)
probe-rs-test.exe!rusb::device_handle::DeviceHandle<rusb::context::Context>::read_product_string_ascii<rusb::context::Context>(rusb::device_descriptor::DeviceDescriptor * self) Line 742 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\rusb-0.8.1\src\device_handle.rs:742)
probe-rs-test.exe!probe_rs::probe::cmsisdap::tools::get_cmsisdap_info(rusb::device::Device<rusb::context::Context> * device) Line 57 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.12.0\src\probe\cmsisdap\tools.rs:57)
probe-rs-test.exe!probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1(probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1 *, rusb::device::Device<rusb::context::Context> device) Line 20 (c:\Users\USER\.cargo\registry\src\github.com-1ecc6299db9ec823\probe-rs-0.12.0\src\probe\cmsisdap\tools.rs:20)
probe-rs-test.exe!core::ops::function::impls::impl$3::call_mut<tuple$<rusb::device::Device<rusb::context::Context>>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>(probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1 * * self, rusb::device::Device<rusb::context::Context> args) Line 270 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\core\src\ops\function.rs:270)
probe-rs-test.exe!core::iter::traits::iterator::Iterator::find_map::check::closure$0<rusb::device::Device<rusb::context::Context>,probe_rs::probe::DebugProbeInfo,ref_mut$<probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>>(core::iter::traits::iterator::Iterator::find_map::check::closure$0 *, ..., rusb::device::Device<rusb::context::Context> x) Line 2407 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\core\src\iter\traits\iterator.rs:2407)
probe-rs-test.exe!core::iter::traits::iterator::Iterator::try_fold<rusb::device_list::Devices<rusb::context::Context>,tuple$<>,core::iter::traits::iterator::Iterator::find_map::check::closure$0,enum$<core::ops::control_flow::ControlFlow<probe_rs::probe::DebugProbeInfo,tuple$<>>, 0, 1, Break>>(rusb::device_list::Devices<rusb::context::Context> * self, ..., core::iter::traits::iterator::Iterator::find_map::check::closure$0 f) Line 1994 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\core\src\iter\traits\iterator.rs:1994)
probe-rs-test.exe!core::iter::traits::iterator::Iterator::find_map<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::DebugProbeInfo,ref_mut$<probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>>(rusb::device_list::Devices<rusb::context::Context> * self, probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1 * f) Line 2413 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\core\src\iter\traits\iterator.rs:2413)
probe-rs-test.exe!core::iter::adapters::filter_map::impl$2::next<probe_rs::probe::DebugProbeInfo,rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>(core::iter::adapters::filter_map::FilterMap<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1> * self) Line 62 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\core\src\iter\adapters\filter_map.rs:62)
probe-rs-test.exe!alloc::vec::spec_from_iter_nested::impl$0::from_iter<probe_rs::probe::DebugProbeInfo,core::iter::adapters::filter_map::FilterMap<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>>(core::iter::adapters::filter_map::FilterMap<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1> iterator) Line 23 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\alloc\src\vec\spec_from_iter_nested.rs:23)
probe-rs-test.exe!alloc::vec::spec_from_iter::impl$0::from_iter<probe_rs::probe::DebugProbeInfo,core::iter::adapters::filter_map::FilterMap<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1>>(core::iter::adapters::filter_map::FilterMap<rusb::device_list::Devices<rusb::context::Context>,probe_rs::probe::cmsisdap::tools::list_cmsisdap_devices::closure$1> iterator) Line 34 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\alloc\src\vec\spec_from_i [... cut ...]
probe-rs-test.exe!std::rt::lang_start<tuple$<>>(void(*)() main, __int64 argc, unsigned char * * argv) Line 62 (f:\rustc\09c42c45858d5f3aedfa670698275303a3d19afa\library\std\src\rt.rs:62)
probe-rs-test.exe!main() (Unknown Source:0)
[Inline Frame] probe-rs-test.exe!invoke_main() Line 78 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78)
probe-rs-test.exe!__scrt_common_main_seh() Line 288 (d:\agent\_work\4\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288)
kernel32.dll!00007fff723d54e0() (Unknown Source:0)
ntdll.dll!00007fff7342485b() (Unknown Source:0)
```

Win 11

Yatekii, 2022-01-04:
Thanks a lot for the extensive report!

We have discussed this via call and Matrix quite a bit yet and sadly at the current time it seems like there is no easy fix. Reshufflin the order at which probes are polled could fix this specifc case but is not a proper general fix.

Eekle, 2022-01-04:
Yeah that's fair.

If rusb is set to spit out the debug logs from libusb, it's a lot easier to see where it fails. But that only prints it to stdout, which is hardly the most useful.

Eekle, 2022-06-15:
Fixed in rusb >=0.9.1

closed 2022-06-15 by Eekle