probe-rs/probe-rs #943: LPC55 Failed to Attach CMSIS-DAP

mriise, 2022-01-03:
**Describe the bug**
Probe-rs fails to attach to target with `Failed to get information for AP ApAddress { dp: Default, ap: 0 }`

**To Reproduce**
Steps to reproduce the behavior:
1. Link2 CMSIS-DAP or Mbed(0257 beta3) firmware on the probe for the lpcExpresso dev board
2. cargo embed --release --probe 0d28:0204

**Expected behavior**
Attach to target, flash, then debug

**Stacktrace**
```
TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching 0d28:0204
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 002: ID 0451:8140
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 001: ID 1d6b:0003
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 004: ID 043e:9a39
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 003: ID 046d:082d
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 002: ID 0451:8142
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 001: ID 1d6b:0002
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 001: ID 1d6b:0003
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 006: ID 1532:0060
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 004: ID 3297:4976
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 007: ID 1038:1294
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 005: ID 1038:1290
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 003: ID 0451:2036
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 035: ID 0d28:0204
       TRACE probe_rs::probe::cmsisdap::tools > DAPLink CMSIS-DAP: CMSIS-DAP device with 6 interfaces
       TRACE probe_rs::probe::cmsisdap::tools > Interface 0 is not HID, skipping
       TRACE probe_rs::probe::cmsisdap::tools > Interface 1 is not HID, skipping
       TRACE probe_rs::probe::cmsisdap::tools > Interface 2 is not HID, skipping
       TRACE probe_rs::probe::cmsisdap::tools >   Interface 3: CMSIS-DAP v1
       TRACE probe_rs::probe::cmsisdap::tools > Will use interface number 3 for CMSIS-DAPv1
       DEBUG probe_rs::probe::cmsisdap::tools > Opening 0d28:0204 in CMSIS-DAPv2 mode
       DEBUG probe_rs::probe::cmsisdap::commands > Draining probe of any pending data.
       DEBUG probe_rs::probe::cmsisdap::commands > Attempt 1 to find packet size
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FF]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 02, 40, 00]...
       DEBUG probe_rs::probe::cmsisdap::commands > Success: packet size is 64
       DEBUG probe_rs::probe::cmsisdap::commands > Configuring probe to use packet size 64
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FE]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 01, 08]...
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, F0]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 01, 35]...
       DEBUG probe_rs::probe::cmsisdap           > Detected probe capabilities: Capabilities { swd_implemented: true, jtag_implemented: false, swo_uart_implemented: true, swo_manchester_implemented: false, atomic_commands_implemented: true, test_domain_timer_implemented: true, swo_streaming_trace_implemented: false, uart_communication_port_implemented: false, uart_com_port_implemented: false }
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FD]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 04, 00, 10, 00]...
       DEBUG probe_rs::probe::cmsisdap           > Probe SWO buffer size: 4096
        INFO cargo_embed                         > Protocol speed 1000 kHz
       DEBUG probe_rs::config::registry          > Searching registry for chip with name LPC55S28JBD100
       DEBUG probe_rs::config::registry          > Exact match for chip name: LPC55S28JBD100
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 10, 00, 80, 00]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [10, 00]...
       DEBUG probe_rs::probe::cmsisdap           > Attaching to target system (clock = 1000kHz)
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 02, 01]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [02, 01]...
       TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 11, 40, 42, 0F, 00]...
       TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [11, 00]...
[... 39 lines cut ...]
       Error failed attaching to target
             
             Caused by:
                 Failed to get information for AP ApAddress { dp: Default, ap: 0 }
        INFO cargo_embed                                          > Metadata {
    chip: Some(
        "Unspecified(\"LPC55S28JBD100\")",
    ),
    probe: None,
    speed: Some(
        "1000",
    ),
    release: "0.12.0",
    commit: "crates.io",
}```

**Desktop (please complete the following information):**
 - Linux Debian 11

**Additional context**
I was having this issue when first adding support for the board, though I assumed it had something to do with #483, well after coming back to it to try and get rtt working its still here! (not so surprise) 

I have the the s16, s28, and s69 series dev boards and soon (provided i dont kill all 5 of my processors) my own boards with a stack of probes to test with. 

The s16 dev board I still haven't tried yet.

wt, 2022-01-03:
NXP devices can't be probed right now (at least not mine), you need to manually specify the chip type.

wt, 2022-01-03:
BTW, I just posted a PR to enable the debug probe for the s16.

mriise, 2022-01-04:
> NXP devices can't be probed right now (at least not mine), you need to manually specify the chip type.

do you mean with cli args? I was using `Embed.toml` to define the chip...

wt, 2022-01-04:
For the lpc55S16-evk, you'd use an arg like this `--chip LPC55S16JBD100`. However, that particular one will not work until my commit is merged. I don't know what the equivalents for your chips are.

Having said that, probing could probably be made to work. I got pretty close the other day, but I realized that the yamls with the chip info didn't include all the needed info to do the probe for the nxp devices, and I kinda lost interest.

mriise, 2022-01-04:
if that's the case I don't mind picking up where you left off.

 I also realized I could probably dig into pyocd to lean how they do it as well since hubris uses it to probe.

wt, 2022-01-04:
Check out `./probe-rs/src/probe`. That's where the probe logic is. The problem is that the info discovered in the debug probe doesn't appear to match directly with what is in the yaml.

Yatekii, 2022-01-04:
specifying the chip in the Embed.tol like you did is fine @mriise.

You seem to need an entry here https://github.com/wt/probe-rs/blob/d8b986e6457c913be6300ae61832f80b86c74467/probe-rs/src/config/target.rs#L92 just like @wt said.
Currently sequences are in a functional but rough state and the LPC55SXX series require sequences. But that brings us closer to having nice sequences :)

mriise, 2022-01-05:
@Yatekii since you were involved in getting the sequences for the s69, are there some sources I can refer to to make sure the s1 and s2 sequences accounted for? looking though pyocd they only have raw hex for the sequences, though even there I can see that there _are_ differences between them. [ref](https://github.com/pyocd/pyOCD/blob/main/pyocd/target/builtin/target_LPC5526Jxxxxx.py) Though the generic family sequence [is the same](https://github.com/pyocd/pyOCD/blob/main/pyocd/target/family/target_lpc5500.py).

some other documents for future notes:
debug sequence overview: https://www.nxp.com/docs/en/application-note/AN13037.pdf
flash programming tips: https://www.nxp.com/docs/en/application-note/AN12949.pdf
DAPLink folder: https://github.com/ARMmbed/DAPLink/tree/develop/source/hic_hal/nxp/lpc55xx (though im pretty sure this is just the HAL to put DAPLink on s69)

this about all ive come across with google

Tiwalun, 2022-01-05:
@mriise The debug sequences are based on the CMSIS-Packs from ARM. These sequences are described here: https://arm-software.github.io/CMSIS_5/Pack/html/debug_description.html.

You can find a CMSIS-Pack for your chip on this ARM site: https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-search, and then look at the *.pdsc file inside. The sequence for the S69 was copied from its CMSIS-Pack, so comparing the .pdsc files could be helpful to figure out the differences.

mriise, 2022-01-05:
Thanks a lot for this @Tiwalun!

mriise, 2022-01-05:
I tried using probe-rs with my s69 dev board using the jtag and CMSIS probe firmware provided by nxp with the same result as above. [gist](https://gist.github.com/mriise/9ded2f5c736b8bfdedae051fb5ec502d)

I also reviewed the debug sequences between the 3 series and found a few differences the rest being between driver files and extra logic needed for the dual core s6 series.

s1 is missing `ENABLE_SECURE_CHECKING` but it doesnt look like its currently implemented anyways. https://github.com/probe-rs/probe-rs/blob/0b057449e2455f0b55aeac0be8e6807622e434c1/probe-rs/src/architecture/arm/sequences/nxp.rs#L142
```rust
// for s28 & s69
// ENABLE_SECURE_CHECKING is set to restrictive mode, access secure addresses
if (interface.read_word_32(0x400ACFFC)? & 0xC) != 0x8 || (interface.read_word_32(0x400ACFF8)? & 0xC) != 0x8 {
    reset_vector = interface.read_word_32(0x1000_0004)?
} else {
    reset_vector = interface.read_word_32(0x0000_0004)?;
}
```
`Read AP0 CSW` is done in `ResetSystem` not `WaitForStopAfterReset`, then only going into `WaitForStopAfterReset` if `ap0_disable` for both s16 and s28.

I would like to at least get my s69 board running as intended to play with changes across the family, so knowing what probe firmware was used for development would be helpful.

Yatekii, 2022-01-05:
Pretty sure this was done with an onboard probe provided with the NXP devkit (MCU-LInk?). It is CMSIS-DAP based.

mriise, 2022-01-05:
I am using the onboard debugger for the LpcXpressoV2 with `LPC432x_IAP_CMSIS_DAP_NB_V5_361` firmware flashed to it using lpcscrypt. the Jlink firmware comes from NXP as well. the device enumerates as `NXP Semiconductors LPC-LINK2 CMSIS-DAP V5.361` with the CMSIS-DAP firmware flashed to it.

wt, 2022-01-06:
FWIW, the sequences are documented in user manuals for the various devices. I have the user manual for the s16 in front fof me and I see that it's in there. I am not sure if the implementation conforms or not.

mriise, 2022-01-06:
> However, I can confirm that my s16 is working with the tip of the tree right now after my changes landed.

Is that including debugging? if so, rtt or some other way?

Yatekii, 2022-04-18:
Hey, could you retry this with current master? :)

mriise, 2022-05-20:
Still the same as above with CMSIS, but I reflashed one of my boards with Jlink and tried out the `Jtag` protocol and got the below
```
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb

       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.26.11724
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: true
       DEBUG jaylink                          > libusb can detach kernel driver: true
       DEBUG jaylink                          > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                          > scanning 3 interfaces
       DEBUG jaylink                          > J-Link interface is #2
       DEBUG jaylink                          > legacy caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx
       DEBUG jaylink::capabilities            > unknown ext. capability bits: 0x2140013C0DB9FF7BBF truncated to 0x1B9FF7BBF (Reserved0 | GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite)
       DEBUG jaylink                          > extended caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 1366:0105 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 3 interfaces
       DEBUG jaylink                                > J-Link interface is #2
       DEBUG jaylink                                > legacy caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx
       DEBUG jaylink::capabilities                  > unknown ext. capability bits: 0x2140013C0DB9FF7BBF truncated to 0x1B9FF7BBF (Reserved0 | GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite)
       DEBUG jaylink                                > extended caps: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
       DEBUG probe_rs::probe::jlink                 > Supported speeds: SpeedInfo { base_freq: 48000000, min_div: 12 }
        INFO cargo_embed                            > Protocol speed 2000 kHz
       DEBUG probe_rs::config::registry             > Searching registry for chip with name LPC55S28JBD100
       DEBUG probe_rs::config::registry             > Exact match for chip name: LPC55S28JBD100
       DEBUG probe_rs::probe::jlink                 > Attaching to J-Link
       DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'JTAG'
        INFO probe_rs::probe::jlink                 > J-Link: S/N: 728552493
[... cut ...]
       DEBUG probe_rs::probe::jlink                               > Write IR: [8], len=4
       Error failed attaching to target
             
             Caused by:
                 Failed to get information for AP ApAddress { dp: Default, ap: 0 }
        INFO cargo_embed                                          > Metadata {
    chip: Some(
        "Unspecified(\"LPC55S28JBD100\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "2000",
    ),
    release: "0.12.0",
    commit: "bf588ae",
}
```
sadly still nothing flashed though :(

mriise, 2022-05-20:
As always a new cable changes things :^) 

*fresh boot = plugged in/replugging
board flashed with JLink firmware using swd protocol has some interesting behavior:
- most scenarios it will error with a timeout
- fresh boot connecting under reset option will infinite loop with NACK

```
       DEBUG probe_rs::probe::jlink::arm            > Performing line reset!
       DEBUG probe_rs::probe::jlink::arm            > Performing 1 transfers (0 additional transfers)
       DEBUG probe_rs::probe::jlink::arm            > Transfer result 0: Err(NoAcknowledge)
       DEBUG probe_rs::probe::jlink::arm            > DAP NACK
```
- fresh boot connecting without connecting under reset`will give (https://gist.github.com/mriise/28a09944a0f59f3904a21f6d83a001b9):
```
       !!! < THE REST IS IN THE GIST > !!!
       DEBUG probe_rs::probe::jlink::arm                          > Performing 1 transfers (0 additional transfers)
       DEBUG probe_rs::probe::jlink::arm                          > Transfer result 0: Ok(0)
        WARN probe_rs::session                                    > Could not clear all hardware breakpoints: ArchitectureSpecific(RegisterRead { address: 12, name: "DRW", source: ArchitectureSpecific(FaultResponse) })
     Erasing sectors ⠁ [00:00:00] [#######################################]       0B/      0B @       0B/s (eta 0s )
       Error failed to flash /home/mriise/Projects/blinky/target/thumbv8m.main-none-eabi/debug/blinky
             
             Caused by:
                 0: Error while flashing
                 1: Something during the interaction with the core went wrong
                 2: A core architecture specific error occurred
                 3: Failed to write register DRW at address 0x0000000c
                 4: An error specific to the selected architecture occurred
        INFO cargo_embed                                          > Metadata {
    chip: Some(
        "Unspecified(\"LPC55S28JBD100\")",
    ),
    probe: Some(
        "JLink",
    ),
    speed: Some(
        "2000",
    ),
    release: "0.12.0",
    commit: "bf588ae",
}
```