probe-rs/probe-rs #949: stm32h7: last word written to RAM may get "reverted" on reset

Dirbaio, 2022-01-05:
STM32H7's  ECC RAM behaves funny on reset [(source)](https://www.st.com/resource/en/application_note/dm00623136-error-correction-code-ecc-management-for-internal-memories-protection-on-stm32h7-series-stmicroelectronics.pdf):

> The ECC is computed on data word. If a data smaller than word is written in the volatile memory, the modification
is done on read-modify-write basis. On an incomplete access, the ECC does not write the value immediately. As it
may be the very next byte or half word, it waits for the next write access. This is a common case in applications
dealing with the backup SRAM. For example in an array of characters, energy is conserved. However, a write
operation will not be completed in case of reset (the memory contents will be retained without the last incomplete
word write). The workaround for this limitation is to write a dummy incomplete word write after each regular one. The dummy
write address must be within the same memory (backup SRAM in this case).

I just hit this in [teleprobe](https://github.com/embassy-rs/teleprobe/commit/e5da6f94084605c8c5b4db02cee231a013d0e424), where "flashing" an ELF to RAM and then resetting would corrupt the last written word.

I'm not sure how to best fix this in probe-rs, maybe making a sequence that does the dummy write on reset?

Yatekii, 2022-01-05:
Hmm a cudtom sequence would work I guess :) Maybe we can even find something according in the packs?

Tiwalun, 2022-01-05:
Maybe having a way to specify this as part of the target description? Some kind of attribute for the memories, which are already described there.