probe-rs/probe-rs #963: Support for STM32F303VCT6 (STM32F3DISCOVERY)

BrendanBall, 2022-01-16:
**Is your feature request related to a problem? Please describe.**
I've been trying to get into embedded with Rust and started following https://docs.rust-embedded.org/discovery which uses the STM32F3DISCOVERY board.
Unfortunately this chip isn't yet supported by probe-rs.

**Describe the solution you'd like**
Support for STM32F303VCT6

**Describe alternatives you've considered**
None

**Additional context**
I'm not sure exactly what's needed for support, but looking at `probe-rs/targets/STM32F3_Series.yaml` it shouldn't be difficult to add support if you know what you're doing, unfortunately I don't yet know what I'm doing :( .

adamgreig, 2022-01-16:
Thanks for the report! This chip is supported under the name "STM32F303VCTx" - the `x` indicates it supports both 6 and 7 in the part number, which refers to the operating temperature range (the names come from ST).

So, you should be able to set `chip = STM32F303VCTx` in `Embed.toml`, or `--chip STM32F303VCTx` with `cargo flash`.

You can use `cargo flash --list-chips` to get the full list, but I guess in this case it would have been helpful if it could have suggested `STM32F303VCTx` as an alternative for `STM32F303VCT6`, maybe that's something we could add.

BrendanBall, 2022-01-16:
Ah awesome. Yes I was following the docs on https://github.com/knurling-rs/probe-run and just did `probe-run --list-chips | grep STM32F303VCT6` which didn't return anything, and also got `Error: The requested chip 'STM32F303VCT6' was not found in the list of known targets` when specifying  `--chip STM32F303VCT6`. I definitely think there needs to be a way to specify the exact chip (or at least discover the correct setting based on the chip). It's already confusing enough with all the versions you need to specify everywhere (arch etc.).
BTW is it theoretically possible to discover what chip you have programmatically? Or can't you even communicate properly without knowing the chip first?

adamgreig, 2022-01-16:
Probe-rs does have functionality for automatically discovering the chip but it's not very comprehensive at the moment as it basically has to be specialised for every vendor - there's no standard way to detect the chip unfortunately. Still, it would be possible to do much better autodetection than we do currently.

BrendanBall, 2022-01-16:
Thanks for the quick responses :smile: . At least someone will find this issue now if they search STM32F303VCT6 or STM32F3DISCOVERY

closed 2022-01-16 by BrendanBall

adamgreig, 2022-01-16:
I've opened #964 so that in future it will select the right chip when you ask for `STM32F303VGT6` and similar, and drop a log message letting you know the canonical name is with the x. Thanks for the inspiration!

BrendanBall, 2022-01-16:
Awesome!