probe-rs/probe-rs #965: cmsisdap: Stackoverflow

jacobrosenthal, 2022-01-16:
Same issue as months ago from here https://github.com/probe-rs/probe-rs/issues/531 I was checking back into, but that issue wants to be for just jlink

Mainly im attempting to use vscode and probe-rs-debugger on embassy. A simple example like this works without embassy, but presumably async stuff is deep enough to trigger?. Uses an nrf52840-mdk which has a cmsis dap on board
https://github.com/jacobrosenthal/blinky-button-demo/tree/nrf52840-mdk

Trace logs here
https://gist.github.com/jacobrosenthal/921f5306e01e3ef83db36c46e3c157d1

noppej, 2022-01-16:
I will test this against #893 to make sure we catch this as well. Let's keep this Issue open for now until we know for sure.

noppej, 2022-01-19:
@jacobrosenthal Are you able to test this with the VSCode extension and `probe-rs-debugger` branches I just uploaded? (details in https://github.com/probe-rs/vscode/issues/26#issuecomment-1016870363)

jacobrosenthal, 2022-01-20:
Yeah it looks good!

jacobrosenthal, 2022-01-20:
So I guess this is a different stackoverflow bug. But I confirmed that [this](https://github.com/probe-rs/vscode/issues/26#issuecomment-1016870363) fixes embassy programs for both cmsis and jlink

noppej, 2022-08-11:
Closing as fixed.

closed 2022-08-11 by noppej