probe-rs/probe-rs #974: vscode debugger popup shows invalid path (does not exist)

darryln, 2022-01-20:
Moved here from [vscode] (https://github.com/probe-rs/vscode/issues/26)  per @noppej.

Getting this popup, but the path is invalid (does not exist):
[image]

See below objdump excerpts below. There is one compilation unit for asm/lib.rs. Its comp_dir is "./", but the code is substituting the full  base path to my crate, resulting in the bogus path.  The second entry shown below is for my own src/lib.rs, for which the comp_dir gives the correct full base path.  

Initially the popup seemed to coincide with setting a bp in my src/lib.rs, but the popup repeated after removing the bp, so may be unrelated to setting the bp.

```
 <1><b8201>: Abbrev Number: 0
  Compilation Unit @ offset 0xb8202:
   Length:        0xa09 (32-bit)
   Version:       4
   Abbrev Offset: 0xd59
   Pointer Size:  4
 <0><b820d>: Abbrev Number: 1 (DW_TAG_compile_unit)
    <b820e>   DW_AT_producer    : (indirect string, offset: 0xb9288): clang LLVM (rustc version 
1.59.0-nightly (c5ecc1570 2021-12-15))
    <b8212>   DW_AT_language    : 28    (Rust)
    <b8214>   DW_AT_name        : (indirect string, offset: 0xe1171): asm/lib.rs
    <b8218>   DW_AT_stmt_list   : 0x5599a
    <b821c>   DW_AT_comp_dir    : (indirect string, offset: 0xb075b): ./
    <b8220>   DW_AT_GNU_pubnames: 1
    <b8220>   DW_AT_low_pc      : 0x0
    <b8224>   DW_AT_ranges      : 0xdf80

```

```
  Compilation Unit @ offset 0xf3:
   Length:        0xf0 (32-bit)
   Version:       4
   Abbrev Offset: 0x87
   Pointer Size:  4
 <0><fe>: Abbrev Number: 1 (DW_TAG_compile_unit)
    <ff>   DW_AT_producer    : (indirect string, offset: 0x6a439): clang LLVM (rustc version 1.5
8.0 (02072b482 2022-01-11))
    <103>   DW_AT_language    : 28      (Rust)
    <105>   DW_AT_name        : (indirect string, offset: 0x9a1b9): src/lib.rs
    <109>   DW_AT_stmt_list   : 0xf6
    <10d>   DW_AT_comp_dir    : (indirect string, offset: 0x39311): /home/embsysdev/rust/embedded/hello-probe-rs
    <111>   DW_AT_GNU_pubnames: 1
    <111>   DW_AT_low_pc      : 0x0
    <115>   DW_AT_ranges      : 0x18
 <1><119>: Abbrev Number: 2 (DW_TAG_namespace)
    <11a>   DW_AT_name        : (indirect string, offset: 0x2969f): hello
 <2><11e>: Abbrev Number: 2 (DW_TAG_namespace)
    <11f>   DW_AT_name        : (indirect string, offset: 0x5a2f6): exit
 <3><123>: Abbrev Number: 3 (DW_TAG_variable)

```

noppej, 2022-01-21:
@darryln I am able to reproduce this. What is happening is that the debugger finds some RUST source statement that uses the `asm::` form to execute assembly code. If you put a breakpoint immediately before this code, it will find it and stop correctly. However, once the processor starts stepping into this code, the PC register points to a location inside the compile unit that has an incomplete directory name, such as the ` <b821c>   DW_AT_comp_dir    : (indirect string, offset: 0xb075b): ./` above.  

At this point we could probably direct the UI client back to the valid call-site of the `asm::` function, but that would be misleading, because the PC value wouldn't match. My opinion is that the best solution when this happens is to show the (as yet unimplemented) disassembly view of that code, so that the developer can step through the assembly code one instruction at a time. 

What do you think?

darryln, 2022-01-21:
I think I know nothing about this code, and I think your description sounds like a plan.

noppej, 2022-02-01:
Update: In addition to the condition described above, there are also situations where the path to `.asm` files are hardcoded and point to locations that are not accessible to the user. The problem and future solution is described [here](https://github.com/rust-embedded/cortex-m/issues/419)

noppej, 2022-02-03:
Closing this as there is already a tracking issue #788

closed 2022-02-03 by noppej