probe-rs/probe-rs #975: Debugger: Panic with "This is a bug. Please report it."

dylanede, 2022-01-20:
**Describe the bug**
When I enable flashing with the debugger, connecting to a Raspberry Pi Pico (RP2040) over CMSIS-DAP v2, the debugger panics after flashing. No problems occur if flashing is disabled (i.e. `--flashing-enabled` is not passed to `probe-rs-debugger`).

**To Reproduce**
Steps to reproduce the behavior:
Simple application based on https://github.com/rp-rs/rp-hal/blob/main/boards/rp-pico/examples/pico_pwm_blink.rs.
`cargo build`
`RUST_BACKTRACE=1 probe-rs-debugger debug --program-binary target/thumbv6m-none-eabi/debug/pico -c RP2040 --flashing-enabled`

**Expected behavior**
Not panic

**Stacktrace**
```text
Welcome to "probe-rs-debugger". Use the 'help' command for more
INFO: FLASHING: Starting write of "/home/dylan/repos/pico-comms-test/target/thumbv6m-none-eabi/debug/pico" to device memory
thread 'main' panicked at 'This is a bug. Please report it.', debugger/src/debugger.rs:1310:21
stack backtrace:
   0: rust_begin_unwind
             at /rustc/02072b482a8b5357f7fb5e5637444ae30e423c40/library/std/src/panicking.rs:498:5
   1: core::panicking::panic_fmt
             at /rustc/02072b482a8b5357f7fb5e5637444ae30e423c40/library/core/src/panicking.rs:107:14
   2: probe_rs_debugger::debugger::Debugger::debug_session
             at /home/dylan/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/d52ad28/debugger/src/debugger.rs:1310:21
   3: probe_rs_debugger::debugger::debug
             at /home/dylan/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/d52ad28/debugger/src/debugger.rs:1621:9
   4: probe_rs_debugger::main
             at /home/dylan/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/d52ad28/debugger/src/main.rs:164:14
   5: core::ops::function::FnOnce::call_once
             at /rustc/02072b482a8b5357f7fb5e5637444ae30e423c40/library/core/src/ops/function.rs:227:5
```

**Desktop (please complete the following information):**
 - WSL2 under Windows 10

**Additional context**
The debug adapter is an Adafruit Trinkey QT2040 running CMSIS-DAP firmware.
Flashing and debugging works fine using the Raspberry Pi Foundation's fork of OpenOCD.
`cargo embed` and `cargo flash` also succeed without issue.

noppej, 2022-01-21:
@dylanede I believe I've fixed this in a recent PR (still in review). If you want, you are welcome to test it with this unpublished code \( [instructions here](https://github.com/probe-rs/vscode/issues/26#issuecomment-1016870363) \), or alternatively I will let you know once that PR is merged into the master branch.

noppej, 2022-02-03:
@dylanede Can you please confirm that this is fixed with the latest (master) of probe-rs-debugger and VSCode extenstion 0.3.4 ?

dylanede, 2022-02-04:
This particular problem appears to be fixed, and I can use the debugger in the terminal (stepping, resetting, halting works), but now the (updated) VSCode extension complains with _"An error with the usage of the probe occurred"_. The only output in the debug console is
```text

Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
probe-rs-debugger CONSOLE: ....Closing session from  :127.0.0.1:38916
probe-rs-debugger CONSOLE: DAP Protocol server exiting
```
Shall I raise a separate issue on the extension repository for this?

Note that there is one bit of unusual behaviour with the debugger when I use it from the terminal - after flashing and entering the debug terminal, the CPU is halted at the start of RAM (0x20000000), which is unexpected. I have to `reset` to get the loaded program to run. Could this be related to the problem with the extension?

noppej, 2022-04-04:
@dylanede Can it be that your `launch.json` has both 'reset-after-flashing` and `halt-after-reset`? That would explain the halt you are seeing.

noppej, 2022-04-08:
Please feel free to re-open if this is still an issue for you.

closed 2022-04-08 by noppej