probe-rs/probe-rs #980: Overlapping Segments/Headers - Differing Behavior from rust-objcopy (llvm-objcopy)

david-boles, 2022-01-25:
**Describe the bug**
A seemingly valid linker scripts can produce program headers/segments in the ELF that overlap. rust-objcopy (llvm-objcopy) somehow handles this fine but probe-rs errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone [this repo](https://github.com/david-boles/pinetime-rust-mcuboot/tree/overlapping-segments).
2. Attempt to flash to a chip, e.g. `cargo flash --elf target/thumbv7em-none-eabihf/debug/firmware --chip nRF52832_xxAA`
3. Doing so results in an error:
    ```
    Flashing target/thumbv7em-none-eabihf/debug/firmware
       Error 0: Error while flashing
             1: Adding data for addresses 00008000..0000B668 overlaps previously added data for addresses 00008020..0000AE24.
    ```
4. Generate a binary without probe-rs: `rust-objcopy -O binary target/thumbv7em-none-eabihf/debug/firmware target/thumbv7em-none-eabihf/debug/firmware-objcopy.bin`
5. This succeeds and can be flashed, e.g. with OpenOCD.

**Expected behavior**
An image equivalent to that produced by `rust-objcopy -O binary` should be flashed to the target. 

**Desktop (please complete the following information):**
Linux, Debian sid

**Additional context**
This isn't a major problem and can be worked around but tiwalun on the matrix chat mentioned that probe-rs should probably copy objcopy's behavior. I believe the relevant probe-rs source is [here](https://github.com/probe-rs/probe-rs/blob/798a9015c5ec850a0e2bec7d4914258b6963c5c7/probe-rs/src/flashing/download.rs#L192).

Dirbaio, 2022-01-26:
A binary with overlapping sections is a broken binary. You should fix your linker script. 

objcopy probably doesn't check for overlapping sections, so the result is probably the 2nd overlapping section overwriting the 1st in the resulting bin.

david-boles, 2022-01-26:
I have fixed my scripts, I just opened this based on what tiwalun said. Happy to close if desired!

>tiwalun: I think copying the behaviour of objcopy would be the best solution for probe-rs for this case.

Yatekii, 2022-01-26:
I think it might be of use for some applications (nRF standard utilities that have a bootloader etc have the issue of overlap. And their mergehex utility can handle this). So it's def. not a good solution and you should ALWAYS fix your ELF, but it might be warranted to think about whether this should be a hard error or a warning. Personally I favor the hard error.

Dirbaio, 2022-01-26:
> nRF standard utilities that have a bootloader etc have the issue of overlap

which binaries? I've never seen that

Yatekii, 2022-06-06:
I think this is intended behavior :)

closed 2022-06-06 by Yatekii