probe-rs/probe-rs #993: JLink reset_target

k0d, 2022-02-02:
https://github.com/probe-rs/probe-rs/blob/bfa7cf6c00ecc47097e386efa30579db74fdd105/probe-rs/src/probe/jlink/mod.rs#L545

I need to do a hard reset of my target, but it's not implemented on JLink, what would you suggest to get this working? assert the reset pin for X milliseconds and then de-assert it? Should it be in this jlink/mod.rs file, or should we make target_reset_assert in probe.rs public?

Tiwalun, 2022-02-02:
If you just want to reset when attaching, you can use the `Probe::attach_under_reset` method.

The easiest way to implement this would be to use the [reset_target] method available on the J-Link, then the error can be removed.

[reset_target]: https://docs.rs/jaylink/latest/jaylink/struct.JayLink.html#method.reset_target

k0d, 2022-02-02:
Sounds good...I'll sort it and do a PR

PR #2333 (2024-04-04): Implement target_reset on J-Link
I'm not sure about changelog, but I'm also not sure if this is useful. `target_reset` seems to be unused in the binary, and maybe carrying it around may be somewhat redundant?

Closes #993
closed 2024-04-04 by Yatekii