probe-rs/probe-rs #994: Can't flash nRF9160 with HS probe

diondokter, 2022-02-03:
**Describe the bug**
I can't flash my nRF9160 using the HS Probe.
I've just updated the HS probe to the newest firmware, but no change.

It does work with the jlink of the Nordic DK's (tested with an nRF53 DK).
It works with both swd and jtag as the protocol.

**To Reproduce**
1. Have an embedded project (in this case https://github.com/tweedegolf/dis-bootloader)
2. Run `cargo flash --release --chip nRF9160_xxAA`

**Desktop (please complete the following information):**
 - Windows 11

**Additional context**
Cargo flash is based on the Jan-28 master version (with a small modification so it works nicely with the nrf53).
Problem is also with the most recent probe-run.

Trace: https://gist.github.com/diondokter/e8cc8c2ce88cb55ec7b685f14194eb8e
(This is not the full trace, because it's so much that terminal forgets the starting lines. But I think it's far enough back that all relevant data is there)

k0d, 2022-04-14:
I also have this issue, but with my own probe (based on rp2040). I can flash it with my jlink in cmsis-dap mode though.

So I did a trace, and then a diff of both the jlink and my probe. This is the result.

```diff --git a/foo.trace b/foo.trace
index 04c634c..38e9314 100644
--- a/foo.trace
+++ b/foo.trace
@@ -443,10 +443,10 @@
        DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
        DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
        TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 10, 20, 00, E0, 0F, 00]...
-       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, 7F, 00]...
+       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, FF, 00]...
        DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
        TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
-       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0x7ffffffe
+       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0xfffffffe
        DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002014 }
        DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002014
        DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
@@ -476,10 +476,10 @@
        DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
        DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
        TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 1C, 20, 00, E0, 0F, 00]...
-       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, DF, FF, FF, 00]...
+       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, BE, DF, FF, FF, 00]...
        DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
        TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
-       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0xffffdffe
+       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0xffffdfbe
        DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002020 }
        DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002020
        DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
@@ -537,4 +537,3 @@
        TRACE probe_rs::flashing::loader                           > Adding data at address 0x00010000 with size 530328 bytes
        TRACE probe_rs::flashing::loader                           > Adding data at address 0x000917b0 with size 4556 bytes
        TRACE probe_rs::flashing::loader                           > Adding data at address 0x00092980 with size 1176 bytes
-    Finished in 29.873s```

No idea what that means though, anyone else have some ideas?

Yatekii, 2022-04-14:
Interesting, thanks! If I understand it correct, the red trace is the one that worked?

k0d, 2022-04-14:
Actually, no...it was the red one :)

Yatekii, 2022-04-14:
But the red one says Finished is excluded, symbolizing to me that it did not work?

k0d, 2022-04-14:
```
       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0x10000080
       DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002008 }
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002008
       DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=AccessPort, addr=12)
       DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
       DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
       TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 08, 20, 00, E0, 0F, 00]...
       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, FF, 00]...
       DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
       TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0xfffffffe
       DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e000200c }
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000200c
       DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=AccessPort, addr=12)
       DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
       DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
       TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 0C, 20, 00, E0, 0F, 00]...
       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, 7F, 00]...
       DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
       TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0x7ffffffe
       DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002010 }
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002010
       DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=AccessPort, addr=12)
       DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
       DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
       TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 10, 20, 00, E0, 0F, 00]...
       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, 7F, 00]...
       DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
       TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0x7ffffffe
       DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002014 }
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002014
       DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=AccessPort, addr=12)
       DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
       DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 2 items
       TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 02, 05, 14, 20, 00, E0, 0F, 00]...
       TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 02, 01, FE, FF, FF, FF, 00]...
       DEBUG probe_rs::probe::cmsisdap                            > 2 of batch of 2 items suceeded
       TRACE probe_rs::probe::cmsisdap                            > Transfer status: ACK
       DEBUG probe_rs::architecture::arm::ap                      > Read register    DRW, value=0xfffffffe
       DEBUG probe_rs::architecture::arm::ap                      > Writing register TAR, value=TAR { address: e0002018 }
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002018
       DEBUG probe_rs::architecture::arm::ap                      > Reading register DRW
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=AccessPort, addr=12)
       DEBUG probe_rs::probe::cmsisdap                            > 2 items in batch
[... cut ...]
        INFO probe_rs::flashing::loader                           >     Multiple sections at 00092980 (1176 bytes)
       TRACE probe_rs::flashing::loader                           > Adding data at address 0x00010000 with size 530328 bytes
       TRACE probe_rs::flashing::loader                           > Adding data at address 0x000917b0 with size 4556 bytes
       TRACE probe_rs::flashing::loader                           > Adding data at address 0x00092980 with size 1176 bytes
    Finished in 29.873s
```

k0d, 2022-04-14:
The above is the trace that worked...with the first 2 differences hightlighted.

Yatekii, 2022-04-14:
Thanks!

Hmm it looks as if some bit went missing ... what implementation do you use on the rp2040? Curious because it's an issue with a) our CMSIS-DAP driver
b) the CMSIS-DAP spec
c) the hs-probe and whatever you use

cc: @korken89 @adamgreig

k0d, 2022-04-14:
https://github.com/majbthrd/DapperMime

k0d, 2022-04-15:
Trying to flash it with python-dap or node-dap fails as well, so I'm assuming it's either b) or c).

So probably not a fault in probe.rs I guess, however it could be good to figure out what the issue is anyway...as it might be something with hs-probe which I guess you'd want to fix.

Yatekii, 2022-04-15:
Yeah, what i think is weird about this is that just the read value is wrong. It does not actually write different data. But maybe if it reads bad bits it also writes bad bits without telling us.

Yatekii, 2022-04-15:
What is weird is that I would have guessed that this is the same issue: https://github.com/probe-rs/probe-rs/issues/1058 but it seems to be okay on the dapper-mime there which fails for you too, no?

k0d, 2022-04-15:
I'll check what my probe sends/receives...

also, that other issue looks interesting...I'll look into it and get back to you...

k0d, 2022-04-15:
```
RX 00 FF 00 00 00 00 00 00                           |  ........
TX 00 00 00 00 00 00 00 00                           |  ........
RX 00 FE 00 00 00 00 00 00                           |  ........
TX 00 02 40 00 00 00 00 00                           |  ..@.....
RX 00 F0 00 00 00 00 00 00                           |  ........
TX 00 01 01 00 00 00 00 00                           |  ........
RX 02 00 00 00 00 00 00 00                           |  ........
TX 00 02 11 01 00 00 00 00                           |  ........
RX 11 40 42 0F 00 00 00 00                           |  .@B.....
TX 02 01 11 01 00 00 00 00                           |  ........
RX 04 00 50 00 00 00 00 00                           |  ..P.....
TX 11 00 11 01 00 00 00 00                           |  ........
RX 13 00 00 00 00 00 00 00                           |  ........
TX 04 00 11 01 00 00 00 00                           |  ........
RX 01 00 01 00 00 00 00 00                           |  ........
TX 13 00 11 01 00 00 00 00                           |  ........
RX 12 33 FF FF FF FF FF FF                           |  .3......
TX 01 00 11 01 00 00 00 00                           |  ........
RX 12 10 9E E7 00 00 00 00                           |  ........
TX 12 00 11 01 00 00 00 00                           |  ........
RX 12 33 FF FF FF FF FF FF                           |  .3......
TX 12 00 11 01 00 00 00 00                           |  ........
RX 12 03 00 00 00 00 00 00                           |  ........
TX 12 00 11 01 00 00 00 00                           |  ........
RX 05 00 01 02 00 00 00 00                           |  ........
TX 12 00 11 01 00 00 00 00                           |  ........
RX 05 00 03 00 1E 00 00 00                           |  ........
TX 05 01 01 77 24 A0 6B 00                           |  ...w$.k.
RX 05 00 01 06 00 00 00 00                           |  ........
TX 05 03 01 00 0F 00 F0 00                           |  ........
RX 05 00 03 04 00 0F 00 F0                           |  ........
TX 05 01 01 00 0F 00 F0 00                           |  ........
RX 05 00 02 08 F0 00 00 01                           |  ........
TX 05 03 01 01 00 77 84 00                           |  .....w..
RX 05 00 02 08 F0 00 00 02                           |  ........
TX 05 02 01 11 00 77 24 00                           |  .....w$.
RX 05 00 02 08 F0 00 00 03                           |  ........
TX 05 02 01 00 00 88 12 00                           |  ........
RX 05 00 02 08 F0 00 00 04                           |  ........
TX 05 02 01 02 00 77 54 00                           |  .....wT.
RX 05 00 02 08 F0 00 00 05                           |  ........
TX 05 02 01 00 00 88 12 00                           |  ........
RX 05 00 02 08 F0 00 00 06                           |  ........
TX 05 02 01 00 00 88 12 00                           |  ........
RX 05 00 02 08 F0 00 00 07                           |  ........
TX 05 02 01 A1 00 88 12 00                           |  ........
RX 05 00 02 08 F0 00 00 00                           |  ........
TX 05 02 01 00 00 00 00 00                           |  ........
RX 05 00 01 0B 00 00 00 00                           |  ........
TX 05 02 01 01 00 77 84 00                           |  .....w..
RX 05 00 01 03 00 00 00 00                           |  ........
TX 05 01 01 03 F0 0F E0 00                           |  ........
RX 05 00 02 08 00 00 00 00                           |  ........
TX 05 01 01 00 00 00 00 00                           |  ........
RX 05 00 02 01 10 00 00 E3                           |  ........
TX 05 02 01 52 00 80 03 00                           |  ...R....
RX 05 00 03 01 52 00 80 03                           |  ....R...
TX 05 02 01 50 00 80 43 00                           |  ...P..C.
RX 05 00 01 0B 00 00 00 00                           |  ........
TX 05 03 01 11 00 77 24 00                           |  .....w$.
RX 05 00 01 03 00 00 00 00                           |  ........
TX 05 01 01 03 F0 0F E0 00                           |  ........
RX 05 00 02 08 00 00 00 01                           |  ........
TX 05 01 01 00 00 00 00 00                           |  ........
RX 05 00 02 01 10 00 00 E3                           |  ........
TX 05 02 01 00 00 00 23 00                           |  ......#.
RX 05 00 03 01 00 00 00 23                           |  .......#
TX 05 02 01 10 00 00 23 00                           |  ......#.
RX 05 00 02 08 F0 00 00 03                           |  ........
[... 213 lines cut ...]
RX 05 00 06 05 F8 ED 00 E0                           |  ........
TX 05 02 01 00 00 00 00 00                           |  ........
RX 05 00 06 05 F8 ED 00 E0                           |  ........
TX 05 06 01 03 00 13 00 00                           |  ........
RX 05 00 06 05 F8 ED 00 E0                           |  ........
TX 05 06 01 03 00 13 00 00                           |  ........
RX 05 00 02 05 F0 ED 00 E0                           |  ........
TX 05 06 01 03 00 13 00 00                           |  ........
RX 05 00 02 05 F0 ED 00 E0                           |  ........
TX 05 02 01 03 00 13 00 00                           |  ........
RX 05 00 04 05 F4 ED 00 E0                           |  ........
TX 05 02 01 0F 00 13 05 00                           |  ........
RX 05 00 02 05 F8 ED 00 E0                           |  ........
TX 05 04 01 0F 00 13 00 00                           |  ........
RX 05 00 02 05 F0 ED 00 E0                           |  ........
[... cut ...]
RX 05 00 02 05 F0 ED 00 E0                           |  ........
TX 05 02 01 D6 00 00 20 00                           |  ...... .
RX 05 00 03 01 12 00 00 23                           |  .......#
TX 05 02 02 D6 00 00 20 00                           |  ...... .
RX 03 00 00 00 00 00 00 00                           |  ........
TX 05 01 02 D6 00 00 20 00                           |  ...... .
RX 01 00 00 00 00 00 00 00                           |  ........
TX 03 00 02 D6 00 00 20 00                           |  ...... .
```

k0d, 2022-04-15:
Not sure if the above log is helpful...but it's what my dapper-mime is RX/TXing.

I'm going to try to cross reference it to the trace I did.

k0d, 2022-04-15:
Some new info:
```
probe-rs ❯ cargo run --bin probe-rs-cli run ./build/zephyr/zephyr.elf --chip nRF9160_xxAA 
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target/debug/probe-rs-cli run ./build/zephyr/zephyr.elf --chip nRF9160_xxAA`
     Erasing sectors ✔ [00:00:00] [------------------------------------------------------------------------]       0B/524.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [------------------------------------------------------------------------]       0B/524.00KiB @       0B/s (eta 0s )
Error: The flashing procedure failed for './build/zephyr/zephyr.elf'.

Caused by:
    0: Failed to erase flash sector at address 0x00010000.
    1: Failed to erase flash sector at address 0x00010000.
    2: A core architecture specific error occurred
    3: Failed to read register DRW at address 0x0000000c
    4: An error specific to the selected architecture occurred
    5: Target device responded with a WAIT response to the request.
```

It seems to want to write to the flash sector at address 0x00010000, but I thought the flash started at 0x0 on the nrf9160. The ELF does indeed say 0x00010000, so I'm quite confused by this.
It seems to erase sector 0x00010000 with jlink just fine though...so that's even more confusing!

Yatekii, 2022-04-15:
Very odd. I mean it does not have flash at 0x00010000 so even if JLink would just wave the erase it would fail programming hmm. Does it also have stuff at 0x0 or only 0x00010000? Also, does it always fail like this?

k0d, 2022-04-15:
JLINK (running as CMSIS-DAP)
```
> Adding command to batch: Read(port=AccessPort, addr=12)
> 2 items in batch
> Attempting batch of 2 items
> 2 of batch of 2 items suceeded
> Read register    DRW, value=0x5100001
> Writing register TAR, value=TAR { address: e000edf0 }
> Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf0
> Reading register DRW
> Adding command to batch: Read(port=AccessPort, addr=12)
> 2 items in batch
> Attempting batch of 2 items
> 2 of batch of 2 items suceeded
> Read register    DRW, value=0x1130003
> Writing register TAR, value=TAR { address: e000edf4 }
> Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf4
> Writing register DRW, value=DRW { data: 0 }
> Adding command to batch: Write(port=AccessPort, addr=12, data=0x00000000
> Writing register TAR, value=TAR { address: e000edf0 }
> Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf0
> Reading register DRW
> Adding command to batch: Read(port=AccessPort, addr=12)
> 4 items in batch
> Attempting batch of 4 items
> 4 of batch of 4 items suceeded
> Read register    DRW, value=0x130003
> Writing register TAR, value=TAR { address: e000edf8 }
> Adding command to batch: Write(port=AccessPort, addr=4, data=0xe000edf8
> Reading register DRW
> Adding command to batch: Read(port=AccessPort, addr=12)
> 2 items in batch
> Attempting batch of 2 items
> 2 of batch of 2 items suceeded
> Read register    DRW, value=0x0
```

DapperMime
```
> Adding command to batch: Read(port=AccessPort, addr=12)
> 2 items in batch
> Attempting batch of 2 items
> 2 of batch of 2 items suceeded
> Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
> Adding command to batch: Write(port=AccessPort, addr=0, data=0x23000012
> Writing register TAR, value=TAR { address: e0002000 }
> Adding command to batch: Write(port=AccessPort, addr=4, data=0xe0002000
> Reading register DRW
> Adding command to batch: Read(port=AccessPort, addr=12)
> 3 items in batch
> Attempting batch of 3 items
> 1 of batch of 3 items suceeded
> Could not clear all hardware breakpoints: ArchitectureSpecific(RegisterRead { address: 12, name: "DRW", source: ArchitectureSpecific(WaitResponse) })
> Detaching from CMSIS-DAP probe
```

Yatekii, 2022-04-15:
Oh, de BP issue. Same as with the atsams then. those have the same issue. I tried debugging this a while ago. Disabling the call do clear all breakpoints makes it just fail later on tho.

Yatekii, 2022-04-15:
Also, those two traces are VERY different. How does this happen lol.

k0d, 2022-04-15:
https://github.com/probe-rs/probe-rs/blob/7e6b23c6bc861d1f29c9272197b6de2047d05982/probe-rs/src/session.rs#L511

This runs when dropping the session, so I guess the error happens before that? And then when it fails, we drop the session?

k0d, 2022-04-15:
https://github.com/probe-rs/probe-rs/blob/7e6b23c6bc861d1f29c9272197b6de2047d05982/probe-rs/src/architecture/arm/sequences/nrf53.rs << what is this? could we need something like this for nrf9160 also?

k0d, 2022-04-15:
https://gist.github.com/k0d/da5bef2a386ebdc1b28b207cd5631bd5 << full diff of the traces

Yatekii, 2022-04-15:
That's the debug sequence to unlock the nrf53. And yes, that could be required for the nrf91

k0d, 2022-04-17:
I've been trying lots of things, nothing fixed it yet. However we're receiving a WAIT response and then quit after that. I just wonder if the JLINK handles that internally itself better than we do. I'm going to hook up the debug probes to a logic analyzer and capture what's actually sent between the probe and the device.

k0d, 2022-04-17:
[CMSIS.csv](https://github.com/probe-rs/probe-rs/files/8501421/CMSIS.csv)
[JLink.csv](https://github.com/probe-rs/probe-rs/files/8501422/JLink.csv)

I've just captured the data, haven't analyzed it myself yet, but I thought I'd post it so you can take a look if/when you have time.

k0d, 2022-04-17:
I've fixed it!!!

Turns out that we don't wait enough times when we get a WAIT response. It's set to 80 in probe.rs, the J-Link ignores that we set that and waits a lot longer.

So in the firmware for my probe (based on dapper-mime), I've set it to wait the maximum 0xFFFF and it now programs the device just fine. Not sure what the best way to do a proper fix is? I guess we should really set it higher in probe.rs as the setting in "TransferConfigure" is there for a reason and we should respect that in the probe.

Yatekii, 2022-04-17:
Awesome job!

Yeah, I am not sure we need 0xFFFF (maybe 0xFF for example would be enough), but we can try for now and if we see that it breaks things we can adjust.
Do you make a PR? :)

k0d, 2022-04-17:
It took around ~650 waits in the JLink...so I think 0xFF isn't enough.

Maybe we set it as 0xFFFF in probe.rs as a default and then have it as a setting the user can change? My guess is that it's ok to have the maximum...it did the ~650 waits in some small amount of milliseconds.

I've set it as 0xFFFF in my probe firmware as I sometimes use pyocd with it and I don't want to have to fix every software...

Yatekii, 2022-04-17:
Let's have it 0xFFFF in probe-rs for now. No need for a setting.

commit 40103e93 (2022-04-18): Increase SWD wait timeout

We currently send to the probe that it should retry when it receives a WAIT response for 80 times. But this doesn't seem to be enough, so I've increased it to the maximum of 0xFFFF.

This should fix #994 and possibly #1058. Both issues should be checked by the original issue reporter.
commit 4108f930 (2022-04-18): Merge #1081

1081: Increase SWD wait timeout r=Yatekii a=k0d

We currently send to the probe that it should retry when it receives a WAIT response for 80 times. But this doesn't seem to be enough, so I've increased it to the maximum of 0xFFFF.

This should fix #994 and possibly #1058. Both issues should be checked by the original issue reporter.
PR #1081 (2022-04-18): Increase SWD wait timeout
We currently send to the probe that it should retry when it receives a WAIT response for 80 times. But this doesn't seem to be enough, so I've increased it to the maximum of 0xFFFF.

This should fix #994 and possibly #1058. Both issues should be checked by the original issue reporter.
closed 2022-04-18 by bors[bot]