probe-rs/probe-rs #995: Suppor for WCH-Link - CMSIS-DAP V2 probe

alexxy, 2022-02-03:
Hi!

There are CMSIS-DAP v2 adapters from China that works with openocd

```
[40975.589210] usb 1-3: USB disconnect, device number 12
[40984.609307] usb 1-3: new full-speed USB device number 13 using xhci_hcd
[40984.754909] usb 1-3: New USB device found, idVendor=1a86, idProduct=8011, bcdDevice= 1.05
[40984.754915] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40984.754917] usb 1-3: Product: WCH-Link
[40984.754918] usb 1-3: Manufacturer: wch.cn
[40984.754920] usb 1-3: SerialNumber: 0001A0000001
[40984.757874] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
[40984.759750] hid-generic 0003:1A86:8011.0004: hiddev96,hidraw0: USB HID v1.00 Device [wch.cn WCH-Link] on usb-0000:00:14.0-3/input2
```

log from openocd

```
info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Warn : gdb services need one or more targets defined
```

probe-run/probe-rs doesnt find it since it detects cmsis-dap adapters by  checking for CMSIS-DAP in product string

adamgreig, 2022-02-03:
Aw, the _one thing_ that makes a cmsis-dap probe cmsis-dap is that it has to say "CMSIS-DAP" in the product string!! It's not like you even have to get tested or approved or anything to do it... why.....

Do you know if openocd has to do anything with these besides talk regular cmsis-dap to them? Could you test just updating our cmsis-dap detection to include their product strings, just to check that works?

alexxy, 2022-02-03:
Hi!

I tryed updating product detection like this

```
diff --git a/probe-rs/src/probe/cmsisdap/tools.rs b/probe-rs/src/probe/cmsisdap/tools.rs
index fd6c2c75..d161bec7 100644
--- a/probe-rs/src/probe/cmsisdap/tools.rs
+++ b/probe-rs/src/probe/cmsisdap/tools.rs
@@ -62,7 +62,7 @@ fn get_cmsisdap_info(device: &Device<rusb::Context>) -> Option<DebugProbeInfo> {
         .ok();
 
     // All CMSIS-DAP probes must have "CMSIS-DAP" in their product string.
-    if prod_str.contains("CMSIS-DAP") {
+    if prod_str.contains("CMSIS-DAP") || prod_str.contains("WCH-Link") {
         let config_descriptor = device.active_config_descriptor().ok()?;
 
         log::trace!(
@@ -124,7 +124,7 @@ fn get_cmsisdap_info(device: &Device<rusb::Context>) -> Option<DebugProbeInfo> {
 /// Checks if a given HID device is a CMSIS-DAP v1 probe, returning Some(DebugProbeInfo) if so.
 fn get_cmsisdap_hid_info(device: &hidapi::DeviceInfo) -> Option<DebugProbeInfo> {
     if let Some(prod_str) = device.product_string() {
-        if prod_str.contains("CMSIS-DAP") {
+        if prod_str.contains("CMSIS-DAP") || prod_str.contains("WCH-Link") {
             log::trace!("CMSIS-DAP device with USB path: {:?}", device.path());
             log::trace!("                product_string: {:?}", prod_str);
             log::trace!(

```

After that I can detect probe using 

```
$ ./probe-rs-cli list
The following devices were found:
[0]: WCH-Link (VID: 1a86, PID: 8011, Serial: 0001A0000001, CmsisDap)
```

but info doesnt work =\

```
$ ./probe-rs-cli info
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.

```

lsusb info for the probe
```
$ lsusb -v -d 1a86:8011  

Bus 001 Device 015: ID 1a86:8011 QinHeng Electronics WCH-Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x8011 
  bcdDevice            1.05
  iManufacturer           1 wch.cn
  iProduct                2 WCH-Link
  iSerial                 3 0001A0000001
  bNumConfigurations      1
[... 122 lines cut ...]

BTW: working config for openocd

```
source [ find  interface/cmsis-dap.cfg ]
adapter speed 500
cmsis_dap_vid_pid 0x1a86 0x8011
transport select swd
```

so it looks like 

```
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz

```

```

alexxy, 2022-02-03:
Seems there were similar issue with https://github.com/pyocd/pyOCD/issues/1121

adamgreig, 2022-02-08:
Interestingly it looks like using the path provided by hidapi let them check if the interface says "CMSIS-DAP" instead of the product string - we could probably do the same when searching for HID devices, as well as checking interface names for composite devices that don't have it in the product string when using rusb. It'l require some tweaks to the probe searching code but seems like it would be possible and work with the WCH-Link.

My guess is your tweak let it detect it when scanning for probes but it then refused to open it, but running with RUST_LOG=trace might give more details. Still, it looks like we might not need to search for "WCH-Link" at all if we can check the interface strings too.

burrbull, 2023-01-14:
cc @adamgreig 
I've updated on 0.14, but it still can't flash: 
```
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       TRACE probe_rs::probe::cmsisdap::tools >   Interface 0: WCH CMSIS-DAP
       TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 1, skipping
       TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 2, skipping
       TRACE probe_rs::probe::cmsisdap::tools > WCH-Link: CMSIS-DAP device with 3 interfaces
       TRACE probe_rs::probe::cmsisdap::tools > No HID interface for CMSIS-DAP found.
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.25.11696
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: true
       DEBUG jaylink                          > libusb can detach kernel driver: true
       TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching 1a86:8011:0001A0000001
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 001: ID 1d6b:0003
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 002: ID 322e:202c
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 005: ID 1a86:8011
       TRACE probe_rs::probe::cmsisdap::tools >   Interface 0: WCH CMSIS-DAP
       TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 1, skipping
       TRACE probe_rs::probe::cmsisdap::tools >   Interface 2: WCH CMSIS-DAP
       TRACE probe_rs::probe::cmsisdap::tools >     HID interface found
       TRACE probe_rs::probe::cmsisdap::tools > WCH-Link: CMSIS-DAP device with 3 interfaces
       TRACE probe_rs::probe::cmsisdap::tools > Will use interface number 2 for CMSIS-DAPv1
       DEBUG probe_rs::probe::cmsisdap::tools > Could not open 1a86:8011 in CMSIS-DAP v2 mode
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 003: ID 13d3:3563
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 001: ID 1d6b:0002
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 001: ID 1d6b:0003
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 001: ID 1d6b:0002
       TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 1a86:8011 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG probe_rs::probe::espusbjtag::protocol  > Acquired libusb context.
       Error Probe could not be created
             
             Caused by:
                 Probe was not found.
        INFO cargo_embed                            > Metadata {
    chip: Some(
        "Unspecified(\"STM32F411CEUx\")",
    ),
    probe: Some(
        "CmsisDap",
    ),
    speed: None,
    release: "0.14.0",
    commit: "crates.io",
}
```

Although `probe-rs-cli` detects probe:
 ```
❯ probe-rs-cli list
The following devices were found:
[0]: WCH-Link (VID: 1a86, PID: 8011, Serial: 0001A0000001, CmsisDap)
```

adamgreig, 2023-01-14:
In the end we merged #1142 to get it in 0.14 for orbtrace and similar probe support, but it still doesn't work with wchlink. Though it seems like we might be closer to getting it to work now that it's detected - I've finally got one so I will try and get it working in a separate PR.

commit 1e8fc057 (2023-01-13): Merge #1142

1142: CMSIS-DAP: Check for CMSIS-DAP string in interface strings too r=Yatekii a=adamgreig

It turns out it's legitimate for CMSIS-DAP devices to just put that in an interface string, even though most put it in the product string too (as required for non-composite devices). This PR tries to check interface strings too, though this requires rusb and permission to access the device. Apparently on some platforms hidapi will report the interface string in the path, but not on mine, so it's included here as a talisman for good luck...

It's a bit hard to read the diff because mostly it [... cut]
PR #1142 (2023-01-13): CMSIS-DAP: Check for CMSIS-DAP string in interface strings too
It turns out it's legitimate for CMSIS-DAP devices to just put that in an interface string, even though most put it in the product string too (as required for non-composite devices). This PR tries to check interface strings too, though this requires rusb and permission to access the device. Apparently on some platforms hidapi will report the interface string in the path, but not on mine, so it's included here as a talisman for good luck...

It's a bit hard to read the diff because mostly it changes indentation on existing code, but [... cut]