probe-rs/probe-rs #997: jlink: Unhelpful error message when multidrop is not supported

Dirbaio, 2022-02-05:
jlink does not support multidrop, so this is not expected to work. 

However, the error message is not helpful at all. This message should show, but for whatever reason it doesn't: https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/jlink/swd.rs#L724

```
[dirbaio@mars ~]$ probe-rs-cli erase --chip rp2040
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

korken89, 2022-02-05:
This seems to not be a JLink specific bug.
I get the same error using a DAP42 and HS probe when trying to connect to the RP2040.

Yatekii, 2022-02-05:
Does the cli unwrap errors properly? I am not sure.

korken89, 2022-03-10:
@taunusflieger I've got it to work with the HS probe at least, still no luck with the DAP42 though.

Dirbaio, 2022-03-10:
For jlink it's indeed a misleading error message, since multidrop is completely unimplemented. It's not supposed to work, it should say so to the user.

Cmsis-dap probes like DAP42 and HS-PRobe not working is a different issue, these should definitely work. Could you open a separate issue @korken89 @taunusflieger  attaching trace-level logs? Thank you!

adamgreig, 2022-07-09:
This came up again in #1157. The issue is that the jlink driver error about not supporting multidrop:

https://github.com/probe-rs/probe-rs/blob/5a7f872aa4dde49db6a9df92f606e6db0ff49388/probe-rs/src/probe/jlink/arm.rs#L933-L935
is being masked by this `map_err` higher up:

https://github.com/probe-rs/probe-rs/blob/5a7f872aa4dde49db6a9df92f606e6db0ff49388/probe-rs/src/architecture/arm/communication_interface.rs#L404-L409

Yatekii, 2022-07-09:
Ah yes, this should be a .context :)

PR #2079 (2024-01-22): J-Link multidrop support
Closes #1272
Closes #1157 
Closes #997

The original PR was pretty close, but missed a few tricky parts:
 - A line reset in `raw_read_register` could immediately deselect the target. This was the root cause of incorrect DPIDR values.
 - IHI0074D contains the following line: `If overrun detection is enabled, then the line reset sets CTRL/STAT.STICKYORUN to 0b1.`. This means we need to clear the sticky overflow bit after target selection, because the selection process starts with a line reset.

Note that I only tested that probe-rs can reset a Raspberry Pico W, I haven't [... cut]
closed 2024-01-22 by Dirbaio