probe-rs/probe-rs #1005: [Feature Request] Update STM32H7 yaml

jg2562, 2022-02-13:
**Is your feature request related to a problem? Please describe.**
The current STM32H7 yaml does not contain several newer targets. I am currently using a microcontroller target that is only in the newer yaml, so I cannot upload to the newer board.

**Describe the solution you'd like**
The current STM32H7 yaml be replaced by the newer version.

**Describe alternatives you've considered**
I tried creating a PR by GitHub rejected the push due to the STM32H7 yaml being too large and recommended using LFS. However, it doesnt seem like you guys this, so I wasnt sure where to proceed. 

**Additional context**
If I can update my PR to help out, please let me know!

adamgreig, 2022-02-13:
Which STM32H7 are you using, and how did you try to update the YAML? I thought GitHub only blocked files >100MB which seems _really_ large for a target YAML file, so maybe something went wrong in generation or you accidentally added some other files?

jg2562, 2022-02-13:
Im using the STM32H723ZGTx. I followed the instructions [here](https://github.com/probe-rs/probe-rs#adding-targets) and I ended up with a single file that is 114Mb, so something definitely could have gone wrong in generation. It looks like the majority of it is the instructions from flash algorithms.

Yatekii, 2022-02-15:
You could try with https://github.com/probe-rs/probe-rs/pull/1010 :)
in release it takes 8s and generates a 900kb file. I am not sure it's correct tho. Lots of changes.
Also, it seems like the right cores are missing from the algos. But I am not sure we ever use that.

jg2562, 2022-02-16:
I was able to upload a program to my dev kit, and it ran correctly, so I assuming it's correct enough? Since its a smaller size, I will open a PR now.

jg2562, 2022-02-16:
Solved by #1011

closed 2022-02-16 by jg2562