probe-rs/probe-rs #1012: nRF5340 fails to flash under multiple circumstances

sphw, 2022-02-16:
**Describe the bug**
I am experiencing a consistent error while flashing the nRF5340. 

## Failed to write register DRW at address
```
       DEBUG cargo_flash::diagnostics                             > 0: A core architecture specific error occurred
       DEBUG cargo_flash::diagnostics                             > 1: Failed to write register DRW at address 0x0000000c
       DEBUG cargo_flash::diagnostics                             > 2: An error specific to the selected architecture occurred
       DEBUG cargo_flash::diagnostics                             > 3: Target device responded with FAULT response to request.
       Error Connecting to the chip was unsuccessful.
```

This error appears to happen only after my bootloader has been flashed. I am not sure why this would be the case, but in the instance where the device does not have a bootloader (i.e a valid vector table at 0x0) everything appears to be fine. I can flash multiple times with no errors. Once the bootloader is flashed (even if an application is not), `probe-rs` refuses to flash the device again. 

I also on occasion get an error with the message: `Failed to get information for AP ApAddress`, but I don't have clear reproduction steps. 

I would be happy to help solve these issues if pointed in the correct direction. I am new to the dark-magic of flashing sequences, and so lack the context to understand what could cause these errors. 

**Expected behavior**
I expect probe-rs to be able to recover from the common failure cases like `nrfjprog` can

**Desktop (please complete the following information):**
 - macOS Monterey - M1 MBP

Yatekii, 2022-02-16:
Hi :)

Can you please paste the full command/code (always include it in general, because without it it's just guesses in the dark (well it might be with it as well :D)) you are running to receive this log?

sphw, 2022-02-16:
>  Can you please paste the full command/code (always include it in general, because without it it's just guesses in the dark (well it might be with it as well :D)) you are running to receive this log?

Oh yup of course (that's what I get for filing bugs late at night). I am using `cargo flash` and the following command: `cargo flash --chip nRF5340_xxAA --log trace --target thumbv8m.main-none-eabihf` 

Here is a link to the bootloader I am flashing: https://github.com/m10io/frumsceaft/blob/main/examples/secure-blinky/src/main.rs

I have attached the full log below:
```
       DEBUG cargo_flash > Changed working directory to .
       DEBUG probe_rs_cli_util > Running '/Users/sphw/.rustup/toolchains/nightly-2021-12-15-aarch64-apple-darwin/bin/cargo' in directory /Users/sphw/code/mono/embedded/m10-spm
[0m[1m[33mwarning[0m[0m[1m: unused import: `rtt_init_print`[0m
[0m [0m[0m[1m[38;5;12m--> [0m[0msrc/main.rs:6:28[0m
[0m  [0m[0m[1m[38;5;12m|[0m
[0m[1m[38;5;12m6[0m[0m [0m[0m[1m[38;5;12m| [0m[0muse rtt_target::{rprintln, rtt_init_print};[0m
[0m  [0m[0m[1m[38;5;12m| [0m[0m                           [0m[0m[1m[33m^^^^^^^^^^^^^^[0m
[0m  [0m[0m[1m[38;5;12m|[0m
[0m  [0m[0m[1m[38;5;12m= [0m[0m[1mnote[0m[0m: `#[warn(unused_imports)]` on by default[0m

[0m[1m[33mwarning[0m[0m[1m: 1 warning emitted[0m

    Flashing /Users/sphw/code/mono/embedded/m10-spm/target/thumbv8m.main-none-eabihf/debug/m10-spm
       DEBUG probe_rs::config::registry > Searching registry for chip with name nRF5340_xxAA
       DEBUG probe_rs::config::registry > Exact match for chip name: nRF5340_xxAA
        WARN probe_rs::config::target   > Using custom sequence for nRF5340
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.24.11584
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: false
       DEBUG jaylink                          > libusb has hotplug support: true
       DEBUG jaylink                          > libusb can detach kernel driver: false
       DEBUG jaylink                          > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x1051,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                          > scanning 6 interfaces
       TRACE jaylink                          > active configuration descriptor: ConfigDescriptor {
    bLength: 0x9,
    bDescriptorType: 0x2,
    wTotalLength: 0xbb,
    bNumInterfaces: 0x6,
    bConfigurationValue: 0x1,
    iConfiguration: 0x4,
    bmAttributes: 0x80,
    bMaxPower: 0x32,
    extra: Some(
        [
            0x8,
            0xb,
            0x0,
            0x2,
            0x2,
            0x2,
            0x0,
            0x5,
        ],
    ),
[... 694 lines cut ...]
}
       TRACE probe_rs::probe::jlink::swd                          > Adding 2 idle cycles after transfer!
       DEBUG probe_rs::probe::jlink::swd                          > Performing 1 transfers (0 additional transfers)
       TRACE jaylink                                              > write 20 bytes: [cf, 0, 3a, 0, ff, 83, ff, ff, ff, ff, ff, 3, 4, 2, a, 0, 0, 0, 0, 0]
       TRACE jaylink                                              > read 9 bytes: [0, 64, 0, 0, 0, 0, 0, 0, 0]
       DEBUG probe_rs::probe::jlink::swd                          > Transfer result 0: Ok(0)
        INFO cargo_flash                                          > Metadata {
    chip: Some(
        "\"nRF5340_xxAA\"",
    ),
    probe: Some(
        "\"J-Link\"",
    ),
    speed: Some(
        "0",
    ),
    release: "0.12.1",
    commit: "5f87526",
}
       DEBUG cargo_flash::diagnostics                             > 0: A core architecture specific error occurred
       DEBUG cargo_flash::diagnostics                             > 1: Failed to write register DRW at address 0x0000000c
       DEBUG cargo_flash::diagnostics                             > 2: An error specific to the selected architecture occurred
       DEBUG cargo_flash::diagnostics                             > 3: Target device responded with FAULT response to request.

```

diondokter, 2022-07-12:
This may be similar to what I'm experiencing on the nRF9160.
Trace: https://gist.github.com/diondokter/0000ddbc112d67b68fe730bb1b90f315 

In my case, the chip has been fully erased and nothing is on it.
I have seen this before and when the chip has our (different) bootloader flashed on it, it works just fine.
This also seems similar to #1162 

My guess is that an erased nrf91 can fault into this state, that the nrf53 of the OP faults into this state when his bootloader jumps into empty flash code.

diondokter, 2022-09-02:
Oh, I've now found out this is actually a regression from 0.12

I used an older probe-run that used 0.12 and the flashing went fine on my nrf9160, but a newer cargo-flash didn't work and now that I've updated probe-run to a version that uses 0.13, it now can't flash the mcu when it is in a crashed state.

(Specifically, I know the current code is hardfaulting by executing a function call at a place where no function is)