probe-rs/probe-rs #1045: Erasing a flash page results in "Error communicating with probe" on stm31f103 "blue pill"

barafael, 2022-03-06:
**Describe the bug**
When flashing a program which calls `stm32f1xx_hal::flash::FlashWriter::erase`, the probe flashes the binary correctly, but RTT communication never happens. On terminal, I see:

```
(HOST) INFO  flashing program (16 pages / 16.00 KiB)
(HOST) INFO  success!
Error: Error communicating with probe: An error with the usage of the probe occured

Caused by:
    0: An error with the usage of the probe occured
    1: An error specific to a probe type occured
    2: Command failed with status SwdApWait
```

**To Reproduce**
Steps to reproduce the behavior:
1. Here's the example:

```rust
//! Write a value to flash.

#![deny(unsafe_code)]
#![no_std]
#![no_main]

use nb::block;

use panic_rtt_target as _;

use cortex_m_rt::entry;
use rtt_target::{rprint, rprintln, rtt_init_print};
use stm32f1xx_hal::flash::FlashWriter;
use stm32f1xx_hal::{pac, prelude::*, timer::Timer};

use stm32f1xx_hal as _;

#[entry]
fn main() -> ! {
    // Get access to the core peripherals from the cortex-m crate
    let cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    rtt_init_print!();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let rcc = dp.RCC.constrain();

    // Freeze the configuration of all the clocks in the system and store the frozen frequencies in
    // `clocks`
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let writer = flash.writer(
        stm32f1xx_hal::flash::SectorSize::Sz1K,
        stm32f1xx_hal::flash::FlashSize::Sz128K,
    );
    rprintln!("Reading page 127");
    let page = writer.read(127 * 1024, 16).unwrap();
    rprintln!("{:?}", &page[..16]);
    let value: u32 = u32::from_le_bytes(page[..4].try_into().unwrap());
    rprintln!("Value on page 127 is: {:?}", value);

    // Page must be erased before writing new value.
    rprintln!("Erasing page");
    let mut writer = writer;
    writer.erase(127 * 1024, 1024);

    let new_value = value + 1;
    rprintln!("Writing new value {}", new_value);
    writer.write(127 * 1024, &new_value.to_le_bytes()).unwrap();

    // Acquire the GPIOC peripheral
    let mut gpioc = dp.GPIOC.split();

    // Configure gpio C pin 13 as a push-pull output. The `crh` register is passed to the function
    // in order to configure the port. For pins 0-7, crl should be passed instead.
    let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
    // LED should be off initially (active-low on blue pill board).
    led.set_high();

    // Configure the syst timer to trigger an update every second
    let mut timer = Timer::syst(cp.SYST, &clocks).counter_hz();
    timer.start(1.Hz()).unwrap();

    let blink_amount = value % 10;
    rprintln!(
        "Blinking 'value % 10' times. 'value % 10 = {}'",
        blink_amount
    );
    // Wait for the timer to trigger an update and change the state of the LED
    for _ in 0..value {
        block!(timer.wait()).unwrap();
        led.set_low();
        block!(timer.wait()).unwrap();
        led.set_high();
    }

    loop {
        continue;
    }
}
```
2. ... Compile the example code. Note if you run this from stm32f1xx_hal examples directory: You have to add `rtt_target` and `panic_rtt_target` as dev dependencies with the cortex-m features and have to compile with `features="stm32f103"`. You probably know this, just for reference :)
3. ... upload to blue pill over stlink with `probe-run --chip STM32F103CB target/thumbv7m-none-eabi/release/examples/flash_writer`
4. ... observe error message above. Also observe the program actually works!

**Expected behavior**
I expect the rtt output to appear on screen.

**Desktop (please complete the following information):**
 - Linux (Ubuntu)

burrbull, 2022-03-06:
Is it official ST chip?

barafael, 2022-03-06:
@burrbull good point. Looks official to me, as per printing on the chip.

burrbull, 2022-03-06:
Try to move `rtt_init_print!();` to first line in `main`. Possibly `unwraps` break something.

Yatekii, 2022-03-06:
Hey :)

This is to be expected. You cannot access the memory bus while flashing is in progress and thus RTT fails to poll the memory. We could make this more robust and retry to fix this.

barafael, 2022-03-06:
I see, that makes sense! Thanks :)

barafael, 2022-03-06:
To be fair, this works just fine:

```rust
//! Reads a value from flash page 127, increments it, writes it back. Then, blinks n times,
//! where n: original value % 10.

#![deny(unsafe_code)]
#![no_std]
#![no_main]

use nb::block;

// use panic_halt as _;
use panic_rtt_target;

use cortex_m_rt::entry;
use rtt_target::{rprint, rprintln, rtt_init_print};
use stm32f1xx_hal::flash::FlashWriter;
use stm32f1xx_hal::{pac, prelude::*, timer::Timer};

use stm32f1xx_hal as _;

#[entry]
fn main() -> ! {
    rtt_init_print!();
    // Get access to the core peripherals from the cortex-m crate
    let cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let rcc = dp.RCC.constrain();

    // Freeze the configuration of all the clocks in the system and store the frozen frequencies in
    // `clocks`
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Configure the syst timer to trigger an update every second
    let mut timer = Timer::syst(cp.SYST, &clocks).counter_hz();
    timer.start(1.Hz()).unwrap();
    block!(timer.wait()).unwrap();

    let writer = flash.writer(
        stm32f1xx_hal::flash::SectorSize::Sz1K,
        stm32f1xx_hal::flash::FlashSize::Sz128K,
    );

    rprintln!("Reading page 127");
    let page = writer.read(127 * 1024, 16).unwrap();
    rprintln!("{:?}", &page[..16]);
    let value: u32 = u32::from_le_bytes(page[..4].try_into().unwrap());
    rprintln!("Value on page 127 is: {:?}", value);

    // Page must be erased before writing new value.
    rprintln!("Erasing page");
    let mut writer = writer;
    writer.erase(127 * 1024, 1024);

    let new_value = value + 1;
    rprintln!("Writing new value {}", new_value);
    writer.write(127 * 1024, &new_value.to_le_bytes()).unwrap();

    // Acquire the GPIOC peripheral
    let mut gpioc = dp.GPIOC.split();

    // Configure gpio C pin 13 as a push-pull output. The `crh` register is passed to the function
    // in order to configure the port. For pins 0-7, crl should be passed instead.
    let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
    // LED should be off initially (active-low on blue pill board).
    led.set_high();

    let blink_amount = value % 10;
    rprintln!(
        "Blinking 'value % 10' times. 'value % 10 = {}'",
        blink_amount
    );
    // Wait for the timer to trigger an update and change the state of the LED
    for _ in 0..blink_amount {
        block!(timer.wait()).unwrap();
        led.set_low();
        block!(timer.wait()).unwrap();
        led.set_high();
    }

    loop {
        continue;
    }
}
```

closed 2022-03-15 by barafael