probe-rs/probe-rs #1046: nrf9160: RTT messages not printed after a reset

folkertdev, 2022-03-07:
**Describe the bug**

RTT logs are not shown after a reset. 

**To Reproduce**
Steps to reproduce the behavior:
1. clone https://github.com/tweedegolf/nrf9160-rust-starter/tree/rtt-after-reset
2. run `cargo run --bin hello` 
3. press the reset button

**Expected behavior**
RTT would reconnect, so we see another "Hello, world!" printed after the reset.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

```
± |main U:2 ✗| → cargo run --bin hello
   Compiling nrf9160-rust-starter v0.1.0 (/home/folkert/tg/nrf9160-rust-starter)
    Finished dev [optimized + debuginfo] target(s) in 0.53s
     Running `probe-run --chip nRF9160_xxAA target/thumbv8m.main-none-eabi/debug/hello`
(HOST) INFO  flashing program (2 pages / 8.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Hello, world!
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:8
Error: A core architecture specific error occured

Caused by:
    0: Failed to write register TAR at address 0x00000004
    1: An error specific to the selected architecture occured
    2: Target device did not respond to request.
```

**Desktop (please complete the following information):**
 - Linux (Ubuntu)

**Additional context**
Add any other context about the problem here.

cc @diondokter

Yatekii, 2022-03-07:
Well, the target goes into WFI, so it most likely interrupts the debug connection, which would make sense.
You could see if the issue persists when you replace the `wfi` instruction with another `rprintln` or the likes :)

As for a nicer error message, that would really be up to `probe-run` :) probe-rs just detects that there was no ack :)

folkertdev, 2022-03-07:
I'm not sure I understand. Another `println` does not work because the main function needs to return `!`. Hence, we must have some diverging operation, either a loop or a panic. In practice `main` most applications would have a loop that waits for interrupts or events.

So are you saying that RTT cannot be used across resets when `main` contains a loop that waits for interrupts?

Yatekii, 2022-03-07:
You have to replace the WFI instruction. not the loop. WFI disables debug communication, unless properly configured. A soft reset should work without issues. Hard reset wont, for obvious reasons. That can be handled hostside tho.

noppej, 2022-03-07:
@Yatekii Could we not solve this more gracefully by configuring a debug sequence to keep the connection alive during wfi?

Yatekii, 2022-03-07:
Well yeah, the framework is there :) It just needs some1 to port them all :) Plus, some targets still fuck up and break ...

Dirbaio, 2022-03-07:
nrf52 and nrf53 have the same issue.

The issue is not WFI. nRF's debug keeps working across WFI. (they honor the DP CDBGPWRUPREQ, unlike stm32 >_>)

The issue is while RST is held, SWD doesn't work. (RST holds the SWD logic in reset, unlike stm32 >_>). I don't think there's anything we can do from probe-rs, other than making RTT resilient to errors (on error, keep trying to reattach...)

Yatekii, 2022-03-07:
Ah yeah thanks, I was sure nRF52s had no issue with wfi, but could not make much other sense, so this was a best guess.

Do they do that in softreset? I think there is two different softresets and maybe the other one would work?

Yeah, I think ultimately being more resilient would be the key. One problem atm I think is that our `Session` is not really recoverable. Would have to make this a bit better.

Dirbaio, 2022-03-07:
the firmware resetting itself (with `SCB::sys_reset()`) works fine. It fails when you hold RST low (for example by pressing the reset button)

Yatekii, 2022-03-07:
Ah yeah, but that is ofc. absolutely correct and desired behavior. You ALWAYS want hard reset to reset EVERYTHING. Imagine that would not be the case. Total mayhem and 0 reliability ...

But yes, we can work around that by recovering the Session :)

I think we can close this, as it is for `probe-run` to handle this gracefully and totally expected behavior.

RTT retrial should be handled by the the binary tools, I guess, or is anyone of different oppinion?

noppej, 2022-04-04:
@Yatekii I agree. The RTT retry should be handled by the binary tools. I just don't see a reasonably way to handle it at the API level.