probe-rs/probe-rs #1057: GDB server cannot load binary?

huming2207, 2022-03-17:
**Describe the bug**

Hi there,

I've installed the latest `probe-rs-cli` from git commit `222264f4b1393ac8bcb743ee9865f92ba63365db`, and I run this to spin up the GDB server: 

```
probe-rs-cli gdb --chip STM32L071KBUx --protocol swd --speed 4000 --reset-halt --connect-under-reset --allow-erase-all
```

The GDB server does start at port 1337. Then when I use a GDB instance to connect to it, it complains something like this:

```
(gdb) target remote :1337
Remote debugging using :1337
warning: while parsing target description (at line 4): Target description specified unknown architecture "armv6-m"
warning: Could not load XML target description; ignoring
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xffffffff in ?? ()
(gdb) info target
Remote serial target in gdb-specific protocol:
Debugging a target over a serial line.
(gdb) load firmware.elf
Loading section .isr_vector, size 0xc0 lma 0x8000000
Load failed
```

However, if I use PyOCD with the command `pyocd gdbserver -t stm32l071kbux`, it seems to work (some confidential info were replaced with ****):

```
(gdb) target remote :3333
Remote debugging using :3333
warning: Architecture rejected target-supplied description
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xffffffff in ?? ()
(gdb) target remote :3333Quit
(gdb) load firmware.elf
Loading section .isr_vector, size 0xc0 lma 0x8000000
Loading section .text, size 0xa778 lma 0x800****
Loading section .rodata, size 0x800 lma 0x800****
Loading section .ARM, size 0x8 lma 0x800b038
Loading section .init_array, size 0x8 lma 0x8000****
Loading section .fini_array, size 0x4 lma 0x800****
Loading section .data, size 0x530 lma 0x800****
Start address 0x0800****, load size ****
Transfer rate: 3 KB/sec, 1602 bytes/write.

```

I guess maybe the architecture is 

**To Reproduce**

See above

**Expected behavior**

The GDB server should load the ELF binary and start debugging

**Stacktrace**

N/A

**Desktop (please complete the following information):**

Ubuntu 21.10, 

**Additional context**

- Rust 1.59.0 (9d1b2106e 2022-02-23), 
- debug probe is DAPLink v2 or ST-Link v2. 
- GDB comes from Ubuntu's apt repo, version: (Ubuntu 11.1-0ubuntu2) 11.1

Yatekii, 2022-03-17:
This is not implemented :) You can download the binary with `probe-rs-cli download` beforehand. The GDB stub was only intended as an intermediate solution until we have proper support for Microsoft DAP going.

huming2207, 2022-03-18:
Hi @Yatekii 

> This is not implemented :) You can download the binary with probe-rs-cli download beforehand. The GDB stub was only intended as an intermediate solution until we have proper support for Microsoft DAP going.

Oh oops, understood. 

I'm using CLion and CLion only supports GDB, so I have to use GDB instead, and personally I'd prefer using GDB as we can do some automation with GDB (e.g. `gdbinit` or other scripting) which I remember DAP can't do that for now. Meanwhile, our project is a mixed C & Rust project, so I'd prefer using CLion instead of VSCode as its C/C++ plugin is quite unreliable compared to CLion. 

I will leave this issue open for now, or if you would like to close it then you may close it instead.

Regards,
Jackson

ryan-summers, 2022-05-21:
@huming2207 You can use cargo-embed with GDB enabled to flash the device and automatically load the ELF into GDB when you start it up. While the GDB stub doesn't load the ELF onto the target, probe-rs does that natively using the probe. Then, you load the ELF + debug symbols into the GDB stub and connect to your target to start debugging. That sounds like it should do what you need it to without having to support reprogramming within the GDB stub.

Yatekii, 2022-05-21:
I thinkt the problem here is the CLion workflow which expects the load command to behave like this. This should actually be really eqsy addition tho :)

Hoohaha, 2023-06-30:
Does there have any plan to support load command for gdbserver?

Yatekii, 2023-09-08:
If someone wants to PR this I am happy to review :)

PR #3394 (2025-06-14): Support gdb load command
Closes #1057

Currently marked as draft because the flash operations extension in gdbstub PR is not merget yet:
(https://github.com/daniel5151/gdbstub/pull/172)
Maybe someone who's interested could test it meanwhile.

Some notes:
- I don't know if it make sense to reset after flash is done. Usually a `monitor reset` is necessary anyways.
- For the board I test flash sector info are wrong for OTP, option bytes etc.. That's not a problem for the load use case, but I wonder if there's some way to filter out those non-suited-for-exec-binary regions.
closed 2025-06-14 by Yatekii