probe-rs/probe-rs #1058: Unable to `cargo flash` ATSAMD51N with `hs-probe`, but it works with a J-Link probe

bradleyharden, 2022-03-20:
**Describe the bug**
Flashing a custom ATSAMD51N board with `cargo flash` and my J-Link probe works just fine. But if I run `cargo flash` with my `hs-probe`, I get the following:

```text
$ cargo flash --release --chip atsamd51n20a --connect-under-reset --log debug
       DEBUG cargo_flash > Changed working directory to .
       DEBUG probe_rs_cli_util > Running '/home/hardebj1/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo' in directory /home/hardebj1/projects/fluxgate
    Finished release [optimized + debuginfo] target(s) in 0.06s
    Flashing /home/hardebj1/projects/fluxgate/target/thumbv7em-none-eabihf/release/fluxgate
       DEBUG probe_rs::config::registry > Searching registry for chip with name atsamd51n20a
       DEBUG probe_rs::config::registry > Exact match for chip name: ATSAMD51N20A
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.24.11584
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: true
       DEBUG jaylink                          > libusb can detach kernel driver: true
       DEBUG probe_rs::probe::cmsisdap::tools > Opening 1209:4853 in CMSIS-DAPv2 mode
       DEBUG probe_rs::probe::cmsisdap::commands > Draining probe of any pending data.
       DEBUG probe_rs::probe::cmsisdap::commands > Attempt 1 to find packet size
       DEBUG probe_rs::probe::cmsisdap::commands > Success: packet size is 512
       DEBUG probe_rs::probe::cmsisdap::commands > Configuring probe to use packet size 512
       DEBUG probe_rs::probe::cmsisdap           > Detected probe capabilities: Capabilities { _swd_implemented: true, _jtag_implemented: true, swo_uart_implemented: true, swo_manchester_implemented: false, _atomic_commands_implemented: false, _test_domain_timer_implemented: false, swo_streaming_trace_implemented: true, _uart_communication_port_implemented: false, uart_com_port_implemented: false }
       DEBUG probe_rs::probe::cmsisdap           > Probe SWO buffer size: 256
        INFO cargo_flash                         > Protocol speed 1000 kHz
       DEBUG probe_rs::probe::cmsisdap           > Attaching to target system (clock = 1000kHz)
       DEBUG probe_rs::probe::cmsisdap           > Adding command to batch: Read(port=DebugPort, addr=0)
       DEBUG probe_rs::probe::cmsisdap           > 1 items in batch
       DEBUG probe_rs::probe::cmsisdap           > Attempting batch of 1 items
       DEBUG probe_rs::architecture::arm::communication_interface > Selecting DP Default
       DEBUG probe_rs::architecture::arm::dp                      > Writing DP register ABORT, value=0x0000001e
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=DebugPort, addr=0, data=0x0000001e
       DEBUG probe_rs::architecture::arm::dp                      > Writing DP register SELECT, value=0x00000000
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Write(port=DebugPort, addr=8, data=0x00000000
       DEBUG probe_rs::architecture::arm::dp                      > Reading DP register CTRL/STAT
       DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=4)
       DEBUG probe_rs::probe::cmsisdap                            > 3 items in batch
       DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 3 items
       DEBUG probe_rs::probe::cmsisdap                            > 1 of batch of 3 items suceeded
       DEBUG probe_rs::probe::cmsisdap                            > Detaching from CMSIS-DAP probe
        INFO cargo_flash                                          > Metadata {
    chip: Some(
        "\"atsamd51n20a\"",
    ),
    probe: Some(
        "\"CMSIS-DAP\"",
    ),
    speed: Some(
        "1000",
    ),
    release: "0.12.1",
    commit: "d4581f7",
}
       Error 0: Failed to get information for AP ApAddress { dp: Default, ap: 0 }
```

**To Reproduce**
1. Run `cargo flash --release --chip atsamd51n20a` with a J-Link probe. Observe that it flashes successfully
2. Run `cargo flash --release --chip atsamd51n20a` with an `hs-probe`. Observe an error.
3. Retry with `cargo flash --release --chip atsamd51n20a --connect-under-reset`. Observe the same error.

**Expected behavior**
Be able to flash successfully with either probe.

**Desktop (please complete the following information):**
Ubuntu 20.04

**Additional context**
I am working with `cargo-flash` and `hs-probe-firmware` from `master`. My J-Link probe firmware is also up-to-date.

9names, 2022-04-03:
Note for anyone trying to reproduce this for debugging: this error appears to be specific to the hs-probe firmware.
Probes that are based on the [CMSIS_5](https://github.com/ARM-software/CMSIS_5) or [DAPLink](https://github.com/ARMmbed/DAPLink) sources connect and program successfully, without requiring --connect-under-reset.
And, as stated above, j-link also works correctly.

My testing was against ATSAMD51G19 (adafruit itsybitsy m4 express) on ubuntu 20.04 with kernel 5.16 w/  probe-rs master commit 228ca5863e5519f46d6a96226ea918bf277fdbf5, running benchmark.rs
For testing hs-probe firmware I used hs-probe and [pico-probe](https://github.com/korken89/pico-probe) (which uses a port of the hs-probe firmware).
The other CMSIS-DAP debug probes I tested against (mculink, lpclink2, dapper-mime, mculink w/ daplink firmware) are able to connect and run the benchmark successfully. I tested running the benchmark with lpclink2 50 times consecutively, no errors.

i was running both dapper-mime and pico-probe firmware on a raspberry pi pico, with both firmwares configured to use the same pins to allow the same wiring harness.
I tested both back-to-back (not changing connections, just changing firmware) running dapper-mime and pico-probe without disconnecting to ensure that the debug core wasn't being left in a weird state - no change.

9names, 2022-04-06:
I've captured some traces with dapper-mime, pico-probe and hs-probe with RUST_LOG=trace and sigrok with saleae clone @ 24MHz, running benchmark.rs with --speed 300

I grabbed the hs-probe one twice just in case it produces different results each time, they look the same to me though

[atsamd51_debugging.zip](https://github.com/probe-rs/probe-rs/files/8425958/atsamd51_debugging.zip)

9names, 2022-04-06:
I fed the captures through sigrok's swd decoder. the output was the same for pico-probe, hs-probe and the second hs-probe trace.
it seems to match probe-rs' interpretation of the data too, so that's nice.

Here's the output from that, in case anyone wants to look at that without loading anything in pulseview or sigrok.
[hsprobe_decode.txt](https://github.com/probe-rs/probe-rs/files/8426592/hsprobe_decode.txt)
[dappermime_decode.txt](https://github.com/probe-rs/probe-rs/files/8426594/dappermime_decode.txt)

9names, 2022-04-23:
Comments on #1081 suggested this might be fixed.
Tested on 6968dc774f86d248368a13126cdf4e5054398216, still fails to connect with hsprobe.

HSProbe (non-turbo) - Trace after a reset, with the chip in bootloader mode:
[probe-rs-hsprobe-atsamd51-trace1.txt](https://github.com/probe-rs/probe-rs/files/8546495/probe-rs-hsprobe-atsamd51-trace1.txt)

HSProbe (non-turbo) - Second attempt, after the first one failed to connect:
[probe-rs-hsprobe-atsamd51-trace2.txt](https://github.com/probe-rs/probe-rs/files/8546496/probe-rs-hsprobe-atsamd51-trace2.txt)

As before, for comparison here is a trace with mculink that succeeds (performed with a reset as per first trace)
[probe-rs-mculink-atsamd51-trace_w_reset.txt](https://github.com/probe-rs/probe-rs/files/8546678/probe-rs-mculink-atsamd51-trace_w_reset.txt)

And since I got it working for my benchmarking, I can now also confirm that rustdap succeeds running benchmark.rs too:
[probe-rs-rustdap-atsamd51-trace_w_reset.txt](https://github.com/probe-rs/probe-rs/files/8546679/probe-rs-rustdap-atsamd51-trace_w_reset.txt)

commit 3710fcaa (2022-09-16): Merge #1225

1225:  Don't expect data in faulty DAP Transfer responses r=Yatekii a=jgust

Debuggers such as HS-Probe (rightfully) send no response data when
DAP Transfer Read requests fail. Instead of always expecting response
data from such requests, only attempt to parse the response data for the
requests that were successful. This propagates the FAULT status properly
up the stack as opposed to triggering a NotEnoughData error.

Fixes #1058
PR #1225 (2022-09-16): Don't expect data in faulty DAP Transfer responses
Debuggers such as HS-Probe (rightfully) send no response data when
DAP Transfer Read requests fail. Instead of always expecting response
data from such requests, only attempt to parse the response data for the
requests that were successful. This propagates the FAULT status properly
up the stack as opposed to triggering a NotEnoughData error.

Fixes #1058
closed 2022-09-16 by bors[bot]