probe-rs/probe-rs #1066: Failed to connect to RP2040 using DAP42

eivindbergem, 2022-04-04:
**Describe the bug**

Connecting to RP2040 using DAP42 on STM32 blue pill fails with vague error message about multi-drop:

```
> probe-rs-cli erase --chip rp2040

Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect SWDIO and SWCLK from DAP42 probe to RP2040
2. Run `probe-rs-cli erase --chip rp2040`

**Expected behavior**

It should connect to target and erase flash.

**Stacktrace**
```
> RUST_LOG=probe_rs=trace probe-rs-cli erase --chip rp2040
 DEBUG probe_rs::config::registry > Searching registry for chip with name rp2040
 DEBUG probe_rs::config::registry > Exact match for chip name: RP2040
 DEBUG probe_rs::config::registry > Partial match for chip name: RP2040_SELFDEBUG
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 TRACE probe_rs::probe::cmsisdap::tools > DAP103-BLUEPILL CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools >   Interface 0: DAP103-BLUEPILL CMSIS-DAP
 TRACE probe_rs::probe::cmsisdap::tools > Will use interface number 0 for CMSIS-DAPv1
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
 TRACE probe_rs::probe::cmsisdap::tools > CMSIS-DAP device with USB path: "/dev/hidraw3"
 TRACE probe_rs::probe::cmsisdap::tools >                 product_string: "DAP103-BLUEPILL CMSIS-DAP"
 TRACE probe_rs::probe::cmsisdap::tools >                      interface: 0
 TRACE probe_rs::probe::cmsisdap::tools > Ignoring duplicate DAP103-BLUEPILL CMSIS-DAP (VID: 1209, PID: da42, Serial: 321913674975524953FF6A06, CmsisDap)
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
 TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching 1209:da42:321913674975524953FF6A06
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 001: ID 1d6b:0002
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 008 Device 003: ID 046d:c050
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 008 Device 002: ID 046d:c31c
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 008 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 007 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 006 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 001: ID 1d6b:0002
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 005 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 044: ID 1209:da42
 TRACE probe_rs::probe::cmsisdap::tools > DAP103-BLUEPILL CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools >   Interface 0: DAP103-BLUEPILL CMSIS-DAP
 TRACE probe_rs::probe::cmsisdap::tools > Will use interface number 0 for CMSIS-DAPv1
 DEBUG probe_rs::probe::cmsisdap::tools > Could not open 1209:da42 in CMSIS-DAP v2 mode
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 001: ID 1d6b:0001
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: Access
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 1209:da42 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::cmsisdap::commands > Draining probe of any pending data.
 DEBUG probe_rs::probe::cmsisdap::commands > Attempt 1 to find packet size
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FF, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 02, 40, 00]...
 DEBUG probe_rs::probe::cmsisdap::commands > Success: packet size is 64
[... 86 lines cut ...]
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 05, 00, 01, 02, 00]...
 TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [05, 00, 07, 00]...
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 TRACE probe_rs::probe::cmsisdap                            > Transfer status: NACK
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 WARN  probe_rs::probe::cmsisdap                            > Giving up on TARGETSEL, too many retries.
 DEBUG probe_rs::probe::cmsisdap                            > Detaching from CMSIS-DAP probe
 TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 03, 00]...
 TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [03, 00]...
 TRACE probe_rs::probe::cmsisdap::commands                  > Transmit buffer: [00, 01, 00]...
 TRACE probe_rs::probe::cmsisdap::commands                  > Receive buffer: [01, 00]...
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

**Desktop (please complete the following information):**
- Arch Linux

**Related issues**

- #997

noppej, 2022-04-04:
@eivindbergem Can you try connecting with a lower speed? Maybe start with a setting of 1000, and work your way up from a successful connect until it fails?

eivindbergem, 2022-04-05:
@noppej Doesn't appear to make a difference:
```
> probe-rs-cli erase --chip rp2040 --speed 1000
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

Trace output is identical regardless of speed.

eivindbergem, 2022-04-05:
I did a system upgrade and a reboot, and now it's working. It worked fine with an STM32 target, so it didn't occur to me that it could be that.

closed 2022-04-05 by eivindbergem