probe-rs/probe-rs #1068: Picoprobe not detected when mounted in WSL

KorvinSzanto, 2022-04-05:
**Describe the bug**
Picoprobe not detected when using ubuntu WSL2 and usbipd

**To Reproduce**
Steps to reproduce the behavior:
1. Set up usbipd in windows so that you can attach a usb device to wsl
    1. Find the picoprobe with `usbipd list`, for me it's BUSID `4-3`
    2. Attach the probe to wsl with `usbipd wsl attach --busid 4-3`
    3. Verify the device is attached in wsl with `lsusb`, for me I see `Bus 001 Device 002: ID 2e8a:0004 Raspberry Pi Picoprobe`
4. Attempt to detect the probe with probe-rs `probe-rs-clie list`, I get "No devices were found."
5. Try again as sudo, same result

**Expected behavior**
I expect that the attached USB device would be detected as an available probe in ubuntu but instead it shows no available probes.

**Additional context**
I originally was trying to get things functioning in a vscode devcontainer. I got as far as getting the usb device properly mounted but ran into this same issue so I opted to simplify and try to get just WSL working first.

Tiwalun, 2022-04-05:
Are you using this probe: https://github.com/raspberrypi/picoprobe?

We unfortunately don't support it, because it's using a custom protocol. The following alternatives should work with probe-rs:

- https://github.com/majbthrd/pico-debug
- https://github.com/majbthrd/DapperMime/

9names, 2022-04-06:
I wouldn't recommend pico-debug, that is currently not working well for C code or at all for Rust.
The list of probes that have been tested with rp2040 (and instructions for setting up DapperMime) are in the rp2040 docs.
https://github.com/rp-rs/rp2040-project-template/blob/main/debug_probes.md

KorvinSzanto, 2022-04-07:
> We unfortunately don't support it, because it's using a custom protocol
Understood, I was originally using `probe-run` but I opened the issue here because I saw the same behavior in `probe-rs-cli` and assumed the issue stemmed from here.

I've been able to get picoprobe + rust working well with openocd and gdb fwiw.

closed 2022-04-07 by KorvinSzanto

elpiel, 2023-01-25:
> Are you using this probe: https://github.com/raspberrypi/picoprobe?
> 
> We unfortunately don't support it, because it's using a custom protocol.

Would you mind elaborating on this?
The documentation said that it uses:

>  A vendor-specific interface for SWD probe data conforming to CMSIS-DAP v2.

Since I'm just getting started with the whole debuggers ecosystem it seems to me that it would be a big win if probe-run is able to work with such a setup (one Pico as debugger using picoprobe).

Tiwalun, 2023-01-25:
The current version should be supported, yes. The initial version used a different protocol.

elpiel, 2023-01-26:
I will add docs soon to `probe-run` but the gist is this:

1. You need the special release of uf2 file that supports CMSIS DAP v2 found here:
https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.0.1 (latest version in the time of writing this)
2. Run probe-run

Interestingly, `src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl` fails:

```
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E660C0D1C7144034
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
```

Notes: I'm using the picoprobe branch of RP pi `openocd` found here: https://github.com/raspberrypi/openocd/tree/picoprobe

Tiwalun, 2023-01-26:
You shouldn't need a special branch of openocd anymore, cmsis-dap should work with standard openocd now. The picoprobe branch seems a bit outdated.

elpiel, 2023-01-27:
Yes, thank you for mentioning this, I did try it with the default branch and updated my comment accordingly.
Hopefully it would help other people having the same issues.