probe-rs/probe-rs #1069: Newest ST-LINK v3 firmware not supported

birkenfeld, 2022-04-06:
**Describe the bug**
Updating my Nucleo STM32G431 to the current STLINK-V3 firmware from STSW-LINK007 (V3J9M3) resulted in an USB device with ID 0483:3754 - this is not in the probe.rs udev rules, and when added, doesn't seem to be recognized by probe-rs (verified by looking in https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/stlink/usb_interface.rs#L24)

**To Reproduce**
Steps to reproduce the behavior:
1. Download current STSW-LINK007 from https://www.st.com/en/development-tools/stsw-link007.html
2. Update a STLINK-V3 with it
3. Try to list probes with probe-run

**Expected behavior**
New STLINK-V3 ID to be recognized

**Desktop (please complete the following information):**
Linux Fedora

birkenfeld, 2022-04-06:
Important addendum: This is after I selected "Change type" to "STM32 Debug+VCP" in the FW upgrade tool.

With the default "STM32 Debug+Mass storage+VCP" the USB ID is 374e which *is* supported.

Tiwalun, 2022-04-06:
Could you please try if it works with the linked branch?

closed 2022-04-06 by bors[bot]

birkenfeld, 2022-04-07:
Yes, that works nicely. Thanks!

commit 225d73f7 (2022-04-06): Merge #1070

1070: Add additional USB id for ST-Link V3 without mass storage r=Yatekii a=Tiwalun

Add additional USB ID for ST-Link V3 variant without mass storage. Fix #1069.
PR #1070 (2022-04-06): Add additional USB id for ST-Link V3 without mass storage
Add additional USB ID for ST-Link V3 variant without mass storage. Fix #1069.