probe-rs/probe-rs #1108: Probe-run is not supporting for ATSAM4L8-Xplained pro board.

UdayakumarHidakal, 2022-05-10:
**Probe-run not supporting for ATSAM4L8-Xplained pro board**

I am using this ```ATSAM4L8-Xplained pro``` which is having ```atmel EDBG (on chip debugger)```
As it is listed below ```probe-run supports ATSAM4LC8C```,but i am unable to flash binary .

```SAM4 Series
    Variants:
        ATSAM4E16C
        ATSAM4E16E
        ATSAM4E8C
        ATSAM4E8E
        ATSAM4LC2A
        ATSAM4LC2B
        ATSAM4LC2C
        ATSAM4LC4A
        ATSAM4LC4B
        ATSAM4LC4C
        ATSAM4LC8A
        ATSAM4LC8B
        ATSAM4LC8C
        ATSAM4LS2A
        ATSAM4LS2B
        ATSAM4LS2C
        ATSAM4LS4A
        ATSAM4LS4B
        ATSAM4LS4C
        ATSAM4LS8A
        ATSAM4LS8B
        ATSAM4LS8C
```
while flashing the code using this command  ```probe-run microchip --chip ATSAM4LC8C``` it's throwing some errors like this
```Error: Error while flashing

Caused by:
    0: Something during the interaction with the core went wrong
    1: A core architecture specific error occured
    2: Failed to read register DRW at address 0x0000000c
    3: An error specific to the selected architecture occured
    4: Target device did not respond to request.
```
**Desktop Info:**
- Linux (Ubuntu)

So i need support for solving this issue.

UdayakumarHidakal, 2022-05-13:
```probe-run``` not working properly , after flashing i am getting error.

```
(HOST) INFO  flashing program (3 pages / 1.50 KiB)
(HOST) INFO  success!
Error: A core architecture specific error occured

Caused by:
    0: Failed to read register DRW at address 0x0000000c
    1: An error specific to the selected architecture occured
    2: Target device did not respond to request.
```

Yatekii, 2022-05-14:
Hey :)

The atmel EDBG is not supported. How did you manage to flash?
Also, ATSAMD have issues. Try probe-rs master which has several fixes.