probe-rs/probe-rs #1124: Error Failed to open the debug probe.

John15321, 2022-06-05:
**Describe the bug**
Im following this tutorial with the same hardware. Im trying to flash the example blinky program onto my Blue Pill but I get this error:

```bash
$ cargo flash --release --chip STM32F103C8
    Finished release [optimized + debuginfo] target(s) in 0.03s
    Flashing /Users/john/Documents/Programming/embedded-stuff/base/target/thumbv7m-none-eabi/release/blue_pill_base
       Error Failed to open the debug probe.

        Hint This could be a permission issue. Check our guide on how to make all probes work properly on your system: https://probe.rs/docs/getting-started/probe-setup
```

My ST-Link is properly connected and visible to the system. When I run `cargo flash --list-probes` I can clearly see it:

```bash
$ cargo flash --list-probes
The following debug probes were found:
[0]: STLink V2 (VID: 0483, PID: 3748, Serial: 34006D063142323713511943, StLink)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Blue Pill with STLinker to the computer
2. Try to flash it by running `cargo flash --release --chip STM32F103C8`

**Expected behavior**
It should flash the blinky program

**Desktop (please complete the following information):**
macOS 12.3.1 arm64 (Darwin 21.4.0 arm64)
CPU: 10-core Apple M1 Max
rust version: `1.59.0`
cargo-flash version: `cargo-flash 0.12.1`

Yatekii, 2022-06-05:
This is most likely because your probe driver is outdated. Can you try updating it? :) the newest version of cargo-flash on master already shows a better hint and if you enable --log=INFO you will see it even with your version :)

John15321, 2022-06-05:
> This is most likely because your probe driver is outdated. Can you try updating it? :) the newest version of cargo-flash on master already shows a better hint and if you enable --log=INFO you will see it even with your version :)

Sure but tbh I have no idea how to do this. Im coming back to embedded after a long time and the STLinker literally has dust on it 😅

John15321, 2022-06-05:
> --log=INFO

that log does give me much tbh
```
    Finished release [optimized + debuginfo] target(s) in 0.01s
    Flashing /Users/john/Documents/Programming/embedded-stuff/base/target/thumbv7m-none-eabi/release/blue_pill_base
        INFO cargo_flash > Metadata {
    chip: Some(
        "\"STM32F103C8\"",
    ),
    probe: None,
    speed: None,
    release: "0.12.1",
    commit: "crates.io",
}
       Error Failed to open the debug probe.

        Hint This could be a permission issue. Check our guide on how to make all probes work properly on your system: https://probe.rs/docs/getting-started/probe-setup
```

John15321, 2022-06-05:
Also this is what happens when I reattach the STLinker

[image]

Im peeking using the Console application on mac, looking at `icdd`

John15321, 2022-06-05:
@Yatekii please help

Tiwalun, 2022-06-05:
@John15321 Please update the ST-Link firmware using the [updater tool](https://www.st.com/en/development-tools/stsw-link007.html) from ST, and try again.

John15321, 2022-06-05:
> Member

After dwonloading the updater and installing an x86 version of java I managed to update it. Now I'm getting a different error

This is now what I get if I just want to flash it:

```bash
    Finished release [optimized + debuginfo] target(s) in 0.01s
    Flashing /Users/john/Documents/Programming/embedded-stuff/base/target/thumbv7m-none-eabi/release/blue_pill_base
        INFO cargo_flash > Protocol speed 1800 kHz
        INFO probe_rs::probe::stlink > Target voltage (VAPP): 3.27 V
        WARN probe_rs::probe::stlink > send_jtag_command 242 failed: JtagGetIdcodeError
        INFO cargo_flash             > Metadata {
    chip: Some(
        "\"STM32F103C8\"",
    ),
    probe: Some(
        "\"ST-Link V2\"",
    ),
    speed: Some(
        "1800",
    ),
    release: "0.12.1",
    commit: "crates.io",
}
       Error Connecting to the chip was unsuccessful.

        Hint A hard reset during attaching might help. This will reset the entire chip. Run with `--connect-under-reset` to enable this feature.
```

And If I add that flag I get this:

```bash
$ cargo flash --release --chip STM32F103C8 --log=INFO --connect-under-reset
    Finished release [optimized + debuginfo] target(s) in 0.01s
    Flashing /Users/john/Documents/Programming/embedded-stuff/base/target/thumbv7m-none-eabi/release/blue_pill_base
        INFO cargo_flash > Protocol speed 1800 kHz
        INFO probe_rs::session > Custom reset sequences are not supported on ST-Link V2.
        INFO probe_rs::session > Falling back to standard probe reset.
        INFO probe_rs::probe::stlink > Target voltage (VAPP): 3.27 V
        WARN probe_rs::probe::stlink > send_jtag_command 242 failed: JtagGetIdcodeError
        INFO cargo_flash             > Metadata {
    chip: Some(
        "\"STM32F103C8\"",
    ),
    probe: Some(
        "\"ST-Link V2\"",
    ),
    speed: Some(
        "1800",
    ),
    release: "0.12.1",
    commit: "crates.io",
}
       Error 0: An error with the usage of the probe occured
             1: An error specific to a probe type occured
             2: Command failed with status JtagGetIdcodeError
```

Yatekii, 2022-06-05:
Did you connect the physical reset line?

John15321, 2022-06-05:
> Did you connect the physical reset line?

no, should I?

Yatekii, 2022-06-05:
Yes. You always should :) And in this case in particular connect-under-reset does not work without :)

Yatekii, 2023-05-08:
This was resolved and user error judging from the comments.

closed 2023-05-08 by Yatekii