probe-rs/probe-rs #1127: error: Found argument '--chip' which wasn't expected

jrmurdoch, 2022-06-09:
Trying to get going with vscode but checking i have it working via command first

E:\Development\Rust\hello-world>probe-rs-debugger debug --chip esp32c3 --program-binary ./target/riscv32imc-esp-espidf/debug/hello-world --protocol jtag
error: Found argument '--chip' which wasn't expected, or isn't valid in this context

        If you tried to supply `--chip` as a value rather than a flag, use `-- --chip`

USAGE:
    probe-rs-debugger debug [OPTIONS]

For more information try --help

E:\Development\Rust\hello-world>

burrbull, 2022-06-09:
you need to specify chip in `launch.json` instead of pass its directly:
https://github.com/burrbull/f411-test/blob/f65db4e772f8d0dce464a0a87bf024d14766147d/.vscode/launch.json#L10

noppej, 2022-06-09:
`probe-rs-debugger` used to accept more command line arguments, and then acted in CLI mode. We eliminated that because of overlap with other tools [see this discussion](https://github.com/probe-rs/probe-rs/issues/828), and now only operates as a server for the VSCode extension - hence the need to code your configuration in a `launch.json`.

The documentation has some [configuration examples](https://probe.rs/docs/tools/vscode/#start-a-debug-session-with-minimum-configuration). 

If you are still having problems after following those steps, please paste your `launch.json` and one of us will help you get going :)

quentinmit, 2022-07-12:
The README still documents the CLI debugger.

(`launch.json` doesn't really help if I'm not already using VSCode... I tried `probe-rs-cli debug` but it doesn't seem like it's anywhere near as complete as `probe-rs-debugger` (was).)

noppej, 2022-07-12:
Oops ... I will update the README. That was an oversight.

I get your point about if you don't have VSCode ... sorry but we are all volunteer contributors, and maintaining overlapping functionality (multiple CLI tools) just steals time away of adding new features. 

Most of the functionality in `probe-rs-debugger` is actually just a front-end for the core `probe-rs::debug` functionality, so if you want to PR additions to `probe-rs-cli`, it should be relatively easy.

closed 2022-08-11 by bors[bot]