probe-rs/probe-rs #1132: Failed to open debug probe on windows using CMSIS-DAP

Alintya, 2022-06-12:
**Describe the bug**
I'm trying to flash a STM32F411 board (black pill) using another one as CMSIS-DAP probe connected via usb.
The probe seems to be recognized and does show up with `--list-probes` but I can't get it to flash.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect probe via usb
2. Connect target board via SWD to probe
3. Run `cargo flash --chip STM32F411CEUx --release --log debug`

**Expected behavior**
Attempting to flash the target after successfully connecting to the probe.

**Log**
```
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.24.11584
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: false
       DEBUG jaylink                          > libusb can detach kernel driver: false
       DEBUG probe_rs::probe::cmsisdap::tools > Could not open c251:f001 in CMSIS-DAP v2 mode
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open c251:f001 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG probe_rs::probe::espusbjtag::protocol  > Acquired libusb context.
        INFO cargo_flash                            > Metadata {
            chip: Some(
                "\"STM32F411CEUx\"",
                ),
                probe: None,
                speed: None,
                release: "0.12.1",
                commit: "crates.io",
          }
       DEBUG cargo_flash::diagnostics               > 0: Probe could not be created
       DEBUG cargo_flash::diagnostics               > 1: Probe was not found.
       Error Failed to open the debug probe.
```

**Desktop (please complete the following information):**
 - Windows 11

**Additional context**
[CMSIS-DAP firmware](https://github.com/WeActTC/MiniSTM32F4x1/tree/master/SDK/CMSIS-DAP)

burrbull, 2022-06-12:
`lsusb -vv`

facuver, 2022-07-03:
Hello, did you found a solution?
I have the same issue but trying to program a RP2040 with the blackpill.

bugadani, 2024-03-30:
I don't have anything to add here except that [CMSIS-DAP firmware](https://github.com/WeActTC/MiniSTM32F4x1/tree/master/SDK/CMSIS-DAP) in the issue description is now a broken link.