probe-rs/probe-rs #1134: xmc4000: breakpoint must be on half-word boundaries

willglynn, 2022-06-13:
**Describe the bug**

`architecture::arm::core::armv7m::FpRev1CompX::get_breakpoint_comparator()` returns an error while attempting to use `probe-run` on an [XMC4800](https://www.infineon.com/cms/en/product/microcontroller/32-bit-industrial-microcontroller-based-on-arm-cortex-m/32-bit-xmc4000-industrial-microcontroller-arm-cortex-m4/xmc4800/).

```
(HOST) WARN  Could not clear all hardware breakpoints: ArchitectureSpecific(Other(Unsupported breakpoint comparator value 0xc80e46a5 for HW breakpoint. Breakpoint must be on half-word boundaries))
└─ probe_rs::session @ /Users/willglynn/IdeaProjects/probe-rs/probe-rs/src/session.rs:496
Error: A core architecture specific error occured

Caused by:
    Unsupported breakpoint comparator value 0xc80e46a5 for HW breakpoint. Breakpoint must be on half-word boundaries
```

**To Reproduce**
Steps to reproduce the behavior:
1. Obtain a [KIT_XMC48_IOT_AWS_WIFI](https://www.infineon.com/cms/en/product/evaluation-boards/kit_xmc48_iot_aws_wifi/).
2. Connect to the onboard J-Link debugger, or power the board and use an external CMSIS-DAP probe.
3. Check out [my xmc4000-enabled probe-run branch](https://github.com/willglynn/probe-run/tree/v0.3.3-xmc4000) to use [my xmc4000-enabled probe-rs branch](https://github.com/willglynn/probe-rs/tree/add_xmc4000) (which is just `target-gen`'d from the [vendor's CMSIS pack](https://www.infineon.com/cmsis_packs/XMC4000/Infineon.XMC4000_DFP.2.14.0.pack)).
4. `cargo run -- -vv path/to/blinky`

I captured some [maximally verbose log output](https://gist.github.com/willglynn/2fdc8fce3542dff77fa5e0cb79cbbf2f).

**Expected behavior**

Flashing should succeed.

**Additional context**

I've learned some things while trying to work this issue:

* The XMC4xxx chips have flash mapped into the address space twice: one for cached accesses, once for uncached accesses. On this device it's cached at 0x08000000 and uncached at 0x0c000000.
* The [vendor's linker script](https://github.com/aws/amazon-freertos/blob/main/vendors/infineon/XMCLib/2.1.20/CMSIS/Infineon/XMC4800_series/Source/GCC/XMC4800x2048.ld#L69-L77) defines both regions and does `.text {…} > FLASH_1_cached AT > FLASH_1_uncached`. In other words, for things which end up in flash, the [VMA](https://sourceware.org/binutils/docs/ld/Output-Section-LMA.html) (seen by the code) is in the cached region while the [LMA](https://sourceware.org/binutils/docs/ld/Output-Section-LMA.html) (seen by the flashing program) is in the uncached region.
* The CMSIS pack contains a whole pile of `*.FLM`s which distinguish these two regions:
```console
% strings < XMC4800_2048.FLM | grep XMC | head -1 
XMC4800 2048kB Flash
% strings < XMC4800c_2048.FLM | grep XMC | head -1
XMC4800 2048kB cached Flash
```
* The `target-gen`'d YAML contains entries for each of the ELF files in the pack:
```yaml
  - name: xmc4800_2048
    description: XMC4800 2048kB Flash
    default: true
…
    flash_properties:
      address_range:
        start: 201326592
        end: 203423744
      page_size: 256
…
…
  - name: xmc4800c_2048
    description: XMC4800 2048kB cached Flash
    default: true
…
    flash_properties:
      address_range:
        start: 134217728
        end: 136314880
      page_size: 256
```
* The `target-gen`'d flash algorithm `xmc4800_2048` and `xmc4800_2048c` live components – the `instructions:` and function pointers and such – are completely identical.
* The `target-gen`'d XMC4800-F100x1024 chip definition refers to both flash algorithms, but the memory map overlaps only the cached flash algorithm's address range. This makes `probe-run` unable to flash binaries linked using a strategy similar to the vendor's strategy.
```yaml
    memory_map:
      - Ram:
          name: ~
          range:
            start: 536797184
            end: 537001920
          is_boot_memory: false
          cores:
            - main
      - Nvm:
          name: ~
          range:
            start: 134217728
            end: 135266304
          is_boot_memory: true
          cores:
            - main
```

(I _think_ this means that the vendor toolchain will flash via `xmc4800_2048` while `probe-rs` will flash via `xmc4800_2048c`. This might be nothing because they're identical, or this might be a problem if the cached one is broken and no one noticed because everyone else uses the uncached one.)

* `openocd` supports this product family via a [hand-written flash driver](https://github.com/openocd-org/openocd/blob/master/src/flash/nor/xmc4xxx.c) which seems to work.
* All these flash details are likely to be red herrings since the reported issue concerns the contents of the FPB's `FPCOMPn` registers, which is super far removed from the flash controller.
* [XMC4700/4800 errata](https://www.infineon.com/dgdl/Infineon-ErrataSheet_XMC4700_XMC4800_AA-ES-v01_04-EN.pdf?fileId=5546d462525dbac40152cfa214cb7beb) has an entry "DEBUG_CM.002 CoreSight logic only reset after power-on reset" describing how debug resets do not reset the debugging logic. I don't _think_ `probe-rs` is performing debug resets, but if it did, this chip would [respond poorly](https://i.kym-cdn.com/photos/images/facebook/000/913/758/a12.jpg).

That's as far as I got before coming here.

willglynn, 2022-06-23:
I learned something new. Plugging into the board's embedded J-Link debugger and trying to run `cargo-flash` tells me the core is locked up:

```console
% cargo flash --chip XMC4800-F100x2048 
…
    Flashing …/target/thumbv7em-none-eabi/debug/blink
       ERROR probe_rs::architecture::arm::core::armv7m > The core is in locked up status as a result of an unrecoverable exception
     Erasing sectors ✔ [00:00:02] [---------------------------------------]       0B/16.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:03] [---------------------------------------]       0B/10.75KiB @       0B/s (eta 0s )
       Error 0: The initialization of the flash algorithm failed.
             1: An error with the usage of the probe occured
             2: Operation timed out
```

Doing it again here causes the titular "breakpoint must be on half-word [boundary]" error:

```console
% cargo flash --chip XMC4800-F100x2048 
…
    Flashing …/target/thumbv7em-none-eabi/debug/blink
        WARN probe_rs::session > Could not clear all hardware breakpoints: ArchitectureSpecific(Other(Unsupported breakpoint comparator value 0xc80e46a5 for HW breakpoint. Breakpoint must be on half-word boundaries))
       Error Connecting to the chip was unsuccessful.

        Hint A hard reset during attaching might help. This will reset the entire chip. Run with `--connect-under-reset` to enable this feature.
```

Trying a third time from this state with the proposed `--connect-under-reset` doesn't help:

```console
% cargo flash --chip XMC4800-F100x2048 --connect-under-reset
…
    Flashing …/target/thumbv7em-none-eabi/debug/blink
       Error 0: Failed to get information for AP ApAddress { dp: Default, ap: 0 }
```

…and in fact it leaves the chip in a state where it won't even fail in the ways above:

```console
% cargo flash --chip XMC4800-F100x2048 
…
    Flashing …/target/thumbv7em-none-eabi/debug/blink
       Error Connecting to the chip was unsuccessful.

        Hint A hard reset during attaching might help. This will reset the entire chip. Run with `--connect-under-reset` to enable this feature.
```

Power cycling the board at this point lets me restart this whole sequence. However, `openocd` is able to recover without power cycling:

```console
% openocd -f scripts/board/xmc4800-relax.cfg 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link Lite-XMC4200 Rev.1 compiled Oct 27 2015 17:41:01
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : xmc4800.cpu: hardware has 6 breakpoints, 4 watchpoints
Error: xmc4800.cpu -- clearing lockup after double fault
Polling target xmc4800.cpu failed, trying to reexamine
Info : xmc4800.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for xmc4800.cpu on 3333
Info : Listening on port 3333 for gdb connections

% cargo flash --chip XMC4800-F100x2048
    Flashing …/target/thumbv7em-none-eabi/debug/blink
       ERROR probe_rs::architecture::arm::core::armv7m > The core is in locked up status as a result of an unrecoverable exception
     Erasing sectors ✔ [00:00:02] [---------------------------------------]       0B/16.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:02] [---------------------------------------]       0B/10.75KiB @       0B/s (eta 0s )
       Error 0: The initialization of the flash algorithm failed.
             1: An error with the usage of the probe occured
             2: Operation timed out
```

"Clearing lockup after double fault" seems significant.

willglynn, 2022-06-23:
I can now `cargo build` a `blink` ELF binary which `openocd` can flash, `gdb` can debug via `openocd`, and which does in fact blink the LEDs on my [KIT_XMC48_IOT_AWS_WIFI](https://www.infineon.com/cms/en/product/evaluation-boards/kit_xmc48_iot_aws_wifi/). My linker script is suboptimal – it's at running code from uncached flash at 0x0C000000 – but at least it works.

I have been unable to coerce `cargo-flash` or `probe-run` into loading this same binary. They complain that 0x0C000000 does not correspond to any memory region, which is true because `target-gen` generated a flash algorithm for that address didn't generate a corresponding region. Adding the uncached flash to the target as an `Nvm:` region did not help, and neither did changing the linker script to place code in cached flash at 0x08000000. In both cases, `cargo-flash` causes the blinking to momentarily stutter, resume, and then fails several seconds later saying:

```
% cargo flash --chip XMC4800-F100x2048 --elf target/thumbv7em-none-eabi/release/blink
    Flashing target/thumbv7em-none-eabi/release/blink
        WARN probe_rs::architecture::arm::core::armv7m > Core is running, but we expected it to be halted
     Erasing sectors ✔ [00:00:02] [---------------------------------------]       0B/16.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:03] [---------------------------------------]       0B/ 5.50KiB @       0B/s (eta 0s )
       Error 0: The initialization of the flash algorithm failed.
             1: An error with the usage of the probe occured
             2: Operation timed out

%
```

`probe-run` behaves similarly:

```
Error: Error while flashing

Caused by:
    0: Something during the interaction with the core went wrong
    1: A core architecture specific error occured
    2: Failed to write register DRW at address 0x0000000c
    3: An error specific to a probe type occured
    4: USB error while reading from device: Operation timed out
```

The program seems to be running almost the entire time that both tools attempt to overwrite it. "Core is running, but we expected it to be halted" indeed.

willglynn, 2022-06-23:
I've familiarized myself with CoreSight™, the standard CMSIS debug sequences, and the XMC4800 reference manual enough to determine that the Infineon XMC4000 series reset is weird. Snippets from the reference manual:

[image]
[image]
[image]
[image]

The "Warm Reset" sequence accomplishes the same goal as the CMSIS [ResetCatchSet](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#resetCatchSet) + [ResetSystem](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#resetSystem) but bears little resemblance to the usual implementations. The Infineon CMSIS pack does not include alternate sequences, and at the moment I have no idea what the official tools do. I'm pretty sure this means adding a `probe_rs::architecture::arm::sequences::infineon` module, so I've started on that.

willglynn, 2022-06-24:
All roads from the official tools seem to converge at SEGGER J-Link. [DAVE](https://en.wikipedia.org/wiki/DAvE_(Infineon)) (the official IDE) [packages J-Link](https://wiki.segger.com/Infineon_DAVE_4). The Infineon XMC PlatformIO support stuff [shells out to J-Link](https://github.com/Infineon/platformio-infineonxmc/blob/master/platform.py#L54-L56). The third-party XMC for Arduino [requres J-Link](https://github.com/Infineon/XMC-for-Arduino#prework-for-segger-j-link).

J-Link works with this chip, and can reset/go/halt/flash:

```console
% JLinkExe -device XMC4800-2048 -if SWD -speed 4000 -autoconnect 1
SEGGER J-Link Commander V7.66d (Compiled Jun 21 2022 17:01:42)
DLL version V7.66d, compiled Jun 21 2022 17:01:36

Connecting to J-Link via USB...O.K.
Firmware: J-Link Lite-XMC4200 Rev.1 compiled Mar  1 2019 11:28:26
Hardware version: V1.00
S/N: 591137765
VTref=3.300V
Device "XMC4800-2048" selected.

Connecting to target via SWD
Performing XMC4500 connection sequence.
DPv0 detected
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
Cortex-M4 identified.
J-Link>
```

The output above indicates that J-Link has a specific "XMC4500 connection sequence" which supports my determination that this product family is weird.

The [SEGGER wiki has some information](https://wiki.segger.com/XMC4xxx_series) beyond the reference manual, including that Infineon requested that they perform a reset under certain circumstances. Experimentally, J-Link situationally performs a power-on reset. (Recall that power on resets do not reset debugging state due to the [DEBUG_CM.002 erratum](https://www.infineon.com/dgdl/Infineon-ErrataSheet_XMC4700_XMC4800_AA-ES-v01_04-EN.pdf?fileId=5546d462525dbac40152cfa214cb7beb).) I don't know exactly which situations those are, other than that I can occasionally create them if I thrash around and put the chip into weird states.

My best guess is that this power on reset is to set SCU_GENERAL->STCON's HWCON field, since HWCON a) is latched after power on reset, b) reflects which are normally configured at the board level with pullups/pulldowns but can be overridden with a debug probe, and c) configures the boot mode.

```
J-Link>Mem32 0x50004010 1
50004010 = 00000000 
```

0x50004010 & 3 == 0 corresponds to normal boot mode, which IMO is the sane and reasonable way to implement the "halt after warm reset" process. It's likely that my effort to create weird states was sometimes doing a power on reset while sending SWD commands, which could cause HWCON to latch nonzero, which J-Link would want to do a power-on reset to clear.

willglynn, 2022-06-24:
🥳 

```console
% RUST_LOG=probe_rs::probe::jlink::swd=info,probe_rs::architecture::arm::ap=info cargo-flash --log debug --chip XMC4800-F100x2048 --elf …/blink --speed 1000
       DEBUG cargo_flash > Changed working directory to .
    Flashing …/blink
       DEBUG probe_rs::config::registry > Searching registry for chip with name XMC4800-F100x2048
       DEBUG probe_rs::config::registry > Exact match for chip name: XMC4800-F100x2048
        WARN probe_rs::config::target   > Using custom sequence for XMC4000
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.24.11584
       DEBUG jaylink                          > libusb has capability API: true
       DEBUG jaylink                          > libusb has HID access: false
…
       DEBUG probe_rs::flashing::flasher                          > Reset and halt
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Set a breakpoint at 0c0001fe
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Cleared SCU_RESET->RSTSTAT
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Resetting via AIRCR.SYSRESETREQ
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Detected reset via S_RESET_ST
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Waiting for XMC4000 to halt after reset
       DEBUG probe_rs::architecture::arm::sequences::infineon     > Halted after reset
       DEBUG probe_rs::architecture::arm::core::armv7m            > Reason for halt has changed, old reason was Halted(Request), new reason is Breakpoint
       DEBUG probe_rs::flashing::flasher                          > Loading algorithm into RAM at address 0x1ffe8000
…
       DEBUG probe_rs::flashing::flasher                                  > Waiting for routine call completion.
       DEBUG probe_rs::architecture::arm::core                            > DFSR: Dfsr { .0: 3, external: false, vcatch: false, dwttrap: false, bkpt: true, halted: true }
     Erasing sectors ✔ [00:00:01] [#######################################] 16.00KiB/16.00KiB @ 23.79KiB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [#######################################]  5.50KiB/ 5.50KiB @  2.14KiB/s (eta 0s )
    Finished in 2.141s
%
```

I pushed this to [my `xmc4000` branch](https://github.com/probe-rs/probe-rs/compare/master...willglynn:xmc4000?expand=1), which is not yet ready for a PR but can (!) flash this chip.

willglynn, 2022-06-24:
I'm pleased to report that `gdb-server` also works in this branch:
[image]

Yatekii, 2022-06-25:
Awesome work! I love the persistence :) Sorry for not helping with any hints :(

closed 2022-12-09 by willglynn