probe-rs/probe-rs #1135: ORBTrace Mini not detected due to CMSIS-DAP string only in interface descriptor

kc8apf, 2022-06-16:
**Describe the bug**
ORBTrace Mini (https://orbcode.org/orbtrace-mini/) implements both CMSIS-DAP v1 and v2.  probe-rs does not find the probe.  https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/tools.rs#L65 is looking for the string "CMSIS-DAP" in the product string but ORBTrace Mini is a composite device so "CMSIS-DAP" is only present in the interface description.  Per https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__ConfigUSB__gr.html#details, ORBTrace Mini is following the spec.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in ORBTrace Mini
2. probe-rs-cli list

**Expected behavior**
ORBTrace Mini detected and listed as a probe.

**Desktop (please complete the following information):**
Ubuntu 21.04

**Additional context**

kc8apf, 2022-06-16:
```
sudo lsusb -d 1209:3443 -v

Bus 001 Device 013: ID 1209:3443 Generic Orbtrace
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1209 Generic
  idProduct          0x3443
  bcdDevice            0.00
  iManufacturer           1 Orbcode
  iProduct                2 Orbtrace
  iSerial                 3 35886192A2189006
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0098
    bNumInterfaces          7
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              11
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 103 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        5
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     80
      bInterfaceProtocol      0
      iInterface              7 Target power
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        6
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     86
      bInterfaceProtocol      0
      iInterface              8 Version: v1.0.0-0-g3ad3fa4
      ```

adamgreig, 2022-06-21:
(cc https://github.com/probe-rs/probe-rs/issues/995)

The main problem with fixing this is that hidapi doesn't expose interface strings, only top-level product/vendor/serial strings, making it hard to check properly. rusb does expose these strings but we can't rely on that always working. It seems in https://github.com/probe-rs/probe-rs/issues/995#issuecomment-1032525318 that we might be able to use the [path](https://docs.rs/hidapi/latest/hidapi/struct.DeviceInfo.html#method.path) method which contains interface strings, which is worth looking in to...

adamgreig, 2022-06-21:
It also looks like pyOCD whitelists a bunch of device names and VID/PID pairs that are known to be cmsis-dap devices, which I guess we could also do... https://github.com/pyocd/pyOCD/pull/1399/files

commit 1e8fc057 (2023-01-13): Merge #1142

1142: CMSIS-DAP: Check for CMSIS-DAP string in interface strings too r=Yatekii a=adamgreig

It turns out it's legitimate for CMSIS-DAP devices to just put that in an interface string, even though most put it in the product string too (as required for non-composite devices). This PR tries to check interface strings too, though this requires rusb and permission to access the device. Apparently on some platforms hidapi will report the interface string in the path, but not on mine, so it's included here as a talisman for good luck...

It's a bit hard to read the diff because mostly it [... cut]
PR #1142 (2023-01-13): CMSIS-DAP: Check for CMSIS-DAP string in interface strings too
It turns out it's legitimate for CMSIS-DAP devices to just put that in an interface string, even though most put it in the product string too (as required for non-composite devices). This PR tries to check interface strings too, though this requires rusb and permission to access the device. Apparently on some platforms hidapi will report the interface string in the path, but not on mine, so it's included here as a talisman for good luck...

It's a bit hard to read the diff because mostly it changes indentation on existing code, but [... cut]
closed 2023-01-13 by bors[bot]