probe-rs/probe-rs #1137: ESP32C3 Error No flash memory contains the entire requested memory range

jeancf, 2022-06-16:
**Describe the bug**
Using `cargo flash` to upload binary on Ai-Thinker NodeMCU Series ESP32-C3S Kit (ESP32C3-based MCU) through the built-in JTAG USB interface fails with `Error No flash memory contains the entire requested memory range 0x42000020..0x4200de82.`

**To Reproduce**
Steps to reproduce the behavior:
1. Install rust nightly toolchain and probe-rs flash crate
2. Clone https://github.com/esp-rs/esp-hal repository
3. Connect USB wires to GPIO18 and GPIO19 on the board
4. From the root of the repo execute cargo flash --example blinky --chip esp32c3
5. result:
```
   Compiling esp32c3-hal v0.1.0 (/home/jeancf/Documents/esp32-c3/esp-hal/esp32c3-hal)
    Finished dev [unoptimized + debuginfo] target(s) in 1.68s
    Flashing /home/jeancf/Documents/esp32-c3/esp-hal/esp32c3-hal/target/riscv32imc-unknown-none-elf/debug/examples/blinky
        WARN probe_rs::config::target > Using custom sequence for ESP32c3
        WARN probe_rs::architecture::riscv > Breakpoint 0: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 1: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 2: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 3: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 4: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 5: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 6: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 7: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 0: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 1: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 2: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 3: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 4: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 5: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 6: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv > Breakpoint 7: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
       Error No flash memory contains the entire requested memory range 0x42000020..0x4200de82.

        Hint Make sure the flash region specified in the linkerscript matches the one specified in the datasheet of your chip.
```

**Expected behavior**
Using Espressif rust tool works fine:
```
❯ cargo espflash --example=blinky
Serial port: /dev/ttyACM0
Connecting...

   Compiling esp32c3-hal v0.1.0 (/home/jeancf/Documents/esp32-c3/esp-hal/esp32c3-hal)
    Finished dev [unoptimized + debuginfo] target(s) in 4.38s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        2MB
Features:          WiFi
MAC address:       [REDACTED]
[00:00:00] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:00] ########################################      28/28      segment 0x10000

Flashing has completed!
```

**Desktop (please complete the following information):**
- Manjaro Linux 5.17

**Additional context**
I tried with another ESP32C3-based device, the Lilygo TTGO T-OI PLUS with the exact same result.

Ardelean-Calin, 2022-06-21:
I can confirm I get the same error with an ESP32-C3 with a 4MB flash size. I am willing to help in any way I can to fix this issue.

Ardelean-Calin, 2022-06-21:
Unfortunately it still doesn't work for me... 

However I tested using probe-rs as a library to halt the esp32c3, as well as reset it, and when at least those two commands functioned correctly. So at least that is a confirmation that the JTAG connection somewhat works, right?

I am currently trying to attach to a running program on the esp32c3 and set some breakpoints, so that I can be sure that at least debugging works. Will get back when I have progress.

Ardelean-Calin, 2022-06-21:
**Update**
It seems flashing works with the direct-boot feature **enabled**. By default, when building/flashing an example, that feature is **disabled**.

The command that works for me:
`cargo flash --example blinky --features direct-boot --chip esp32c3`

jeancf, 2022-06-21:
@noppej I originally built with `cargo install cargo-flash`. Installing from github gives the same error. I could not try using probe-rs-cli.

However I can confirm that `cargo flash --example blinky --features direct-boot --chip esp32c3` works for me as well, albeit very slowly (transfer at 82B/s).

Thanks @Ardelean-Calin !

bugadani, 2024-02-25:
I fixed a similar issue in `cargo embed` a short while ago (#2104), and now `cargo embed`, `cargo flash` and `probe-rs` have been using the same code to flash. I'm closing this issue as I assume this to be fixed, but let us know if that's not the case!

closed 2024-02-25 by bugadani