probe-rs/probe-rs #1140: STM32H723ZGTx board support

imrank03, 2022-06-20:
Hi, I am using `stm32h723zg` board I'm trying to erase or flash bin file using `probe-rs-cli` which shows that the chip was not found, but the chip was found in the `probe-run` list, below are the stack-traces.

I have installed the new version but I am still getting the same error, how can I fix this error.

**Stacktrace probe-rs-cli**
```
❯ probe-rs-cli erase --chip STM32H723ZGTx
Error: The chip 'STM32H723ZGTx' was not found in the database.

Caused by:
    The requested chip 'STM32H723ZGTx' was not found in the list of known targets.
```

**Stacktrace probe-run**
```
❯ probe-run --list-chips
...
...
STM32H7 Series
    Variants:
        STM32H723VEHx
        STM32H723VETx
        STM32H723VGHx
        STM32H723VGTx
        STM32H723ZEIx
        STM32H723ZETx
        STM32H723ZGIx
        STM32H723ZGTx
        STM32H725AEIx
        STM32H725AGIx
        STM32H725IEKx
        STM32H725IETx
        STM32H725IGKx
        STM32H725IGTx
        STM32H725REVx
        STM32H725RGVx
        STM32H725VEHx
        STM32H725VETx
        STM32H725VGHx
        STM32H725VGTx
        STM32H725VGYx
        STM32H725ZETx
        STM32H725ZGTx
        STM32H730ABIxQ
        STM32H730IBKxQ
        STM32H730IBTxQ
        STM32H730VBHx
        STM32H730VBTx
        STM32H730ZBIx
        STM32H730ZBTx
        STM32H733VGHx
        STM32H733VGTx
        STM32H733ZGIx
        STM32H733ZGTx
        STM32H735AGIx
        STM32H735IGKx
        STM32H735IGTx
        STM32H735RGVx
        STM32H735VGHx
        STM32H735VGTx
        STM32H735VGYx
        STM32H735ZGTx
        STM32H742AGIx
        STM32H742AIIx
        STM32H742BGTx
        STM32H742BITx
        STM32H742IGKx
        STM32H742IGTx
        STM32H742IIKx
        STM32H742IITx
        STM32H742VGHx
        STM32H742VGTx
        STM32H742VIHx
        STM32H742VITx
        STM32H742XGHx
        STM32H742XIHx
        STM32H742ZGTx
        STM32H742ZITx
        STM32H743AGIx
        STM32H743AIIx
        STM32H743BGTx
        STM32H743BITx
        STM32H743IGKx
        STM32H743IGTx
        STM32H743IIKx
        STM32H743IITx
        STM32H743VGHx
        STM32H743VGTx
        STM32H743VIHx
        STM32H743VITx
        STM32H743XGHx
        STM32H743XIHx
        STM32H743ZGTx
        STM32H743ZITx
        STM32H745BGTx
        STM32H745BITx
        STM32H745IGKx
        STM32H745IGTx
        STM32H745IIKx
        STM32H745IITx
        STM32H745XGHx
        STM32H745XIHx
        STM32H745ZGTx
        STM32H745ZITx
        STM32H747AGIx
        STM32H747AIIx
        STM32H747BGTx
        STM32H747BITx
        STM32H747IGTx
        STM32H747IITx
        STM32H747XGHx
        STM32H747XIHx
        STM32H747ZIYx
        STM32H750IBKx
        STM32H750IBTx
        STM32H750VBTx
        STM32H750XBHx
        STM32H750ZBTx
        STM32H753AIIx
        STM32H753BITx
        STM32H753IIKx
        STM32H753IITx
        STM32H753VIHx
        STM32H753VITx
        STM32H753XIHx
        STM32H753ZITx
        STM32H755BITx
        STM32H755IIKx
        STM32H755IITx
        STM32H755XIHx
        STM32H755ZITx
        STM32H757AIIx
        STM32H757BITx
        STM32H757IITx
        STM32H757XIHx
        STM32H757ZIYx
        STM32H7A3AGIxQ
        STM32H7A3AIIxQ
        STM32H7A3IGKx
        STM32H7A3IGKxQ
        STM32H7A3IGTx
        STM32H7A3IGTxQ
        STM32H7A3IIKx
        STM32H7A3IIKxQ
        STM32H7A3IITx
        STM32H7A3IITxQ
        STM32H7A3LGHxQ
        STM32H7A3LIHxQ
        STM32H7A3NGHx
        STM32H7A3NIHx
        STM32H7A3QIYxQ
        STM32H7A3RGTx
        STM32H7A3RITx
        STM32H7A3VGHx
        STM32H7A3VGHxQ
        STM32H7A3VGTx
        STM32H7A3VGTxQ
        STM32H7A3VIHx
        STM32H7A3VIHxQ
        STM32H7A3VITx
        STM32H7A3VITxQ
        STM32H7A3ZGTx
        STM32H7A3ZGTxQ
        STM32H7A3ZITx
        STM32H7A3ZITxQ
        STM32H7B0ABIxQ
        STM32H7B0IBKxQ
        STM32H7B0IBTx
        STM32H7B0RBTx
        STM32H7B0VBTx
        STM32H7B0ZBTx
        STM32H7B3AIIxQ
        STM32H7B3IIKx
        STM32H7B3IIKxQ
        STM32H7B3IITx
        STM32H7B3IITxQ
        STM32H7B3LIHxQ
        STM32H7B3NIHx
        STM32H7B3QIYxQ
        STM32H7B3RITx
        STM32H7B3VIHx
        STM32H7B3VIHxQ
        STM32H7B3VITx
        STM32H7B3VITxQ
        STM32H7B3ZITx
        STM32H7B3ZITxQ
```

noppej, 2022-06-20:
Hi Imran,

The command you are using to `list-chips`, provides a list of all supported chips, and is no indication of what is connected to your computer.

Try `probe-rs-cli list` ... this will list the actual probes that are connected. Your probe should show up, and then you can try to force the erase to that chip, by using the `--probe` option (see `probe-rs-cli erase --help` output for the format of that)

imrank03, 2022-06-20:
> Hi Imran,
>
> The command you are using to `list-chips`, provides a list of all supported chips, and is no indication of what is connected to your computer.
>
> Try `probe-rs-cli list` ... this will list [...]

Hello @noppej,

I tried what you suggested, but I'm still getting the error. Below is the command line output.

```
❯ probe-rs-cli list                                            
The following devices were found:
[0]: STLink V3 (VID: 0483, PID: 374e, Serial: 002800173156501520323443, StLink)

❯ probe-rs-cli erase --probe 0483:374e                         
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: Unable to load specification for chip
    1: The connected chip could not automatically be determined.

❯ probe-rs-cli erase --probe 0483:374e:002800173156501520323443
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: Unable to load specification for chip
    1: The connected chip could not automatically be determined.
```

noppej, 2022-06-20:
I am sorry, but I wasn't specific enough in my instructions. To help debug this, can you please try the following, and let me know if you get the same error:

`probe-rs-cli erase --chip STM32H723ZGTx --probe 0483:374e`

imrank03, 2022-06-20:
> I am sorry, but I wasn't specific enough in my instructions. To help debug this, can you please try the following, and let me know if you get the same error:
> 
> `probe-rs-cli erase --chip STM32H723ZGTx --probe 0483:374e`

I tried, I still have the same error.
```
❯ probe-rs-cli erase --chip STM32H723ZGTx --probe 0483:374e
Error: The chip 'STM32H723ZGTx' was not found in the database.

Caused by:
    The requested chip 'STM32H723ZGTx' was not found in the list of known targets.
```

- I got the same error when I was using `prob-run` and then @burrbull  suggested this command `cargo install probe-run --git https://github.com/burrbull/probe-run --branch h7 --force` to install [probe-run](https://github.com/knurling-rs/probe-run/issues/309), after installing this branch I can see the chip `stm32h723zgtx` in the `probe-run` list, but I am unable to erase the chip using `probe-rs-cli`.
- I think the chip `stm32h723zg` is not included in the latest version of `probe-rs 0.12.0`.

noppej, 2022-06-20:
I can see that chip in the target list of the master branch of probe-rs. Similar to what @burrbull suggested, perhaps you can try to get the lastest probe-rs-cli, by doing ...
`cargo install --force --git https://github.com/probe-rs/probe-rs probe-rs-cli`

imrank03, 2022-06-21:
@noppej Thanks for the support, it's working now.

```
❯ probe-rs-cli erase --chip stm32h723zgtx
❯ probe-rs-cli download --format Bin --base-address 0x8020000 --chip stm32h723zgtx stm32h723_bootfw_v1234_signed.bin
Erasing sectors ✔ [00:00:00] [##################################################################################################]  3.00KiB/ 3.00KiB @  7.30KiB/s (eta 0s )
Programming pages   ✔ [00:00:00] [##################################################################################################]  3.00KiB/ 3.00KiB @  3.85KiB/s (eta 0s )
Finished in 0.611s
❯ probe-rs-cli download --format Bin --base-address 0x8060000 --chip stm32h723zgtx stm32h723_updtfw_v1235_signed.bin
Erasing sectors ✔ [00:00:00] [##################################################################################################]  3.00KiB/ 3.00KiB @  7.17KiB/s (eta 0s )
Programming pages   ✔ [00:00:00] [##################################################################################################]  3.00KiB/ 3.00KiB @  3.77KiB/s (eta 0s )
Finished in 0.613s
❯ cargo flash --chip stm32h723zgtx --release
Finished release [optimized] target(s) in 0.04s
Flashing /home/imran/Desktop/rustBoot/boards/target/thumbv7em-none-eabihf/release/stm32h723
Erasing sectors ✔ [00:00:02] [##################################################################################################] 47.00KiB/47.00KiB @ 17.60KiB/s (eta 0s )
Programming pages   ✔ [00:00:03] [##################################################################################################] 47.00KiB/47.00KiB @  7.60KiB/s (eta 0s )
Finished in 5.773s
``

closed 2022-06-21 by Yatekii