probe-rs/probe-rs #1157: Segger J-Link fails to connect to RP2040

jebiga52, 2022-07-08:
**Describe the bug**
I have a custom board with an RP2040 on it. I can connect to the RP2040 using probe-rs-cli with an Raspberry Pi Pico running DapperMime firmware. I can also connect to the board with my Segger J-Link Base if I use Segger's JLinkExe program (all else remains constant, including the computer it's plugged in to etc). However, if I try to connect to the board using probe-rs-cli, it fails to connect.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect RP2040 to Segger J-Link.
2. Try to connect to the device.
3. probe-rs-cli fails with the following error:

```
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

**Expected behavior**
probe-rs-cli should be able to connect and debug RP2040s when using a Segger J-Link.

**Stacktrace**
This board can be flashed when using DapperMime firmware:
```
$ RUST_LOG=probe_rs=trace probe-rs-cli erase --chip rp2040
 DEBUG probe_rs::config::registry > Searching registry for chip with name rp2040
 DEBUG probe_rs::config::registry > Exact match for chip name: RP2040
 DEBUG probe_rs::config::registry > Partial match for chip name: RP2040_SELFDEBUG
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 TRACE probe_rs::probe::cmsisdap::tools > RP2040 CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 0, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 1 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 2 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > No HID interface for CMSIS-DAP found.
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
 TRACE probe_rs::probe::cmsisdap::tools > CMSIS-DAP device with USB path: "/dev/hidraw2"
 TRACE probe_rs::probe::cmsisdap::tools >                 product_string: "RP2040 CMSIS-DAP"
 TRACE probe_rs::probe::cmsisdap::tools >                      interface: 0
 TRACE probe_rs::probe::cmsisdap::tools > Ignoring duplicate RP2040 CMSIS-DAP (VID: cafe, PID: 4005, Serial: 6E1646703EA2AA22, CmsisDap)
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
 TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching cafe:4005:6E1646703EA2AA22
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 002: ID cafe:4005
 TRACE probe_rs::probe::cmsisdap::tools > RP2040 CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 0, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 1 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 2 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > No HID interface for CMSIS-DAP found.
 DEBUG probe_rs::probe::cmsisdap::tools > Could not open cafe:4005 in CMSIS-DAP v2 mode
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open cafe:4005 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::cmsisdap::commands > Draining probe of any pending data.
 DEBUG probe_rs::probe::cmsisdap::commands > Attempt 1 to find packet size
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FF, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 02, 40, 00]...
 DEBUG probe_rs::probe::cmsisdap::commands > Success: packet size is 64
 DEBUG probe_rs::probe::cmsisdap::commands > Configuring probe to use packet size 64
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, FE, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 01, 01, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 00, F0, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [00, 01, 11, 00]...
 DEBUG probe_rs::probe::cmsisdap           > Detected probe capabilities: Capabilities { swd_implemented: true, jtag_implemented: false, swo_uart_implemented: false, swo_manchester_implemented: false, atomic_commands_implemented: true, test_domain_timer_implemented: false, swo_streaming_trace_implemented: false, uart_communication_port_implemented: false, uart_com_port_implemented: false }
 DEBUG probe_rs::probe::cmsisdap           > Attaching to target system (clock = 1000kHz)
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 02, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [02, 01, 11, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 11, 40, 42, 0F, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [11, 00, 11, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 04, 00, 50, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [04, 00, 11, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 13, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [13, 00, 11, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 01, 00, 01, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Receive buffer: [01, 00, 11, 00]...
 TRACE probe_rs::probe::cmsisdap::commands > Transmit buffer: [00, 12, 33, FF, FF, FF, FF, FF, FF, 07, 00]...
[... 105 lines cut ...]
 DEBUG probe_rs::probe::jlink                 > Attaching with protocol 'SWD'
 INFO  probe_rs::probe::jlink                 > J-Link: S/N: 50102718
 DEBUG probe_rs::probe::jlink                 > J-Link: Capabilities: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
 INFO  probe_rs::probe::jlink                 > J-Link: Firmware version: J-Link V10 compiled Nov  2 2021 12:14:50
 INFO  probe_rs::probe::jlink                 > J-Link: Hardware version: J-Link 10.10.0
 INFO  probe_rs::probe::jlink                 > J-Link: Target voltage: 3.31 V
 DEBUG probe_rs::probe::jlink                 > Attached succesfully
 DEBUG probe_rs::probe::jlink::swd            > Performing 1 transfers (0 additional transfers)
 TRACE probe_rs::probe::jlink::swd            > DAP read 197207159.
 DEBUG probe_rs::probe::jlink::swd            > Transfer result 0: Ok(bc12477)
 DEBUG probe_rs::architecture::arm::communication_interface > Selecting DP Multidrop(1002927)
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

**Desktop (please complete the following information):**
Reproducible on both macOS and Linux.

**Additional context**
```
$ probe-rs-cli --version        
Probe-rs CLI 0.12.0
```

adamgreig, 2022-07-09:
Unfortunately multidrop SWD isn't supported by probe-rs for jlink probes (I believe it would need implementing in the [jaylink](https://github.com/jonas-schievink/jaylink) crate first, and then adding to the probe-rs jaylink driver). The error message is confusing because the [real error](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/jlink/arm.rs#L934) is being masked by this [higher-level error](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/architecture/arm/communication_interface.rs#L404), but we have an open issue about that (https://github.com/probe-rs/probe-rs/issues/997).

In short, it's not currently possible to use a j-link probe to talk to the rp2040, since that requires multidrop SWD which isn't supported.

jebiga52, 2022-07-09:
I see, thank you for the clarification. If this is a duplicate of #997 please feel free to close it.

jebiga52, 2022-12-16:
Turns out the jaylink crate actually does support this, see the maintainer's comment here: https://github.com/jonas-schievink/jaylink/issues/39#issuecomment-1181186669

Perhaps we should re-open this issue.

jounathaen, 2022-12-16:
In my PR, I tried to implement it, but it doesn't work, and I don't know why. My logic analyzer shows the waveforms I'd expect, but there must be some bug somewhere left. But I think it can be definitely be done with jaylink.

jebiga52, 2022-12-16:
I would be open to placing a bounty on this bug. @jounathaen would you be interested? We can come to an agreement privately over email.

jounathaen, 2022-12-16:
That's a nice idea, but it's not the money that is holding me back ;-)
But if somebody else can finalize it faster than me, I'm not opposing a bounty for them.

jebiga52, 2022-12-16:
Understood. If you do change your mind please let me know. I would love to help but unfortunately my knowledge of SWD is minimal at best.

jounathaen, 2022-12-16:
Actually, this very part is not \*that\* difficult. If you are curious: It is described in https://developer.arm.com/documentation/ihi0074/latest/ in sections B4.3.4 & B5.3.4.

Basically one has to send a special long but fixed sequence via the interface to activate the cores, do a 50 cycle 0-1 sequence and then do a read on the `DPIDR` register. But as always in embedded systems: It's about the details and Murphy is your worst enemy.

PR #2079 (2024-01-22): J-Link multidrop support
Closes #1272
Closes #1157 
Closes #997

The original PR was pretty close, but missed a few tricky parts:
 - A line reset in `raw_read_register` could immediately deselect the target. This was the root cause of incorrect DPIDR values.
 - IHI0074D contains the following line: `If overrun detection is enabled, then the line reset sets CTRL/STAT.STICKYORUN to 0b1.`. This means we need to clear the sticky overflow bit after target selection, because the selection process starts with a line reset.

Note that I only tested that probe-rs can reset a Raspberry Pico W, I haven't [... cut]
closed 2024-01-22 by Dirbaio