probe-rs/probe-rs #1162: Attempting to flash nRF52 after double-fault fails

TilBlechschmidt, 2022-07-10:
**Describe the bug**
When a double-fault occurs on the nRF52, it locks up. In such a scenario, probe-rs is unable to resolve the state and successfully flash the chip:

```
(HOST) INFO  flashing program (4 pages / 16.00 KiB)
Error: Error while flashing

Caused by:
    0: Something during the interaction with the core went wrong
    1: A core architecture specific error occured
    2: Failed to write register DRW at address 0x0000000c
    3: An error specific to a probe type occured
    4: USB error while reading from device: Operation timed out
```

This situation can be remedied by either power-cycling the nRF52 and J-Tag, or starting openocd which will clear the fault:

```bash
> openocd -c 'interface jlink; transport select swd; source [find target/nrf52.cfg]'
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link ARM-OB STM32 compiled Jun 30 2009 11:14:15
Info : Hardware version: 7.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Error: nrf52.cpu -- clearing lockup after double fault # <-- this is the relevant action which probe-rs does not do
Polling target nrf52.cpu failed, trying to reexamine
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked

# Now cargo run will succeed
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a double-fault (throw a hard fault while handling a hard fault). For me this usually occurs when the panic handler tries writing via RTT to the host but the panic occurs by the JTag disconnecting in the first place which makes the RTT channel unavailable.
2. Attempt to flash/run the program with probe-rs
3. It fails with the above error

**Expected behavior**
It should either give a descriptive warning saying "hey, your processor crashed while it was crashing. You should check this out and make sure this is intentional or fix it", or it should just resolve the lockup by its own. A mixture of both is probably sensible. During prototyping it can happen that the panic handler panics and having to resort to openocd for this is mildly annoying.

**Stacktrace**
See at the top.

**Desktop:**
 - macOS Ventura 13.0 Beta 2 (but happened on non-beta OS as well)

noppej, 2022-07-11:
Does NRF52 support the `connect_under_reset` option? Did you try that?

TilBlechschmidt, 2022-07-11:
I have not, will try to reproduce and attempt to flash with that flag in a couple of hours 🙂

diondokter, 2022-07-12:
I think this issue may be linked with #1012 where I've also added some info about something I ran into

Tiwalun, 2023-01-28:
I tried to reproduce this with a nRF52-DK, but was not able to. I tried to manually cause a lockup, and then program the chip afterwards, and that worked without problems.

This is probably another error, and not related to the double fault / lockup. Do you have an example where this occurs?