probe-rs/probe-rs #1178: "Probe was not found" even though probe is detected by `probe-rs-cli list`

doesnotcompete, 2022-07-19:
**Describe the bug**
I am attempting to use a Raspberry Pi Pico with the (DapperMime)[https://github.com/majbthrd/DapperMime/] firmware to connect to a second Pico with `probe-rs`. The same issue occurs with the similar `pico-debug` project.
`probe-rs-cli list` detects the probe, but trying to actually connect to the probe results in a "Failed to open the debug probe." error.
The specified cause "Probe was not found." seems misleading, as the probe was previously detected by the `list` command.
The probe is detected as `RP2040 CMSIS-DAP (VID: cafe, PID: 4005, Serial: 6E060C2631C41153, CmsisDap)` by `lsusb`, and I have configured `udev` rules as documented on the probe-rs website.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the Pico CMSIS-DAP probe.
2. Run `probe-rs-cli list` to ensure the probe is detected.
3. Running `probe-rs-cli debug` fails.

**Expected behavior**
I am having difficulties troubleshooting this problem, as the error cause reported by `probe-rs-cli` does not seem to capture the actual underlying issue. I'd expect either a different error cause than "Probe was not found", or the probe to not be detected by the `list` command either.

**Stacktrace**
This is the probe in `sudo lsusb -vvv`:
```
Bus 002 Device 029: ID cafe:4005 TinyUSB RP2040 CMSIS-DAP
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0xcafe
  idProduct          0x4005
  bcdDevice            1.02
  iManufacturer           1 TinyUSB
  iProduct                2 RP2040 CMSIS-DAP
  iSerial                 3 6E060C2631B40253
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         1
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

When trying to deploy some firmware with `probe-run` I can get this output:
```
probe-run --chip RP2040 --verbose target/thumbv6m-none-eabi/debug/app
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 2003fb90, hard_fault: 100061c5 }
└─ probe_run::elf @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/elf.rs:29
(HOST) DEBUG RAM region: 0x20000000-0x2003FFFF
└─ probe_run::target_info @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/target_info.rs:54
(HOST) DEBUG section `.data` is in RAM at 0x2003FB90..=0x2003FBBF
└─ probe_run::target_info @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/target_info.rs:91
(HOST) DEBUG section `.bss` is in RAM at 0x2003FBC0..=0x2003FBD3
└─ probe_run::target_info @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/target_info.rs:91
(HOST) DEBUG section `.uninit` is in RAM at 0x2003FBD4..=0x2003FFFB
└─ probe_run::target_info @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/target_info.rs:91
(HOST) DEBUG valid SP range: 0x20000000..=0x2003FB8C
└─ probe_run::target_info @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/target_info.rs:105
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ /home/kevin/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.3/src/probe.rs:25
Error: Probe could not be created

Caused by:
    Probe was not found.
```

**Desktop (please complete the following information):**
 - Gentoo (systemd, glibc2.34, Linux 5.15.52, x86_64)

adamgreig, 2022-07-19:
Could you try running probe-rs-cli with RUST_LOG=trace to get better log messages?

doesnotcompete, 2022-07-20:
Thanks for the suggestion, I am still quite new to Rust and was not aware this flag existed. This is the output of `RUST_LOG=trace probe-rs-cli debug`:
```
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 TRACE probe_rs::probe::cmsisdap::tools > RP2040 CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 0, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 1 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 2 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > No HID interface for CMSIS-DAP found.
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
 DEBUG jaylink                          > libusb 1.0.26.11724
 DEBUG jaylink                          > libusb has capability API: true
 DEBUG jaylink                          > libusb has HID access: true
 DEBUG jaylink                          > libusb has hotplug support: true
 DEBUG jaylink                          > libusb can detach kernel driver: true
 TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching cafe:4005:6E060C2631B40253
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 005 Device 001: ID 1d6b:0003
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 006: ID 06cb:00bd
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 005: ID 058f:9540
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 004: ID 04f2:b681
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 003: ID 05e3:0610
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 002: ID 8087:0025
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 004 Device 001: ID 1d6b:0002
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 003: ID 17ef:1020
 TRACE probe_rs::probe::cmsisdap::tools > Error getting languages: Pipe
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 004: ID 17ef:3062
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 002: ID 17ef:101f
 TRACE probe_rs::probe::cmsisdap::tools > Error getting languages: Pipe
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 001: ID 1d6b:0003
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 013: ID cafe:4005
 TRACE probe_rs::probe::cmsisdap::tools > RP2040 CMSIS-DAP: CMSIS-DAP device with 3 interfaces
 TRACE probe_rs::probe::cmsisdap::tools > Could not read string for interface 0, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 1 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > Interface 2 is not HID, skipping
 TRACE probe_rs::probe::cmsisdap::tools > No HID interface for CMSIS-DAP found.
 DEBUG probe_rs::probe::cmsisdap::tools > Could not open cafe:4005 in CMSIS-DAP v2 mode
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 009: ID 17ef:3063
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 007: ID 04d9:2013
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 008: ID 046d:c52b
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 006: ID 05e3:0608
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 005: ID 17ef:1021
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 004: ID 17ef:1026
 TRACE probe_rs::probe::cmsisdap::tools > Error getting languages: Pipe
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 003: ID 17ef:3060
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 002: ID 17ef:1025
 TRACE probe_rs::probe::cmsisdap::tools > Error getting languages: Pipe
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 001: ID 1d6b:0002
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 001: ID 1d6b:0002
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open cafe:4005 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::espusbjtag::protocol  > Acquired libusb context.
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
```
I have also tried to connect to the probe with OpenOCD (which I really should have done before opening this issue) and encountered a similar problem.

However, I was now able to resolve the problem based on an issue report in the `pico-debug` project (inmajbthrd/pico-debug#16) though. The issue describes a problem connecting to CMSIS-DAP v1 probes due to missing kernel features in Windows Subsystem for Linux, which can apparently affect real Linux systems as well.
I have recompiled my kernel with the `CONFIG_HIDRAW`, `CONFIG_USB_HIDDEV` and `CONFIG_USB_SERIAL_DEBUG` flags set and am now able to connect to the probe with `probe-rs`.
Thanks again for your help.