probe-rs/probe-rs #1181: Generalized Jtag scan?

MabezDev, 2022-07-24:
The only probe that implements jtag chain scanning is ftdi. AFAIK (and I'm not an expert on JTAG) scanning can be implemented with simple ir & dr transfers. It it worth adding a `scan` trait method to `JTAGAccess` with a default implementation using ir & dr transfers?

Yatekii, 2022-07-24:
I guess that would be useful, yes :) Tho I am very much a novice when it comes to JTAG features and how they are used in RISC-V and the likes :)

MabezDev, 2022-07-24:
You and me both! Seems like a fun thing to try to implement though. Feel free to assign it to me if you like :)

Yatekii, 2022-07-24:
Without hesitation, SIr!

adamgreig, 2022-07-24:
I wrote this just for CMSIS-DAP with a vague idea to one day build it into probe-rs, but you might find it useful: https://github.com/adamgreig/jtagdap/blob/master/src/jtag.rs

Yatekii, 2022-07-24:
Oh yeah that old thought also never made it :D

PR #2066 (2024-01-22): Delete duplicate JTAG logic
FTDI, JLink and espusbjtag are essentially the same high level JTAG code. This PR removes the duplication from them, getting rid of a considerable amount of code.

This PR also skips IR detection if the configuration contains a scan chain specification. This is now consistent with the CMSIS-DAP implementation. On the other hand, there exists #1796 ... This PR is big enough and blocks enough others so that I would vote for resolving 1796 later.

I'm willing to say this closes #1181
closed 2024-01-22 by Yatekii