probe-rs/probe-rs #1194: Error while flashing No flash memory contains the entire requested memory range 0x42000020..0x4200a1da.

noppej, 2022-08-08:
**Describe the bug**
Attempting to flash the examples from https://github.com/esp-rs/esp-hal/tree/main/esp32c3-hal

This used to work, but somewhere in the latest updates between probe-rs and esp32c3-hal it has stopped working.

**Error**
`Error while flashing No flash memory contains the entire requested memory range 0x42000020..0x4200a1da.`

**Reproduce**
` cargo-flash --chip esp32c3 --elf target/riscv32imc-unknown-none-elf/debug/examples/blinky`
I get the same error with `probe-rs-debugger`.

**Other notes**
I can workaround this by fiddling with the `esp32.yaml` file in the `targets` folder, but cannot find enough documentation about the entries in that file to map the ESP32-C3 Technical Reference information into the proper memory regions / memory map for this device.

MabezDev, 2022-08-08:
Are you passing `--features direct-boot`? The direct boot feature lets the chip act like a more traditional chip. I haven't figured out how to do flash the normal method yet.

noppej, 2022-08-09:
@MabezDev When I try the `--features direct-boot` I get a linker error:
```
rust-lld: warning: address (0x42000008) of section .text is not a multiple of alignment (256)
          rust-lld: warning: address (0x3c00a0c4) of section .rodata is not a multiple of alignment (16)
          rust-lld: error: section .bss load address range overlaps with .rwtext
          >>> .bss range is [0xC54C, 0xCDAB]
          >>> .rwtext range is [0xC554, 0xCBFF]
```

noppej, 2022-08-10:
Update: The `direct-boot` issue has been resolved with the fix in https://github.com/esp-rs/esp-hal/pull/145.

However, the original problem still persists, and I cannot flash unless I use `direct-boot`.

Does anyone know who created the original `ESP32C3.yaml` target spec? I think it just needs an update to adjust to latest esp32c3-hal changes, because it used to work.

MabezDev, 2022-08-10:
> Does anyone know who created the original ESP32C3.yaml target spec? I think it just needs an update to adjust to latest esp32c3-hal changes because it used to work.

That was me :).

I don't think this ever worked. At least, I never even tried because it _shouldn't_ have worked. If you inspect the ELF produced in normal bootloader mode the physical address is wrong. This is because on Espressif chips, the two busses, instruction & data alias the same physical flash space.

On top of the differences above, when not using `direct-boot`, the ROM bootloader inside the chip expects the flash to contain a 32-byte program header, as well as segment headers for each loadable elf segment.

To fix this will require some changes to probe-rs to support the "esp" binary format, which of course would be up to the maintainers of probe-rs as to whether its acceptable or not.

noppej, 2022-08-10:
So weird. I've been testing and developing debug capability on this board for the last 6 months, and flashing continuously as part of the process. It was only after I updated my esp32c3-hal crate yesterday that it stopped working. Do you want me to reproduce a working setup on past branches, or should we just move on?

noppej, 2022-08-10:
> To fix this will require some changes to probe-rs to support the "esp" binary format, which of course would be up to the maintainers of probe-rs as to whether its acceptable or not.

@probe-rs/core ... What are your thoughts on this question?

MabezDev, 2022-08-10:
> Do you want me to reproduce a working setup on past branches, or should we just move on?

I'd be interested to see if it does! Maybe I'm misunderstanding some parts of the esp boot process, or maybe we just got lucky that it did work :D.

MabezDev, 2022-08-10:
> https://github.com/orgs/probe-rs/teams/core ... What are your thoughts on this question?

Just to touch on this, the extra code inside probe-rs should be minimal. We can depend on https://crates.io/crates/espflash to do the elf to esp format bin conversion, and then we can use the current bin flashing code that already exists in probe-rs to write that to flash. 

Where this might get a bit trickier is the CLI tooling, does `cargo-flash` assume elf? How would it know to switch to the esp format, it would be really nice if we didn't have to specify it all - perhaps this is possible by setting something in the .yaml target files.

If there is any interest, I could write a draft PR and see how it slots in.

noppej, 2022-08-10:
@MabezDev I've managed to create a test case where flashing with probe-rs works(ed).   Here's what I did:

- Forked and older branch of `esp-hal`  to https://github.com/noppej/esp-hal/tree/bootloader_probe_rs
- Added a custom linker script (this is from my own personal project where flashing used to work).  If you look at it, you will see that it is a merge of the hal's linker scripts from some time ago.
- I updated the `build.rs` to use this new linker script.

You can flash with either `cargo flash --chip esp32c3 --example blinky` or use the newly added `.vscode` files to use `probe-rs-debugger` to flash and debug the example.

MabezDev, 2022-08-11:
> * https://github.com/noppej/esp-hal/tree/bootloader_probe_rs

Thanks for the test case! It turns out however that even there you're using direct boot mode: https://github.com/esp-rs/esp-hal/commit/337b9990e11a96a5c1a52a5df972ad7668e99318#diff-2481625f2157ed7115547f87f4c57154dd8f9c2c121de63f504cb30606caaed5R48. This is probably because direct boot was the default on the C3 for a while.

noppej, 2022-08-11:
Aha ... that makes perfect sense. OK, so that resolves that mystery. 

Let's focus on your [suggestion/question](https://github.com/probe-rs/probe-rs/issues/1194#issuecomment-1210849663)

TheButlah, 2022-09-19:
@noppej I was able to find this, if thats helpful :)

noppej, 2022-09-19:
@TheButlah Thanks, that is a good overview, and I suspect that using it alongside looking at their source, it should be enough.

@Yatekii / @Tiwalun / @Dirbaio / @adamgreig ... I have no experience with probe-rs flash algorithms, so would appreciate your input in this. Is there any chance we can create a flash algorithm with this level of complexity, or do we need to resort to something as described by @MabezDev [here](https://github.com/probe-rs/probe-rs/issues/1194#issuecomment-1210849663)

MabezDev, 2022-11-25:
> I'm hoping there is some way we can use existing flash algo capability to implement the desired behaviour, rather than build a dependency on a chip-specific crate. What do you think? Too optimistic?

What did you have in mind? The algorithms define how to erase & write to a block of flash, but does not have a concept of the format, it just receives binary data.

noppej, 2022-11-25:
@MabezDev I really don't know enough to comment. It was more of a question than a suggestion, so I will repeat my question from above :) .... 

@Yatekii / @Tiwalun / @Dirbaio / @adamgreig ... I have no experience with probe-rs flash algorithms, so would appreciate your input in this. Is there any chance we can create a flash algorithm with this level of complexity, or do we need to resort to something as described by @MabezDev [here](https://github.com/probe-rs/probe-rs/issues/1194#issuecomment-1210849663). Either way, it would be nice to get some input from you on this request from @MabezDev :)

Tiwalun, 2022-11-25:
The flash algorithm doesn't really care about file formats, it can basically only erase and program. The probe-rs code running on the host so far can handle ELF files, Intel Hex files, and some more formats, and then extracts the necessary addresses and data for the flash algorithm. 

Adding support for a new ESP-specific format sounds reasonable to me, if it's just a different translation from ELF to addresses and data. If it's more involved, I'm not sure if the probe-rs library itself is the right place for it.

In any case, that doesn't mean that adding support in cargo-flash / cargo-embed wouldn't be possible. I'm just a bit worried that adding support for "everything" in a single probe-rs library makes the architecture even more complicated that it already is.

jneem, 2023-09-08:
If I understood correctly, this was done by #1629

MabezDev, 2023-09-08:
Correct, this can be closed!

closed 2023-09-08 by Dirbaio