probe-rs/probe-rs #1204: Add support for CC1352R

rockboynton, 2022-08-16:
**Is your feature request related to a problem? Please describe.**
No, only requesting to add a new supported chip 

**Describe the solution you'd like**
Support CC1352R

**Describe alternatives you've considered**
None

**Additional context**
None

jminer, 2022-08-23:
I'd like to use probe-rs with the CC2650 and CC2652, which should be pretty similar to the CC1352R. These chips use cJTAG, and I imagine it will be a decent amount of work to add cJTAG support. I think by default they use cJTAG, but they can be switched to use regular JTAG. However, with JTAG, you can't use SWO, and I know you can with cJTAG.

Here are some pages with information about cJTAG:
- https://web.archive.org/web/20220130184446/https://sudonull.com/post/106128-We-disassemble-the-2-wire-JTAG-protocol
- https://s3-eu-west-1.amazonaws.com/downloads-mips/mips-documentation/login-required/mips_cjtag_adapter_users_manual.pdf
- https://wiki.segger.com/J-Link_cJTAG_specifics

SeanLyonsTI, 2026-11-16:
`CC1352R` and `CC2652R` are now supported and so is cJTAG. I think we can close this one @Yatekii  or @bugadani

closed 2026-11-16 by bugadani

jminer, 2026-11-16:
I looked in the changelog and saw where those chips were added. However, I don't see where cJTAG was added. How can you select cJTAG? Looking in the docs, the `WireProtocol` enum only has `Swd` and `Jtag`. I'm not very familiar with the code, but I found that in `debug_port_setup` it looks like it unconditionally calls a `ctag_to_jtag` function: https://github.com/seanmlyons22/probe-rs/blob/5be904dea594b2d9dce7e857b403db47c7ff5113/probe-rs/src/config/sequences/cc13xx_cc26xx.rs#L380
Without cJTAG, it won't work with PCBs that only have the two cJTAG pins connected to the debug connector. And you can't use SWO for debugging, which can be very useful.

bugadani, 2026-11-16:
It helps if you're looking at something more recent, a lot has changed in the last two years :) https://github.com/probe-rs/probe-rs/blob/f42ee9beb247914611c91f180ef1b4fe5067db5d/probe-rs/src/vendor/ti/sequences/cc13xx_cc26xx.rs#L110

jminer, 2026-11-16:
Oh, I see, sorry. I thought GitHub switched to the newest, but it links to the old commit more places than I realized. Thanks for the link.