probe-rs/probe-rs #1220: Attaching to LPC55S69 seems to stop code execution

MikeJongen, 2022-09-05:
**Describe the bug**
A clear and concise description of what the bug is.

When executing 
`probe.attach("LPC55S69JBD100", Permissions::default())`
it seems to stop the execution of the code running on the lpc55S69.
Because this is also done as part of the flash command, code execution is stopped during the flash procedure, and a manual reset is required. This breaks the RTT connection, making it hard to use the RTT logging (with cargo-embed).

**To Reproduce**
Steps to reproduce the behavior:
(Using the LPC55S69-EVK)
1. Flash a simple led blink program to the lpc55S69, for example:
```#![no_std]
#![no_main]

use cortex_m::delay::Delay;
use cortex_m_rt::entry;
use hal::drivers::pins::Level;
use hal::prelude::*;
use lpc55_hal as hal;
use panic_rtt_target as _;
use rtt_target;

#[entry]
fn main() -> ! {
    rtt_target::rtt_init_print!();
    rtt_target::rprintln!("Starting I2C Test");

    let hal = hal::new();

    let mut anactrl = hal.anactrl;
    let mut pmc = hal.pmc;
    let mut syscon = hal.syscon;
    let mut iocon = hal.iocon.enabled(&mut syscon);

    let clocks = hal::ClockRequirements::default()
        .system_frequency(96.MHz())
        .configure(&mut anactrl, &mut pmc, &mut syscon)
        .unwrap();

    let systick = hal.SYST;
    let mut delay = Delay::new(systick, 96_000_000);

    let mut gpio = hal.gpio.enabled(&mut syscon);
    let pins = hal::Pins::take().unwrap();

    let mut red = pins
        .pio1_6
        .into_gpio_pin(&mut iocon, &mut gpio)
        // on = low, off = high
        .into_output(Level::High);

    loop {
        red.set_low().unwrap();
        rtt_target::rprintln!("Test");
        delay.delay_ms(1000);
        red.set_high().unwrap();
        delay.delay_ms(1000);
    }
}
```
Cargo.toml:
```
[package]
name = "startup_test"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
lpc55-hal = "0.3.0"
rtt-target = { version = "0.3.0", features = ["cortex-m"] }
cortex-m-rt = "0.6"
cortex-m = "0.7.6"
panic-rtt-target = { version = "0.1.1", features = ["cortex-m"] }
```

You should not see any led blinking now. 
3. press the reset button on the board. The led will start blinking now.
4. Run this script on the host PC:
```
use probe_rs::{CoreStatus, Permissions, Probe};
use std::{thread, time};

fn main() {
    // Get a list of all available debug probes.
    let probes = Probe::list_all();

    // Use the first probe found.
    let probe = probes[0].open().unwrap();

    // Attach to a chip.
    let mut session = probe
        .attach("LPC55S69JBD100", Permissions::default())
        .unwrap();
}
```
The led will now stop blinking again.

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected the led to keep blinking. (Thus indicating that the code on the LPC55S69 keeps running).

I ran a similar test on the LPC845. Here, the code execution was not stopped by `probe.attach()` (as expected). 

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

not applicable

**Desktop (please complete the following information):**
 - Windows 10

**Additional context**
Add any other context about the problem here.

After connecting using `probe.attach()`, the status can be read. This status indicates the board is `running`.
Functions such as `core.halt()`, `core.reset()` and `core.run()` change the status as expected, but do not seem to correspond with the actual status of the board (I never see the led blinking again, until a manual reset using the button on the board).

noppej, 2022-11-07:
@MikeJongen Can you please retest against #1276 ? 

I don't expect this to fix the problem, but noticed that the target yaml has updated flash values when generated against the latest pack files. Before continuing, I'd like to make sure that this doesn't make things worse :)

MikeJongen, 2022-11-08:
with #1276, I get the error:
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ArchitectureSpecific(RegisterRead { address: 12, name: "DRW", source: ArchitectureSpecific(NoAcknowledge) 
})', src\main.rs:20:10

line 20 is the attach:
let mut session = probe
        .attach("LPC55S69JBD100", Permissions::default())
        .unwrap();

noppej, 2022-11-13:
#1276 Helps bring us closer, but this issue still needs resolution.

bugadani, 2024-03-30:
Based on https://github.com/probe-rs/probe-rs/pull/1907#issuecomment-1831832480 I believe this issue should be fixed. Please feel free to reopen if it's not the case.

commit 1c92c2d6 (2022-11-13): Merge #1276

1276: Fix flash algorithm values for LPC55S69.yaml r=Tiwalun a=noppej

Fixes #1220
closed 2024-03-30 by bugadani