probe-rs/probe-rs #1221: GDB fails to connect when using RTT

TheButlah, 2022-09-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Use espressif's [`riscv32-esp-elf-gdb`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#riscv32-esp-elf-gdb) on macos (I have not tried with normal linux gdb)
2. Start cargo embed with rtt and gdb server
3. run `target remote :port` in gdb client

**Expected behavior**
GDB connects without issue

**Stacktrace**
```
(gdb) target remote :1337
Remote debugging using :1337
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;hwbreak+;qXfer:features:read+;qXfer:memory-map:read+
quit)
```

And in cargo embed, I see:
```
During the execution of GDB an error was encountered: Client nack'd the last packet, but `gdbstub` doesn't implement re-transmission.
```

**Desktop (please complete the following information):**
 - macOS Monterey (12.5.1)

**Additional context**
I am on an arm64 macbook (apple m1)
**The bug does not happen if I have `rtt` turned off**

noppej, 2022-10-20:
I thought this was fixed [here](https://github.com/probe-rs/cargo-embed/pull/159). Can you confirm that you have the latest `cargo-embed` please? If you do, then this is likely an issue related to the gdb implementation, so it would be good to know how it fares on a non-espressif gdb. 

I will close this issue here, but if it persists, please re-open at https://github.com/probe-rs/cargo-embed/issues

closed 2022-10-20 by noppej

nvx, 2022-10-20:
I ran into this recently as well on an RP2040 with msys gdb. Can grab exact version numbers later if needed, but nothing would be more than a few weeks old at most.