probe-rs/probe-rs #1222: ESP32-C3 error reading variable

TheButlah, 2022-09-10:
**Describe the bug**
The debugger sucessfully connects and can step through and pause breakpoints. But the variable values are not working.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to cargo-embed with RTT *off* (due to #1221)
2. Connect vscode to gdb (I am using [this](https://github.com/SlimeVR/SlimeVR-Rust/blob/6a7281bfc2d9d45bd4b31bf97754fcc50a054109/firmware/.vscode/launch.json) launch.json)
3. Try to look at the value of a variable

**Expected behavior**
The value of the variable is present without error

**Stacktrace**
I get this in the variables list for my variable:
```
<error reading variable: Cannot access memory at address 0x3fccf608>
```

[image]

**Desktop (please complete the following information):**
 - macOS Monterey (12.5.1)

**Additional context**
I am on an arm64 mac (apple m1) and using the [espressif-provided gdb](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#riscv32-esp-elf-gdb)

Yatekii, 2022-09-10:
I would dare say this is a GDB/espressif/vscode issue but ofc bugs in probe-rs are never to be ruled out. Can you enable login to see if the memory location that should be read is the one that is read by probe-rs? Rust debug symbols are kinda trash at times and it could well be it tries to access the wrong address.

TheButlah, 2022-09-10:
sure, this is with passing `RUST_LOG=trace` to `cargo embed`?

noppej, 2022-09-12:
@TheButlah Are you open to testing this with the `probe-rs-debugger` and probe-rs extension for VSCode? That way we can determine where the issue lies, and if it is in the probe-rs libs, then it gives us the ability to debug to where exactly the problem lies. 

- Install the **probe-rs-debugger** extension in VS Code, by downloading the _latest available_ `probe-rs-debugger-x.x.x.vsix` from the [Releases](https://github.com/probe-rs/vscode/releases) page in our repo (we will publish to the Microsoft Extension marketplace when we exit the 'Alpha' phase of this project)

  - Install the downloaded extension file from the command line, for example: `code --install-extension probe-rs-debugger-0.3.3.vsix` in the terminal

- Install the **probe-rs-debugger** server component, from the comand line with: `cargo install --git https://github.com/probe-rs/probe-rs --force --branch master probe-rs-debugger`

- Instructions and samples for setting up launch.json are in the [official docs](https://probe.rs/docs/tools/vscode/#usage-and-configuration)

TheButlah, 2022-09-13:
Here is what happens with this launch.json:
```json
{
            // "preLaunchTask": "${defaultBuildTask}",
            "type": "probe-rs-debug",
            "request": "launch",
            "flashingConfig": {
                "flashingEnabled": false,
                "resetAfterFlashing": true,
                "haltAfterReset": true
            },
            "consoleLogLevel": "Info",
            "name": "probe_rs Executable Test",
            "chip": "esp32c3", //!MODIFY
            "wireProtocol": "Jtag",
            "cwd": "${workspaceFolder}",
            "coreConfigs": [
                {
                    "programBinary": "${workspaceFolder}/target/riscv32imac-unknown-none-elf/debug/firmware" //!MODIFY
                }
            ]
        }
```

DebugConsole output:
```
[2022-09-13T08:39:07Z WARN  probe_rs::architecture::riscv] Breakpoint 4: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }

[2022-09-13T08:39:07Z WARN  probe_rs::architecture::riscv] Breakpoint 5: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }

[2022-09-13T08:39:07Z WARN  probe_rs::architecture::riscv] Breakpoint 6: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }

[2022-09-13T08:39:07Z WARN  probe_rs::architecture::riscv] Breakpoint 7: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
[2022-09-13T08:39:07Z WARN  probe_rs::architecture::riscv] Setting breakpoint 0

   Sent DAP Response sequence #13 : setBreakpoints
```

TheButlah, 2022-09-13:
> I would dare say this is a GDB/espressif/vscode issue but ofc bugs in probe-rs are never to be ruled out. Can you enable login to see if the memory location that should be read is the one that is [...]

Here is my attempt at reproducing the bug with RUST_LOG=debug. I cannot even get the breakpoints working now, not sure what changed. Here is the log when launching `cargo embed monitor` (no flashing):
[log_start.txt](https://github.com/probe-rs/probe-rs/files/9555298/log_start.txt)

Here is the log when connecting with vscode "native gdb debug" launch task (as described in original post):
[log_connect.txt](https://github.com/probe-rs/probe-rs/files/9555326/log_connect.txt)

noppej, 2022-09-13:
@TheButlah For the test you did with `probe-rs-debugger` ... in an ideal world, I'd like to see the launch.json do both the build and the flash, just to be 100% sure that the host debugger and the target binary are in synch. It is obvious that the debug info is computing incorrect memory addresses for variables and source code lines, and this is most likely something to do with how the code is built versus how it is flashed. 

Any chance you can give me access to your repo, so that I can debug directly? If this is private code, I understand, but will ask that you provide a workable test case, that I can build locally. That means cargo config files (to reproduce the build) as well as launch.json and minimal source code.

TheButlah, 2022-09-13:
> I'd like to see the launch.json do both the build and the flash
I did this once already, and then to save time I simply turned off flashing

> Any chance you can give me access to your repo, so that I can debug directly?
Sure, I should have posted that in the original github issue. I've edited the first post with a link to the repo.

noppej, 2022-09-13:
I will start debugging using your code. As an aside, I had to edit your `cargo.toml` to remove your defmt_fix, because of ...
```
    Updating git repository `https://github.com/TheButlah/esp-hal`
error: failed to load source for dependency `esp32c3-hal`

Caused by:
  Unable to update https://github.com/TheButlah/esp-hal?branch=defmt_fix

Caused by:
  failed to find branch `defmt_fix`

Caused by:
  cannot locate remote-tracking branch 'origin/defmt_fix'; class=Reference (4); code=NotFound (-3)
```

TheButlah, 2022-09-13:
This should fix it, I deleted that branch after it got merged to `esp-hal`:

```
[patch.crates-io]
# Currently patching esp32c3-hal with a change to the linker script that makes defmt work.
esp32c3-hal = { git = "https://github.com/esp-rs/esp-hal", rev = "c896e6c8d2062dd9222508b55e23197c29f8e98f" }
```

That being said, even after doing this the code no longer produces *any* rtt info with `cargo embed`. I'm very confused, as this most definitely used to work. Here is my log:

```
warning: 11 warnings emitted

      Config monitor
      Target D:\Ryan\Programming\slimevr\rust\firmware\target\riscv32imac-unknown-none-elf\debug\firmware
        INFO cargo_embed > Protocol speed 0 kHz
        WARN probe_rs::config::target > Using custom sequence for ESP32c3
        INFO probe_rs::probe::espusbjtag > JTAG IDCODE: 0x00005c25
        INFO probe_rs::architecture::riscv::sequences::esp32c3 > Disabling esp32c3 watchdogs...
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 0 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 1 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 2 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 3 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 4 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 5 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 6 and type 2
        INFO probe_rs::architecture::riscv                     > Discovered trigger with index 7 and type 2
        WARN probe_rs::architecture::riscv                     > Breakpoint 0: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 1: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 2: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 3: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 4: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 5: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 6: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        WARN probe_rs::architecture::riscv                     > Breakpoint 7: Mcontrol { .0: 601882624, type_: 2, dmode: false, maskmax: 31, hit: false, select: false, timing: false, sizelo: 0, action: 0, chain: false, match_: 0, m: false, s: false, u: false, execute: false, store: false, load: false }
        INFO cargo_embed                                       > Initializing RTT (attempt 1)...
        INFO cargo_embed                                       > RTT initialized.
Shutting down.
        WARN probe_rs::session                                 > Could not clear all hardware breakpoints: ArchitectureSpecific(Timeout)
```

Could it be that I set too many breakpoints, and they are still saved on the hardware somehow? And probe-rs is having trouble clearing them (the last WARN)?

TheButlah, 2022-09-13:
Oops, I forgot that I have `todo!()`s in there, so of course the RTT doesnt work. 😅

noppej, 2022-09-13:
@TheButlah I have sent you a PR on your code to show a working config with probe-rs-debugger, which allows the variables to be queried, and has working RTT.

What this proves to me is that the `probe-rs` lib is working as expected, and the problem is likely in the gdb layers.  

Do you agree? Is there something I can do to help you further?

[image]

noppej, 2022-09-13:
PS. With `probe-rs`, you no longer need the following in your `.cargo/config.toml` file ...
```
  "-C",
  "force-frame-pointers",
```

TheButlah, 2022-09-17:
@noppej I just attempted your PR. It doesn't work for me. Here is the output of the Debug Console in vscode:
```
eceived DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
INFO: FLASHING: Starting write of "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/firmware" to device memory
INFO: FLASHING: Completed write of "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/firmware" to device memory
WARNING: Cannot set breakpoint here. Try reducing `opt-level` in `Cargo.toml`, or choose a different source location: Other(No valid breakpoint information found for file: "/Users/ryan.butler/P/slimevr/rust/firmware/src/main.rs", line: 26, column: Some(1))
WARNING: Cannot set breakpoint here. Try reducing `opt-level` in `Cargo.toml`, or choose a different source location: Other(No valid breakpoint information found for file: "/Users/ryan.butler/P/slimevr/rust/firmware/src/main.rs", line: 26, column: Some(1))
WARNING: Warning: Could not set breakpoint at memory address: 0x420020fc: A core architecture specific error occurred
```

TheButlah, 2022-09-17:
Here is what I see with log level debug:
[log.txt](https://github.com/probe-rs/probe-rs/files/9592593/log.txt)

TheButlah, 2022-09-17:
If I run this with log level Info once again, and ensure that no breakpoints are set, this is what I see:
```

Received DAP Request sequence #1 : initialize
   Sent DAP Response sequence #1 : initialize

Received DAP Request sequence #2 : launch
INFO: FLASHING: Starting write of "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/firmware" to device memory

Triggered DAP Event: progressStart

Triggered DAP Event: progressUpdate

Triggered DAP Event: progressUpdate

... omitting ...

Triggered DAP Event: progressUpdate

Triggered DAP Event: progressUpdate

Triggered DAP Event: progressEnd
INFO: FLASHING: Completed write of "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/firmware" to device memory
   Sent DAP Response sequence #2 : launch

Triggered DAP Event: initialized

Received DAP Request sequence #3 : setInstructionBreakpoints
   Sent DAP Response sequence #3 : setInstructionBreakpoints

Received DAP Request sequence #4 : configurationDone
   Sent DAP Response sequence #4 : configurationDone

Received DAP Request sequence #5 : threads
   Sent DAP Response sequence #5 : threads

```
Then I attempt to hit the "pause" button:
```

Received DAP Request sequence #6 : pause

Triggered DAP Event: stopped
   Sent DAP Response sequence #6 : pause

Received DAP Request sequence #7 : threads
   Sent DAP Response sequence #7 : threads

Received DAP Request sequence #8 : stackTrace
   Sent DAP Response sequence #8 : stackTrace

Received DAP Request sequence #9 : stackTrace
   Sent DAP Response sequence #9 : stackTrace

Received DAP Request sequence #10 : scopes
   Sent DAP Response sequence #10 : scopes

Received DAP Request sequence #11 : stackTrace
   Sent DAP Response sequence #11 : stackTrace

```

However, I see this in the GUI:
[image]

Please lmk if there is any more information I can provide.

noppej, 2022-09-17:
@TheButlah ... this is weird. I've just tried again, and everything works exactly the way I would expect it to.

- Can you try adding `"speed": 1000` to your `launch.json` file? Just to slow USB comms down a bit and test if that is a factor?
- Can you give me details about which board and probe specifically you are using? I am using [this one](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp8684/esp8684-devkitm-1/index.html) and using [these instructions](https://visualgdb.com/tutorials/esp32/esp32-c3/) to connect to USB.

noppej, 2022-09-17:
> Then I attempt to hit the "pause" button:

In this log, it tries to set instruction breakpoints. That surprised me. 
- Did you set these on purpose?
- Do you have the "Disassembly" window open? 
- Is it possible to clear all breakpoints, and then set a single source code breakpoint early in `main()`?

TheButlah, 2022-09-17:
I am using [this](https://github.com/TheButlah/slimevr_pcb/tree/5b4e7ea92693ecd4c8b72c9c5600073246561c61/hardware/ferrous_slime) board, which I designed myself. It has worked successfully in the past, so it is not a hardware problem here. I based the design on [espressif's offiicial rust dev board](https://github.com/esp-rs/esp-rust-board), although the design diverged somewhat.

Setting `"speed": 1000` appeard to have no effect.

To answer your other questions:
* No, I have made sure to remove all breakpoints from the code in vscode's GUI. I have no breakpoints set unless in the post I said "I try turning on a breakpoint". I was also confused why the logs talk about breakpoints.
* I do not know what the "Disassembly" window is, but I don't see anything that says disassembly.
* No, I tried this. It gives the same errors, at least as best as I can tell. Do you want me to upload logs for that just in case?

noppej, 2022-09-18:
@TheButlah I think we are dealing with two different issues here.

1) For some reason `probe-rs-debugger` doesn't recognize when the board has halted (that is why you had to press "pause" to get a stack trace. I can reproduce this issue and will have a look at it.
2) The problem with setting breakpoints and reading variable memory ... since this is why you logged this call, let's focus this thread on that issue. 

I note that you say this used to work, and based on that, I have a couple of questions ...
- Can you systematically roll back your code, as well as esp-hal version, and the same for other key dependencies, so that we can pin down where the breaking change happened? (both esp-hal and probe-rs had recent changes regarding flash memory handling).
- The esp `direct-boot` feature implies that the code will be running from flash memory. The flash memory map in `probe-rs` is hardcoded in the target file, and that memory map will impact how the debug code will address breakpoints (mapping from source to memory addresses) as well as for where to read variable values from. Since both these are failing, I wonder if we have a memory map problem. Can you confirm your flash config for your custom board?
- Have you tried using esp tooling to flash without the `direct-boot` and debugging that?

noppej, 2022-09-18:
@TheButlah 

> 1. For some reason `probe-rs-debugger` doesn't recognize when the board has halted (that is why you had to press "pause" to get a stack trace. I can reproduce this issue and will have a look at it.

I've just created a draft PR #1228 for this issue. Can you please test it and confirm that it improves the halt problem?

PS. You can install with `cargo install --git https://github.com/probe-rs/probe-rs --force --branch debug_launch_status probe-rs-debugger`

TheButlah, 2022-09-18:
Sorry, I should clarify what I mean when I say it used to work. It works with espressif's official tooling (openocd, gdb, esp-idf etc). I have not gotten it to work with probe-rs before

TheButlah, 2022-09-18:
> Can you please test it and confirm that it improves the halt problem?

sure, here is a log. A few things to note: This is with debug level logs, and I have turned `"haltAfterReset": true` in the `launch.json` file to demonstrate the halt without me manually pushing the pause button.
[log.txt](https://github.com/probe-rs/probe-rs/files/9595304/log.txt)

> For some reason probe-rs-debugger doesn't recognize when the board has halted (that is why you had to press "pause" to get a stack trace. I can reproduce this issue and will have a look at it.

I have been using your PR, which does uses `"haltAfterReset": false`. I assume this implies that the only way the board would pause is if a breakpoint was set or if I hit "pause". Should I be using `"haltAfterReset": true` from here on out while testing?

TheButlah, 2022-09-19:
OK, some new details have come to light. I'm not sure if it is because I am using [your PR](https://github.com/SlimeVR/SlimeVR-Rust/pull/33) or what, but I can now set breakpoints on *non-macro* functions. So for example I can set a breakpoint on a `let p = self::peripherals::get_peripherals();` line but not on a `error!("yeet")` line. 

[image]

I can confirm that stepping (when not using macros) and inspecting variables works.

Here is a preliminary list of other ssues I experience. LMK which of these if any pertain to this opened issue and which should have a new issue opened:

* I see only stuff in the variables box but the other boxes are empty. However, the variables box has subsections for registers and statics.
[image]

* If I try setting a breakpoint on a macro line, I get:
```
WARNING: Cannot set breakpoint here. Try reducing `opt-level` in `Cargo.toml`, or choose a different source location: Other(No valid breakpoint information found for file: "/Users/ryan.butler/P/slimevr/rust/firmware/src/main.rs", line: 26, column: Some(1))
```

* Stepping over lines (when the next line is a macro) acts as if you just hit "resume". I assume this is because the plugin doesn't understand how to break on a macro. I can confirm that if lines don't include macros, they can be stepped over.

* Attempting to expand the subsection under the variables box called "Statics" triggers a stack overflow, with the following log (collected on debug log level, log cleared just before expanding the subsection):
[log.txt](https://github.com/probe-rs/probe-rs/files/9597071/log.txt)

noppej, 2022-09-19:
> I'm not sure if it is because I am using [your PR](https://github.com/SlimeVR/SlimeVR-Rust/pull/33) or what,

Have you made any changes to that PR, or can I assume I can test with it as an "exact" copy of the code you run?

noppej, 2022-09-19:
> * Attempting to expand the subsection under the variables box called "Statics" triggers a stack overflow, with the following log (collected on debug log level, log cleared just before expanding the subsection):
>   [log.txt](https://github.com/probe-rs/probe-rs/files/9597071/log.txt)

Can you please open a new issue for this and assign it to me?

noppej, 2022-09-19:
> * Stepping over lines (when the next line is a macro) acts as if you just hit "resume". I assume this is because the plugin doesn't understand how to break on a macro. I can confirm that if lines don't include macros, they can be stepped over.

Can you please open a new issue for this and assign it to me?

noppej, 2022-09-19:
> * I see only stuff in the variables box but the other boxes are empty. However, the variables box has subsections for registers and statics.

The default VSCode UX for debugging only has a `VARIABLES` panel, with additional panels for `WATCH`, `CALL STACK`, `BREAKPOINTS`, and `MODULES`.  I am not sure where your `REGISTERS` and `STATICS` panels come from, but I am guessing they are remnants that were created by another debug extension. If they bother you, you can hide them, or disable other debug extensions. For `probe-rs` debug extension, all the variables, registers, etc. will be subfolders in the `VARIABLES` panel.

noppej, 2022-09-19:
> I can confirm that stepping (when not using macros) and inspecting variables works.

Does this mean we can now close this issue, and we can deal with the other outstanding ones individually?

closed 2022-09-19 by bors[bot]

TheButlah, 2022-09-19:
Created #1230 and #1231