probe-rs/probe-rs #1223: Not being able to flash the nRF9160

macthestack, 2022-09-13:
I think this is related to #470 that I filed almost two years ago.

I'm trying to get the nRF9160 working on a new project and I'm running in to the same problem I thought was fixed long ago. Could it be that I have misconfigured something or is this a regression of the probe code?

```
cargo embed
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
      Config default
      Target /Users/xxx/repos/xxx/target/thumbv8m.main-none-eabi/debug/xxx
     Erasing sectors ✔ [00:00:00] [-------------------------------------------------------------------------------------------------------------------------]       0B/32.00KiB @       0B/s (eta 0s )
 Programming pages   ✔ [00:00:00] [-------------------------------------------------------------------------------------------------------------------------]       0B/32.00KiB @       0B/s (eta 0s )
       Error failed to flash /Users/xxx/repos/xxx/target/thumbv8m.main-none-eabi/debug/xxx

             Caused by:
                 0: Error while flashing
                 1: Failed to erase flash sector at address 0x00040000.
                 2: Failed to erase flash sector at address 0x00040000.
                 3: A core architecture specific error occurred
                 4: Failed to read register DRW at address 0x0000000c
                 5: An error specific to the selected architecture occurred
                 6: Target device responded with a FAULT response to the request.
```

Thanks!

macthestack, 2022-09-13:
Just tested that it works with cargo embed version 0.11.0

eflukx, 2022-12-01:
> [@eflukx](https://github.com/eflukx) Are you sure it's probe-run 0.3.3 that fixed it? Because for me it's going back to 0.3.0 that fixes it because all versions after that use probe-rs 0.12 instead of the working 0.11

@diondokter I see you removed your (quoted) question about the exact version used, so probably it's not relevant anymore. Anyway I'd just like to confirm that I'm successfully using probe-run v0.3.3 with a nRF9160 module.

Tiwalun, 2023-01-18:
What kind of debug probe are you using? Does this happen with the nRF9160 DK?

diondokter, 2023-01-19:
> What kind of debug probe are you using? Does this happen with the nRF9160 DK?

Happens on the DK too yeah. When using nRF chips, I usually use a jtag probe of one of my DK boards

korken89, 2023-02-02:
I also have issues using an nRF9160-DK

diondokter, 2023-04-14:
I think this issue can be closed now. The PR of Dario seems to have solved this.

closed 2023-04-14 by Yatekii