probe-rs/probe-rs #1230: probe-rs doesn't cant break on or step over macros

TheButlah, 2022-09-19:
**Describe the bug**
Breakpoints are not able to be set on lines that invoke macros.
In addition, stepping to a line with a macro invocation fails, and execution resumes until the next breakpoint is hit.

**To Reproduce**
Steps to reproduce the behavior:
1. Try setting a breakpoint on a macro invocation like `defmt::error!()`

OR 
1. Pause execution on a line before a macro
3. Try to step to next line

**Stacktrace**
If I try setting a breakpoint on a macro line, I get:
```
WARNING: Cannot set breakpoint here. Try reducing `opt-level` in `Cargo.toml`, or choose a different source location: Other(No valid breakpoint information found for file: "/Users/ryan.butler/P/slimevr/rust/firmware/src/main.rs", line: 26, column: Some(1))
```

**Desktop (please complete the following information):**
 - macOS Monterey (12.5.1)

**Additional context**
I am using an `ESP32-C3` via the onboard usb `JTAG` peripheral (USB port is directly connected to the D+ and D- pins of the MCU)

TheButlah, 2022-09-19:
@noppej

noppej, 2022-09-19:
FYI. Instead of hardcoding `DEFMT_LOG` in your `.cargo/config.toml` file, you can just use the `launch.json` setting for `consoleLogLevel` and the `DEFMT_LOG` value will follow whatever that setting is.

noppej, 2022-09-19:
PS. Finally, with `probe-rs-debugger` you do NOT need the `.cargo.config.toml` setting:
```
  "-C",
  "force-frame-pointers",
```

By deleting this, you will find your binary is both smaller and faster :)

TheButlah, 2022-09-19:
> By deleting this, you will find your binary is both smaller and faster :)

I had it because of `esp-backtrace`. But I am no longer using the backtrace stuff from that crate and have removed it, so I will remove that flag now :)

TheButlah, 2022-09-19:
> Can please you change your .cargo/config.toml appropriately and confirm that this works for you?

I have been flashing with `DEFMT_LOG=debug cargo embed flash` ([link](https://github.com/SlimeVR/SlimeVR-Rust/blob/1f2fd208996ee82b560f16232a34d1760f2ce8e1/firmware/Embed.toml#L24)), and have been running the vscode debugger without flashing. When doing it this way, I do see see the log statements in the RTT console run by the vscode debugger so they are definitely not compiled out. Yet I still cannot set breakpoints on the macros.

Edit: Actually, *sometimes* the breakpoints work. For example, with `DEFMT_LOG=info` in my `.cargo/config.toml` and `DEFMT_LOG=debug` passed on the CLI to `cargo embed flash`, I have this code:

[image]

I can step each line one by one, getting to line 25 successfully. Please note that the RTT window shows no output so far (maybe a separate bug, or maybe its something to do with how defmt flushes its data?).

Then, when I try to step to line 26 (macro invocation), the program resumes without pausing at 26, running until it panics later in the program.  At this point, I can see the RTT logs, which include all of the DEBUG statements.

So clearly the binary has the DEBUG level logs baked in and is actually executing those instructions. But for some reason probe-rs still doesn't always break on the macro lines.

TheButlah, 2022-09-19:
I just tested again with debug level logs in both the cargo toml, the CLI, and the launch.json. The behavior remains the same as described above.

This time I also got:
[image]

TheButlah, 2022-09-19:
Another test: I just tried deleting the `[env]` section from the `.cargo/config.toml` and turning flashing on in my `launch.json`. As before, I have `debug` log levels in the `launch.json`.

Behavior remains the same as described above. I did not get the "core must be halted" popup this time, maybe that is unrelated?

TheButlah, 2022-09-19:
BTW, the panic I am referring to occurs much later in the program, after the async executor has already been initalized and the tasks begin attempting IO. I have confirmed that if I were to set a breakpoint at line 29, then when the debugger fails to break at line 26, it will continue until line 29.

noppej, 2022-09-19:
@TheButlah This is running into a situation where the debug info doesn't provide enough information to know where to set a breakpoint for that statement.  The code to step correctly relies on knowing which is the next statement (there can be multiple statements on a single line in the source, and each statement is made up of multiple machine instructions which can be grouped into multiple execution sequences). When a single statement compiles into multiple machine instructions which belong to multiple execution sequences, then we exceed the current logic of `probe-rs` stepping capability (Open a VSCode Disassembly window at the line before the debug macro, and you will see how complex the execution sequence is for that single line of code). 

I don't believe this is unique to macros, but certainly the defmt macros are complex enough to trigger this limitation more frequently.

I will work on this, but it is a non-trivial fix, so in the mean time, I recommend you use breakpoints after the macros to make sure you can halt when expected.

TheButlah, 2022-09-19:
Can this scenario be detected before actually resuming execution? It would make the experience much better if the step function didn't step and instead warned you that it would cause execution to be fully resumed. That way I could use the step function without it randomly (from the user's perspective) resuming execution

noppej, 2022-09-19:
> It would make the experience much better if the step function could detect this beforehand, and then warn you that it would cause execution to be fully resumed instead of stepping.

This is what I've been aiming for in the code, but there are obviously some scenarios that escape detection. But your point is 100% valid :)

noppej, 2022-12-13:
@TheButlah Are you able to validate https://github.com/probe-rs/probe-rs/pull/1327. I can no longer reproduce the problem on my own repo, or the one you linked above.

closed 2022-12-14 by bors[bot]

TheButlah, 2022-12-14:
> @TheButlah Are you able to validate #1327. I can no longer reproduce the problem on my own repo, or the one you linked above.

I don't have access to my hardware rn as I'm on vacation. If I ever notice the issue again I'll let you know.

noppej, 2022-12-15:
Breakpoints and stepping will be an onion that we have to continue to peel one layer at a time ... with maybe some specific requests for improvements on the DWARF that is generated.

commit 122e51ef (2022-12-14): probe-rs doesn't break on or step over macros
Fixes #1230