probe-rs/probe-rs #1231: Stack overflow when expanding "static" section in probe-rs-debugger

TheButlah, 2022-09-19:
**Describe the bug**
When expanding this static section in the VSCode plugin:
[image]
I get this:
[image]

The log ends with:
```
thread 'main' has overflowed its stack
fatal runtime error: stack overflow
```

**To Reproduce**
Steps to reproduce the behavior:
1. Pause execution on a breakpoint
2. Attempt to expand the static section

**Expected behavior**
The box should not crash but instead show the static variables in memory.

**Stacktrace**
Here is a log of the error with log level "Debug":
[log.txt](https://github.com/probe-rs/probe-rs/files/9597071/log.txt)

**Desktop (please complete the following information):**
 - macOS Monterey 12.5.1

**Additional context**
Add any other context about the problem here.

TheButlah, 2022-09-19:
@noppej

noppej, 2022-09-19:
Thanks. I am able to reproduce this on a simple test case that uses the embassy executor. They are using a linked list for task management that causes an infinite loop in determining the task header's value.

I am working on a fix and will ping you when it is ready to test.

noppej, 2022-09-19:
@TheButlah Can you please test your static variable issue against PR # 1232?

Reminder, you can install locally with ... 
```
cargo install --git https://github.com/probe-rs/probe-rs --force --branch noppej/issue1231 probe-rs-debugger
```

TheButlah, 2022-09-19:
Yep, this appears to fix it!

noppej, 2022-09-19:
Thanks - I appreciate your patience and commitment with logging issues, testing and helping to improve this repo!

PS. That PR branch has an outstanding (newly introduced problem), so it needs a bit more testing/work from me before I submit it for review/merge.

commit 13a9ee7a (2022-09-20): Stack overflow when expanding "static" section in probe-rs-debugger
Fixes #1231
commit 844169bc (2022-09-20): Merge #1232

1232: Stack overflow when expanding "static" section in probe-rs-debugger r=Yatekii a=noppej

There was an error in the logic that prevented cetain variable types from being loaded in a deferred/lazy way. This resulted in situations where recursively referenced types (as in a linked list) would loop forever trying to resolve all children of a structured variable.

Fixes #1231

Additional fixes: 
- This also fixes a new issue (exposed by testing of this PR). Recent changes in RUST have removed the intermediate `DW_AT_name` attribute for the `DW_AT_type` of the `data_ptr` child of ` [... cut]
PR #1232 (2022-09-20): Stack overflow when expanding "static" section in probe-rs-debugger
There was an error in the logic that prevented cetain variable types from being loaded in a deferred/lazy way. This resulted in situations where recursively referenced types (as in a linked list) would loop forever trying to resolve all children of a structured variable.

Fixes #1231

Additional fixes: 
- This also fixes a new issue (exposed by testing of this PR). Recent changes in RUST have removed the intermediate `DW_AT_name` attribute for the `DW_AT_type` of the `data_ptr` child of `&str` variables, resulting in a failure [... cut]
closed 2022-09-20 by bors[bot]