probe-rs/probe-rs #1234: cargo-flash does not work with seeed xiao nrf52840

OleStrohm, 2022-09-20:
**Describe the bug**
`cargo flash --list-probes` does not list the seeed xiao nrf52840, and so none of the other functionality works either.

**To Reproduce**
1. Plug in the seeed xiao nrf52840
2. Double click the button to enter bootloader mode
3. Run `cargo flash --list-probes`

**Expected behavior**
It should list the seed xiao nrf52840 as a probe, and then the rest of the functionality should work.

**Desktop (please complete the following information):**
Arch linux 5.19.7-arch1-1

**Additional context**
The chip in question is the [Seeed Xiao nRF52840](https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html).
I followed the [Getting Started](https://wiki.seeedstudio.com/XIAO_BLE/) guide, and managed to flash a blink program on the chip, so I know it's a functional chip.

Noteworthy is that this required me to install [adafruit-nrfutil](https://github.com/adafruit/Adafruit_nRF52_Arduino).

Here are the outputs from some relevant commands:
```
$ lsusb
Bus 001 Device 030: ID 2886:0045 Seeed Technology Co., Ltd. XIAO nRF52840 Sense

$ cat /etc/udev/rules.d/50-oxidize-global.rules
# Allow accessing the nrf52840 seeed xiao without root

# nrf52840 seeed xiao
ATTRS{idVendor}=="2886", ATTRS{idProduct}=="0045", TAG+="uaccess"

$ `ls -l /dev/bus/usb/001/029
crw-rw----+ 1 root plugdev 189, 29 Sep 20 22:56 /dev/bus/usb/001/030
```

Finally, I know it says `nrf52840 Sense` sometimes (it says the same thing in the arduino ide as well), but unless they sent me the wrong device, they probably just use the same id for both of their products or they write the wrong name inside it.

Hopefully I've done something wrong or this can be fixed, because I'd love to program this chip with Rust! :D

Yatekii, 2022-09-21:
So probe-rs supports CMSIS-DAP probes, ST-Links and J-Links. Everything else is not supported (yet).
I am not sure what this board uses. Glancing over the page you linked didn't tell me anything. But since it's apparently used with the Arduino IDE, I am kind of sure it uses the USB-Bootloader instead of a debug probe, which is out of scope of this project.

OleStrohm, 2022-09-21:
Ah, I see, that's understandable!

It seems like the actual program used to upload in the end is adafruit-nrfutil (which is a patched version of nrfutil made by nordicsemi), which seems to basically be a special version of a dfu flasher, which I believe is the bootloader used.

Do you know how other nrf52840 boards differ in their implementation, do you have to connect via a JTAG connector or do they use a different method of usb uploading to support these debug probes?

Yatekii, 2022-09-21:
probe-rs can communicate over SWD and JTAG with more protocols being in the making but moving slowly. We are a **debug probe** library, so DFU is out of scope. That would go directly into `cargo-flash` and I would accept PRs there :)

We flash nRF52840 over SWD, as pretty much all ARM MCUs are flashed :)

OleStrohm, 2022-09-22:
Thanks for the help! Closing this as it's outside the scope :)

closed 2022-09-22 by OleStrohm

ariofrio, 2023-09-06:
`cargo-flash` was moved into this repository. Are you still opening to adding DFU support to `cargo-flash` (but not, obviously, debugging using probe-rs)?

Yatekii, 2023-09-08:
Yeah I am still open to that :)