probe-rs/probe-rs #1235: micro:bit v2.20 fails to keep stable connection

ctron, 2022-09-21:
**Describe the bug**

After a while (depending on the using `probe-run` or `cargo embed`) both terminate with:

```
RTT error: Error communicating with probe: A core architecture specific error occurred
────────────────────────────────────────────────────────────────────────────────
Error: A core architecture specific error occurred

Caused by:
    0: Failed to read register DRW at address 0x0000000c
    1: An error specific to a probe type occurred
    2: Error handling CMSIS-DAP command Transfer
    3: Timeout in USB communication.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Flash
2. Let it run
3. boom

probe-run fails after 5-30 seconds. cargo embed after ~15 minutes.

**Expected behavior**

It should be as stable as the micro:bit v2.00.

**Stacktrace**

n/a

**Desktop (please complete the following information):**

Fedora 36

**Additional context**

* https://support.microbit.org/support/solutions/articles/19000132336-bbc-micro-bit-v2-interface-processor-change-v2-2-

I also tried upgrading the interface firmware from `0256` (shipped) to `0257`. This doesn't seem to make a difference.

noppej, 2022-09-21:
@ctron Can you please provide a little bit of additional information about your environment?

For example:
- Which RTT libraries are you using `defmt` or basic `rtt-target`?
- Which versions of the various dependencies?
- How big are the RTT buffers?
- What speed are you connecting at?
- How frequent are you writing to RTT, and how much data at a time?

There are literally 100's or variables that can cause instability, and although your request for stability is 100% valid, we need to be able to reproduce this issue before we can pin down the root cause.  

In an ideal world, a small reproduce-able test case will go a looong way towards getting you the help you request :)

ctron, 2022-09-21:
Sure, I will try to. The full code is here: https://github.com/Eclipse-IoT/eclipsecon-2022-hackathon/tree/main/firmware

* It is using defmt
* The list of dependencies can be seen in the [Cargo.toml](https://github.com/Eclipse-IoT/eclipsecon-2022-hackathon/blob/main/firmware/Cargo.toml) file
* > How big are the RTT buffers?
  
  I have no idea.
* I tried speeds of: 100, 1000, 10000 kHz … didn't make a difference
* After the startup, nothing is written unless I press a button (which I don't). I however did not test if sending output would keep the connection alive. I will try that.

I actually have this issue with all embassy based (don't have others) examples that I tried. So maybe I can try to really boil it down to tiny example.

noppej, 2022-09-21:
@lulf When I clone and follow your README to run this, I get this ...
```
error[E0432]: unresolved import `embassy_futures::Either`
  --> display/src/main.rs:12:31
   |
12 | use embassy_futures::{select, Either};
   |                               ^^^^^^ no `Either` in the root

error[E0423]: expected function, found module `select`
  --> display/src/main.rs:27:15
   |
27 |         match select(btn_a.wait_for_low(), btn_b.wait_for_low()).await {
   |               ^^^^^^ not a function
```

Am I missing something obvious?

lulf, 2022-09-21:
@noppej apologies, i had some local changes that were not committed. Please pull and try again now

noppej, 2022-09-21:
@lulf Thanks. That works, but all I've managed to prove is that it works on V2 as you said LOL

While doing that I confirmed that V2.20 boards are on backorder in Canada, so it seems the fastest I can get one is Oct 19th :-(

lulf, 2022-09-21:
I'm receiving one tomorrow so I can debug it further. I'll post whatever I find here . Thanks anyway ☺️

lulf, 2022-09-26:
Quick update on whether or not writing to RTT has an effect. Printing to RTT every 4 seconds, I'm seeing the device crash after running up to the following times:

27 s
441 s
766 s
16 s
81 s

noppej, 2022-09-26:
Does this mean the connection remains stable when you have no RTT output?

lulf, 2022-09-26:
No, it remains unstable even if there is no RTT output.

Tiwalun, 2022-11-27:
I tried to reproduce this on MacOS, but it's been running for over an hour without problems. @lulf just to confirm, the crash happens when you just have a loop printing to RTT, without defmt involved? And which OS were you using?

Yatekii, 2022-11-28:
Well the Question is basically if this is really an instablility with probe-rs/RTT or with defmt. defmt has some unrecoverable characteristics. Even though I think they introduced some framing. cc @Dirbaio

Tiwalun, 2022-11-28:
Why do you think that should lead to a USB error? 

For me, it seems that the error is in the USB connection, not on a higher level. That would also explain better why it seems to break on Linux, but not on MacOS.

Yatekii, 2022-11-28:
Sorry, my brain completely messed up the context. Disregard my previous comment.

BartMassey, 2024-03-08:
I am having the same issue on Microbit v2.2. I am at firmware 0258-pre currently, but it doesn't seem to matter.

With `rtt-target` and `cargo-embed`, I haven't seen the disconnects (but maybe I haven't waited long enough). With `rtt-target` and `probe-rs run`, the disconnects occur within seconds. This is *very* reliably reproducible for me, on a variety of boards and with a variety of programs.

I am happy to help with further debugging, and to look for the problem, if given some direction.