probe-rs/probe-rs #1236: probe-rs-cli-util keeps panicking

LeandroMarceddu, 2022-09-24:
**Describe the bug**
[image]
This keeps happening in both attach mode and launch mode. 

**Stacktrace**
I need more info on how to pass the backtrace environment variable to VSCode's debug console. 

**Desktop (please complete the following information):**
 - Windows 11

Yatekii, 2022-09-24:
Hey :)

Please provide a bit more info around your setup.
It sounds like it's actually an error with `defmt` . Do you use the correct version (the version you use in the firmware needs to match the one used by `probe-rs-cli-util`)? Do you have `defmt` set up correctly? According to the error the `defmt` symbol lookup fails, which points to `defmt` troubles.

Best,
Noah

cc: @noppej

LeandroMarceddu, 2022-09-24:
Hey there.

The weird thing is it works pretty well for about a minute and then it suddenly panicks. 

Project is generated via cargo generate from https://github.com/rp-rs/rp2040-project-template.

Project is currently viewable at https://github.com/LeandroMarceddu/rusty-kiln.

Yatekii, 2022-09-24:
Maybe it tries to access a specific symbol only after 1 minute and for some whatever reason the symbol is not generated by `defmt`?

LeandroMarceddu, 2022-09-24:
```rust
        info!(
            "Set {} min {} reached {} step {}",
            setpoint_c, count_minutes, reached, step
        );```

Is the only thing that's outputting constantly. I'm not sure what symbols you are referring to.

Yatekii, 2022-09-24:
Hmm yeah that's odd indeed.
`defmt` generates debug symbols to reconstruct the log statements on the host. And the error you get tells us that it fails to look up a certain symbol, which is odd.

noppej, 2022-09-26:
@LeandroMarceddu I looked at your code, and I couldn't see anything obvious that contributes to the issue. That said, you have quite a few dependencies and I was wondering if you'd be willing to try to systematically reduce code functionality and dependencies, in the hope that it is one specific conflict that results in that. 

My thinking is that this will either identify a conflict, or reduce your code to a very simple reproduce-able test case that we can debug. Are you willing to try that approach?

LeandroMarceddu, 2022-09-26:
> @LeandroMarceddu I looked at your code, and I couldn't see anything obvious that contributes to the issue. That said, you have quite a few dependencies and I was wondering if you'd be willing to try [...]

Would simply commenting them out work? I will try that.
As of right now I do have indeed quite a few unused deps, however I *do* need them in the future.

LeandroMarceddu, 2022-09-26:
> Sure, commenting out dependencies will work. I was mostly thinking about doing that with 'used' dependencies, and the code that is associated with them. But try it one-by-one, to see if we can identify a conflict.

https://github.com/LeandroMarceddu/rusty-kiln/tree/e566761a56fb6784deea01bd6e393c96f5c2ac48

I've reduced it to about the bare minimum to get atleast some functionality out of the board, yet it still occurs.

noppej, 2022-09-26:
Awesome. I can actually reproduce it with this test-case. Let me have a look at it and get back to you.

noppej, 2022-09-26:
@LeandroMarceddu . Can you please test this against the PR?
`cargo install --git https://github.com/probe-rs/probe-rs --force --branch noppej/issue1236 probe-rs-debugger`

commit 2b1d0f45 (2022-09-26): Prevent panicking in probe-rs-cli-util.
Fixes #1236
commit ea675f6b (2022-09-26): Merge #1241

1241: Prevent panicking in probe-rs-cli-util. r=Yatekii a=noppej

Fixes #1236

The current code panics when `defmt-decoder` reports an invalid frame index for the source location of the string.

This update intercepts (and reports) that condition, rather than panicking.

The user's test case only prints a single string, and for that string, the frame index is `8`, but every so often, `defmt_decoder` reports a frame index of `2`, which used to panic when trying to lookup the source location using that index. 

The updated code will now report the error as part of the output, e.g.
` [... cut]
PR #1241 (2022-09-26): Prevent panicking in probe-rs-cli-util.
Fixes #1236

The current code panics when `defmt-decoder` reports an invalid frame index for the source location of the string.

This update intercepts (and reports) that condition, rather than panicking.

The user's test case only prints a single string, and for that string, the frame index is `8`, but every so often, `defmt_decoder` reports a frame index of `2`, which used to panic when trying to lookup the source location using that index. 

The updated code will now report the error as part of the output, e.g.
```
Set 24.0 min 0 reached true step 1
└ [... cut]
closed 2022-09-26 by bors[bot]