probe-rs/probe-rs #1238: Debugger not working on Pico

cstrahan, 2022-09-25:
**Describe the bug**

In VSCode, attempting to launch the debugger against a Raspberry Pi  Pico (using another Pico as a CMSIS-DAP debugger, using [DapperMime](https://github.com/majbthrd/DapperMime)) results in this error:

```
`probe-rs-debugger`: [2022-09-25T11:57:41Z ERROR probe_rs_debugger::debugger::debug_entry] probe-rs-debugger session ended: An error with the usage of the probe occurred
```

However, `cargo embed` works just fine (I can flash and view my defmt logs via the RTTUI), so I can rule out hardware issues.

**To Reproduce**

Using the following `launch.json`, and the aforementioned debugger setup:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe-rs debug",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": true,
            "chip": "rp2040",
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": false
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "./target/thumbv6m-none-eabi/debug/pim",
                    "rttEnabled": true,
                    "rttChannelFormats": [
                        {
                            "channelNumber": 0,
                            "dataFormat": "Defmt",
                        }
                    ]
                }
            ]
        }
    ]
}
```

**Expected behavior**

Expected to successfully connect to the debugger.

cstrahan, 2022-09-25:
D'oh -- I gave the options another once over, and noticed `"connectUnderReset": false` -- don't know how I kept overlooking that. Suspected that might be the issue -- sure enough, setting that to `false` resolves the issue.

noppej, 2022-09-26:
Thanks for the sleuthing. I'm re-opening this, because I think it would be helpful if we caught this in the code, and warned the user (or ignore and warn).

closed 2022-10-26 by bors[bot]