probe-rs/probe-rs #1242: Errors flashing XMC1400 with JLink on BOOT-KIT board

TheTinderstick, 2022-09-26:
**Describe the bug**
Unable to flash a XMC1400 (Cortex-M0) on the BOOT-KIT eval board.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate chip family description file from CMSIS pack using target-gen
2. Build example executable
3. Try flashing example using cargo flash (command see stacktrace)

**Expected behavior**
Erase and flash chip

**Stacktrace**
<details><summary>Huge trace</summary>
PS C:\Users\stefa\Rust\test-xmc1400> cargo flash --chip-description-path ./chip_family/XMC1000.yaml --chip XMC1404-Q064x0200 --log debug --elf .\target\thumbv6m-none-eabi\release\hello --
       DEBUG cargo_flash > Changed working directory to .
    Flashing .\target\thumbv6m-none-eabi\release\hello
       DEBUG probe_rs::config::registry > Searching registry for chip with name XMC1404-Q064x0200
       DEBUG probe_rs::config::registry > Exact match for chip name: XMC1404-Q064x0200
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb      
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
       DEBUG jaylink                          > libusb 1.0.26.11724
       DEBUG jaylink                          > libusb has capability API: true       
       DEBUG jaylink                          > libusb has HID access: true
       DEBUG jaylink                          > libusb has hotplug support: false     
       DEBUG jaylink                          > libusb can detach kernel driver: false
       DEBUG jaylink                          > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                          > scanning 3 interfaces
       DEBUG jaylink                          > J-Link interface is #2
       DEBUG jaylink                          > legacy caps: GetHwVersion | ReadConfig | WriteConfig | SpeedInfo | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | Indicators | TestNetSpeed | GetCapsEx
       DEBUG jaylink::capabilities            > unknown ext. capability bits: 0x2140067C8DB8EA5A33 truncated to 0x1B8EA5A33 (Reserved0 | GetHwVersion | ReadConfig | WriteConfig | SpeedInfo | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite)
       DEBUG jaylink                          > extended caps: GetHwVersion | ReadConfig | WriteConfig | SpeedInfo | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 1366:0105 in CMSIS-DAP v1 mode
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
       DEBUG jaylink                                > open_usb: device descriptor: DeviceDescriptor {
    bLength: 0x12,
    bDescriptorType: 0x1,
    bcdUSB: 0x200,
    bDeviceClass: 0xef,
    bDeviceSubClass: 0x2,
    bDeviceProtocol: 0x1,
    bMaxPacketSize: 0x40,
    idVendor: 0x1366,
    idProduct: 0x105,
    bcdDevice: 0x100,
    iManufacturer: 0x1,
    iProduct: 0x2,
    iSerialNumber: 0x3,
    bNumConfigurations: 0x1,
}
       DEBUG jaylink                                > scanning 3 interfaces
       DEBUG jaylink                                > J-Link interface is #2
       DEBUG jaylink                                > legacy caps: GetHwVersion | ReadConfig | WriteConfig | SpeedInfo | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | Indicators | TestNetSpeed | GetCapsEx
       DEBUG jaylink::capabilities                  > unknown ext. capability bits: 0x2140067C8DB8EA5A33 truncated to 0x1B8EA5A33 (Reserved0 | GetHwVersion | ReadConfig | WriteConfig | SpeedInfo | GetMaxBlockSize | GetHwInfo | 
[... 391 lines cut ...]
    ),
    speed: Some(
        "0",
    ),
    release: "0.13.0",
    commit: "crates.io",
}
     Erasing sectors ⠁ [00:00:00] [########################################################################################################################################]       0B/      0B @       0B/s (eta 0s )
       Error Something during the interaction with the core went wrong

  Caused by:
          0: Something during the interaction with the core went wrong
          1: A core architecture specific error occurred
          2: Failed to write register DRW at address 0x0000000c
          3: An error specific to the selected architecture occurred
          4: Target device did not respond to request.
</details>

**Desktop (please complete the following information):**
 - Windows 10

**Additional context**
The BOOT-KIT has an attached J-Link debugger. The board and debugger work with DAVE using Segger drivers. Drivers have been replaced using Zadig as described in documentation. The debugger firmware is updated.

I am a bit lost here. Am I missing something? Is this supposed to work?

Yatekii, 2022-09-26:
Can you please paste an unmangled trace? :)
Also, the trace suggests that the device id not respond to a request, but it's hard to tell from the mangled trace.
Have you tried `--connect-under-reset`? That fixes some issues.

TheTinderstick, 2022-09-26:
Hi,

I attached logs for the command without --connect-under-reset ([xmc1400_flash_error.log](https://github.com/probe-rs/probe-rs/files/9649572/xmc1400_flash_error.log)) and with ([xmc1400_flash_error_2.log](https://github.com/probe-rs/probe-rs/files/9649574/xmc1400_flash_error_2.log)). It doesnt work with the flag but the error looks different and I cant see what went wrong from the trace.

I also looked through older issues with similar error messages but didnt find any hints really.

willglynn, 2022-12-13:
@TheTinderstick I recently added support Infineon XMC4xxx processors (#1301), which I found had a bizarre boot strategy and a CMSIS pack that failed to describe the unusual debug sequences. The XMC1xxx reference manual indicates similar but not identical requirements.

[image]

This target family will need careful translation of the manual into a custom sequence too.

willglynn, 2022-12-13:
Like the XMC4s, the XMC1s have system software which runs after reset and before the user code. Unlike the XMC4s, the way to warm-reset-and-halt is to attach, halt, write the `UMHAR` boot mode to `SCU.GCU`->`SSW0` (making sure the upper and lower half-words are inverses of each other 🙄), command a reset, wait for the system software to finish and enter a NOP loop, halt the core again, and update the PC to point to the start of the application code.

[image]

[image]

Ugh.

closed 2026-09-30 by 9names