probe-rs/probe-rs #1251: Specify stack size for flash algorithm

mciantyre, 2022-10-01:
**Is your feature request related to a problem? Please describe.**

I'm looking for a way to configure the stack size for my device's flashing algorithm. I'll use this to avoid a stack overrun that corrupts my flashing algorithm.

<details>
<summary>Observations and testing...</summary>

When developing #1250, probe-rs would fail to flash the Cortex M7 on an MIMXRT1170 EVK. It would timeout when trying to program the *second* flash page. Fault status registers revealed an instruction bus error (`IBUSERR`).

Here's how my flash algorithm is assembled in RAM. I guessed that if a stack overrun occured, the data section and `ProgramPage` instructions would change. This could generate an `IBUSERR` depending on how the stack overran the memory.

```
INFO  probe_rs::flashing::flasher  > chosen RAM to run the algo: RamRegion { name: Some("OCRAM"), range: 20240000..20380000, is_boot_memory: false, cores: ["cm7", "cm4"] }
INFO  probe_rs::flashing::flasher  > Flash algorithm assembly:
INFO  probe_rs::flashing::flasher  >    Load address: 0x20240000
INFO  probe_rs::flashing::flasher  >            Init: 0x20240093
INFO  probe_rs::flashing::flasher  >          Uninit: 0x20240173
INFO  probe_rs::flashing::flasher  >     EraseSector: 0x2024018D
INFO  probe_rs::flashing::flasher  >     ProgramPage: 0x202401B1
INFO  probe_rs::flashing::flasher  >     Static base: 0x20242AF8
INFO  probe_rs::flashing::flasher  >           Stack: 0x20242D30
INFO  probe_rs::flashing::flasher  >      Begin data: 0x20242D30
INFO  probe_rs::flashing::flasher  >   Page buffer 1: 0x20242D30
INFO  probe_rs::flashing::flasher  >   Page buffer 2: 0x20242E30
```

I think I showed that `ProgramPage` was overwriting its own instructions. I took the instruction check routine and [ran it after each failed programming operation](https://github.com/mciantyre/probe-rs/commit/5c9b4dbdce9b42183ff95817becae967610c0ed9). Logs indicate that a word after the `ProgramPage` function address changed after programming the first flash page.

```
DEBUG probe_rs::flashing::flasher  > Preparing Flasher for operation Program
DEBUG probe_rs::flashing::flasher  > Running init routine.
DEBUG probe_rs::flashing::flasher  > Calling routine 20240093(Some(805306368), Some(0), Some(2), None, init=true)
DEBUG probe_rs::flashing::flasher  > content of PC 0xf: 0x20240092 should be: 0x20240093
DEBUG probe_rs::flashing::flasher  > content of a1 0x0: 0x30000000 should be: 0x30000000
DEBUG probe_rs::flashing::flasher  > content of a2 0x1: 0x00000000 should be: 0x00000000
DEBUG probe_rs::flashing::flasher  > content of a3 0x2: 0x00000002 should be: 0x00000002
DEBUG probe_rs::flashing::flasher  > content of R9 0x9: 0x20242af8 should be: 0x20242af8
DEBUG probe_rs::flashing::flasher  > content of SP 0xd: 0x20242d30 should be: 0x20242d30
DEBUG probe_rs::flashing::flasher  > content of LR 0xe: 0x20240001 should be: 0x20240001
DEBUG probe_rs::flashing::flasher  > Waiting for routine call completion.
INFO  probe_rs::flashing::flasher  > Flashing page at address 0x30000000 with size: 256
DEBUG probe_rs::flashing::flasher  > Calling routine 202401b1(Some(805306368), Some(256), Some(539241776), None, init=false)
DEBUG probe_rs::flashing::flasher  > content of PC 0xf: 0x202401b0 should be: 0x202401b1
DEBUG probe_rs::flashing::flasher  > content of a1 0x0: 0x30000000 should be: 0x30000000
DEBUG probe_rs::flashing::flasher  > content of a2 0x1: 0x00000100 should be: 0x00000100
DEBUG probe_rs::flashing::flasher  > content of a3 0x2: 0x20242d30 should be: 0x20242d30
DEBUG probe_rs::flashing::flasher  > content of LR 0xe: 0x20240001 should be: 0x20240001
DEBUG probe_rs::flashing::flasher  > Waiting for routine call completion.
INFO  probe_rs::flashing::flasher  > Flashing took: 85.003185ms
INFO  probe_rs::flashing::flasher  > Flashing page at address 0x30000100 with size: 256
DEBUG probe_rs::flashing::flasher  > Calling routine 202401b1(Some(805306624), Some(256), Some(539241776), None, init=false)
DEBUG probe_rs::flashing::flasher  > content of PC 0xf: 0x202401b0 should be: 0x202401b1
DEBUG probe_rs::flashing::flasher  > content of a1 0x0: 0x30000100 should be: 0x30000100
DEBUG probe_rs::flashing::flasher  > content of a2 0x1: 0x00000100 should be: 0x00000100
DEBUG probe_rs::flashing::flasher  > content of a3 0x2: 0x20242d30 should be: 0x20242d30
DEBUG probe_rs::flashing::flasher  > content of LR 0xe: 0x20240001 should be: 0x20240001
DEBUG probe_rs::flashing::flasher  > Waiting for routine call completion.
ERROR probe_rs::flashing::flasher  > Failed to verify flash algorithm. Data mismatch at address 0x2024228c
ERROR probe_rs::flashing::flasher  > Original instruction: 0x7fffffff
ERROR probe_rs::flashing::flasher  > Readback instruction: 0x000000
Error: The flashing procedure failed for 'path/to/program.elf'.

Caused by:
The flash algorithm RAM contents did not match the expected contents.
```

After quadrupling [the default stack size](https://github.com/probe-rs/probe-rs/blob/7fb62550d68e3813b4e75dadc09b60ff07c876f1/probe-rs/src/flashing/flash_algorithm.rs#L157), I could program my device without error. Given these observations, I think that my black-box FLM needs more than 512 bytes of stack.
</details>

**Describe the solution you'd like**

The simplest approach might be a new target configuration, like `stack_size`, that overrides the default stack size. This would let us allocate extra stack space for this one target without affecting other targets.

**Describe alternatives you've considered**

These are just ideas in my head. I'm still naive about flash algorithms and their constraints.

_Align the stack to the highest address of each device's RAM_. I would expect this approach to safely give us more than 512 bytes of stack for each supported device, but I'm looking for concerns. This is similar to [how `cortex-m-rt` defaults to placing the stack](https://docs.rs/cortex-m-rt/0.7.1/cortex_m_rt/#_stack_start).

The flash algo's placement in RAM matters with this approach. If we were to place the stack above the data buffers, a stack overrun might corrupt the data buffers, resulting in a corrupt program image in flash. Without a way to detect stack overruns during programming, this could be really gnarly to troubleshoot.

_Align the stack below all other flash algo content_. This would be like [flip-link](https://github.com/knurling-rs/flip-link) for flash algorithms. I think we still need to know / guess the stack size in this approach, but it might make detection easier on some targets.

**Additional context**

In this issue, I'm focusing on "configure the stack size _after you know your overrunning your stack._" Before we solve this, we might want to think of ways to detect stack overruns during flash algorithm execution. It could help others who are adding new devices. I'm treating this as a separate issue.

I'd be happy to contribute whatever stack size configuration / overrun detection solutions we come up with.

mciantyre, 2022-10-20:
I was curious if we could [increase the default stack size](https://github.com/probe-rs/probe-rs/blob/7fb62550d68e3813b4e75dadc09b60ff07c876f1/probe-rs/src/flashing/flash_algorithm.rs#L157) for all targets. I wrote [a quick script](https://github.com/probe-rs/probe-rs/commit/ae332321f3dde8be60dbc32068ec784ca1826c49) to reveal the built-in target with the smallest RAM region, and I found we're supporting the LPC810 with just 1K of RAM. Doesn't look like changing the default stack size would be an option.

mciantyre, 2022-12-07:
Sorry for not following up after #1260. OK with me to close this. I'll note that I didn't create any separate issue for stack overrun detection, and I'm no longer positioned to contribute it right now.

Yatekii, 2022-12-07:
No worries, thanks a ton for the initial implementation :) I suggest leaving this open to track canary implementation.

Tiwalun, 2023-01-18:
Created a new issue to track stack overflow detection, and closing this to make the current state clearer.

closed 2023-01-18 by Tiwalun