probe-rs/probe-rs #1253: Raspberry Pi Pico + pico-debug: Failed to get information for AP ApAddress  { dp: Multidrop(11002927), ap: 0 }

afternoon, 2022-10-10:
**Describe the bug**
I'm using `probe-run` to debug a rust binary for the Raspberry Pi Pico/RP2040. I'm using [pico-debug](https://github.com/majbthrd/pico-debug/) as a probe.

**To Reproduce**
1. Install `pico-debug` by flashing the provided `pico-debug-gimmecache.uf2` to a Raspberry Pi Pico.
2. `$ probe-run --chip RP2040 target/thumbv6m-none-eabi/debug/microgroove --verbose --speed 1000`

**Expected behavior**
`probe-run` communicates with the `pico-debug` probe successfully.

**Desktop (please complete the following information):**
 - macOS Mojave

**Additional context**
Output from the `probe-run` command:

```
$ probe-run --chip RP2040 target/thumbv6m-none-eabi/debug/microgroove --verbose --speed 1000
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 2003fbb8, hard_fault: 10001f53 }
└─ probe_run::elf @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/elf.rs:31
(HOST) DEBUG RAM region: 0x20000000-0x2003FFFF
└─ probe_run::target_info @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/target_info.rs:115
(HOST) DEBUG section `.data` is in RAM at 0x2003FBB8..=0x2003FBEF
└─ probe_run::target_info @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/target_info.rs:153
(HOST) DEBUG section `.bss` is in RAM at 0x2003FBF0..=0x2003FBFB
└─ probe_run::target_info @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/target_info.rs:153
(HOST) DEBUG section `.uninit` is in RAM at 0x2003FBFC..=0x2003FFFB
└─ probe_run::target_info @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/target_info.rs:153
(HOST) DEBUG valid SP range: 0x20000000..=0x2003FBB4
└─ probe_run::target_info @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/target_info.rs:167
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/probe.rs:25
(HOST) DEBUG opened probe
└─ probe_run::probe @ /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/probe.rs:33
Error: Failed to get information for AP ApAddress { dp: Multidrop(11002927), ap: 0 }
```

Other additional context:

I'm a rust and embedded n00b. I'm drowning not waving. ;-)

afternoon, 2022-10-10:
BTW, I'm aware that other probe options exist, e.g. DapperMime, but I wanted to flag that pico-debug might not be supported by probe-rs correctly right now. pico-debug has the nice property that it is a virtual probe, running on the 2nd core of a RP2040. For apps only using a single core it does not require additional hardware.

Yatekii, 2022-10-10:
This issue should be fixed on `probe-rs` master. `probe-run` does not use master afaik. You could try with `probe-rs-cli` form master :)

afternoon, 2022-10-11:
Thanks for the speedy reply! Great to know this is fixed. I'm actually trying to use `cargo-embed`. I pulled the latest version and tried to build it with the master version of `probe-rs`, but got a stack of errors. I think I'll wait until the next releases of `probe-rs` and `cargo-embed` are cut. Thanks again for your help, and for an amazing project!

Yatekii, 2022-10-11:
You can use `--locked` to make `cargo-embed` build :)

afternoon, 2022-10-11:
OK, thanks! I'm afraid I'm not sure how to do that. I tried updating `Cargo.toml` to point only to git for `probe-rs` and running `cargo update`. I tried doing `cargo build --locked`. Both gave the following error output:

```
warning: skipping duplicate package `binary_project` found at `/Users/ben/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/7fb6255/probe-rs-cli-util/tests/data/binary_project`
warning: skipping duplicate package `library_project` found at `/Users/ben/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/7fb6255/probe-rs-cli-util/tests/data/library_with_example_project`
    Updating crates.io index
  Downloaded iana-time-zone v0.1.45
  Downloaded utf8-width v0.1.5
  Downloaded serde_yaml v0.9.11
  Downloaded bitvec v1.0.0
  Downloaded paste v1.0.7
  Downloaded semver v1.0.10
  Downloaded parking_lot v0.12.0
  Downloaded yansi v0.5.0
  Downloaded regex-syntax v0.6.26
  Downloaded unicode-linebreak v0.1.2
  Downloaded unicode-segmentation v1.9.0
  Downloaded libc v0.2.126
  Downloaded unsafe-libyaml v0.2.2
  Downloaded uncased v0.9.6
  Downloaded clap_derive v3.1.2
  Downloaded clap v3.1.3
  Downloaded camino v1.0.7
  Downloaded byte-unit v4.0.13
  Downloaded hidapi v1.3.3
  Downloaded rusb v0.9.0
  Downloaded console v0.15.0
  Downloaded lock_api v0.4.6
  Downloaded io-lifetimes v0.7.2
  Downloaded pkg-config v0.3.24
  Downloaded signal-hook v0.3.13
  Downloaded mio v0.8.0
  Downloaded parking_lot_core v0.9.1
  Downloaded once_cell v1.13.0
  Downloaded regex v1.5.6
  Downloaded syn v1.0.92
  Downloaded time v0.3.11
  Downloaded libusb1-sys v0.6.0
  Downloaded rustix v0.35.7
  Downloaded jep106 v0.2.6
  Downloaded gimli v0.26.1
  Downloaded gdbstub v0.6.2
  Downloaded 36 crates (4.4 MB) in 1.60s
   Compiling proc-macro2 v1.0.36
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.92
   Compiling libc v0.2.126
   Compiling memchr v2.4.1
   Compiling pkg-config v0.3.24
   Compiling itoa v1.0.1
   Compiling ryu v1.0.9
   Compiling regex-syntax v0.6.26
   Compiling hashbrown v0.12.3
   Compiling unsafe-libyaml v0.2.2
   Compiling rusb v0.9.0
   Compiling once_cell v1.13.0
   Compiling unicode-width v0.1.9
   Compiling parking_lot_core v0.9.1
   Compiling io-lifetimes v0.7.2
   Compiling signal-hook v0.3.13
   Compiling rustix v0.35.7
   Compiling yansi v0.5.0
   Compiling smallvec v1.8.0
   Compiling semver v1.0.10
   Compiling camino v1.0.7
   Compiling termcolor v1.1.2
   Compiling either v1.6.1
   Compiling paste v1.0.7
   Compiling utf8-width v0.1.5
   Compiling textwrap v0.14.2
[... 165 lines cut ...]

error[E0308]: mismatched types
   --> src/main.rs:368:59
    |
368 |                         erase_progress.enable_steady_tick(100);
    |                                        ------------------ ^^^ expected struct `std::time::Duration`, found integer
    |                                        |
    |                                        arguments to this function are incorrect
    |
note: associated function defined here
   --> /Users/ben/.cargo/registry/src/github.com-1ecc6299db9ec823/indicatif-0.17.1/src/progress_bar.rs:152:12
    |
152 |     pub fn enable_steady_tick(&self, interval: Duration) {
    |            ^^^^^^^^^^^^^^^^^^

error[E0599]: no method named `join` found for struct `MultiProgress` in the current scope
   --> src/main.rs:420:32
    |
420 |                 multi_progress.join().unwrap();
    |                                ^^^^ method not found in `MultiProgress`

Some errors have detailed explanations: E0308, E0599.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `cargo-embed` due to 6 previous errors
```

Yatekii, 2022-10-11:
`--locked` only works with install :) You can try: https://github.com/probe-rs/cargo-embed/pull/386

afternoon, 2022-10-11:
Wow! Loving the full service. 😆

From the `cargo-embed` clone root, I did:

```
$ git pull
$ git co 054841e3d672aee858f1446d098783f6ec4c85fa
$ cargo uninstall cargo-embed
$ cargo install --locked --path .
```

Back in my project running `cargo embed` still spits out the same error. I can see from the `Cargo.lock` that its pulling in `probe-rs` version `7fb62550`, so this is surprising.

```
    Finished dev [optimized + debuginfo] target(s) in 0.07s
      Config default
      Target /Users/ben/Desktop/microgroove-rs/target/thumbv6m-none-eabi/debug/microgroove
        WARN probe_rs::probe::cmsisdap > Giving up on TARGETSEL, too many retries.
       Error failed attaching to target

             Caused by:
                 Failed to get information for AP ApAddress { dp: Multidrop(11002927), ap: 0 }
```

`probe-rs-cli list` finds the `pico-debug` probe:

```
The following devices were found:
[0]: CMSIS-DAP (VID: 1209, PID: 2488, Serial: E661410403292C32, CmsisDap)
```

afternoon, 2022-10-12:
I’ll close this issue since a fix is in master. I’ll reopen if I bump into this issue again later. Thanks so much for your help @Yatekii!

closed 2022-10-12 by afternoon

Urhengulas, 2022-12-09:
> This issue should be fixed on `probe-rs` master. `probe-run` does not use master afaik. You could try with `probe-rs-cli` form master :)

That's great news that this is already fixed! We had some people report that issue. Is is possible to get a patch release with the fix?

Yatekii, 2022-12-09:
@Urhengulas Yes, I am trying to get up to speed again with all development things and once I am I will release a minor release even :)

Urhengulas, 2022-12-09:
> @Urhengulas Yes, I am trying to get up to speed again with all development things and once I am I will release a minor release even :)

That sounds great. Thank you!